LibreELEC/LibreELEC.tv #7750: [BUG] Linux Realtek patch fails

gonciarz, 2023-04-05:
### Describe the bug
I cannot build Libreelec for Odroid C2.
This patch cannot be applied: packages/linux/patches/default/linux-122-rtw88-sdio-rfc-v1.patch

### To Reproduce
Steps to reproduce the behaviour:

I used Archlinux this package to build (I didn't have any problem to build previous revisions):
https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=libreelec-git
but the problem is not distribution related.

This is the command / project I execute:
`PROJECT=Amlogic DEVICE=AMLGX ARCH=arm UBOOT_SYSTEM=odroid-c2 make image`

### Informations
 - LE Version: master, rev: d4ddf07 (12 dev), linux git rev: d4ddf07
 - Hardware Platform: Amlogic (Odroid C2)

### Log file
```
    APPLY PATCH (common - default)      packages/linux/patches/default/linux-122-rtw88-sdio-rfc-v1.patch
patching file include/linux/mmc/sdio_ids.h
patching file drivers/net/wireless/realtek/rtw88/rtw8821c.c
patching file drivers/net/wireless/realtek/rtw88/rtw8821c.h
patching file drivers/net/wireless/realtek/rtw88/rtw8822b.c
patching file drivers/net/wireless/realtek/rtw88/rtw8822b.h
patching file drivers/net/wireless/realtek/rtw88/mac.c
patching file drivers/net/wireless/realtek/rtw88/rtw8822c.c
patching file drivers/net/wireless/realtek/rtw88/rtw8822c.h
patching file drivers/net/wireless/realtek/rtw88/hci.h
patching file drivers/net/wireless/realtek/rtw88/mac.c
Hunk #1 FAILED at 269.
1 out of 1 hunk FAILED -- saving rejects to file drivers/net/wireless/realtek/rtw88/mac.c.rej
patching file drivers/net/wireless/realtek/rtw88/mac.c
Hunk #1 succeeded at 1017 (offset -24 lines).
Hunk #2 succeeded at 1182 (offset -24 lines).
patching file drivers/net/wireless/realtek/rtw88/mac.c
Hunk #5 FAILED at 301.
Hunk #6 succeeded at 465 (offset -8 lines).
Hunk #7 succeeded at 1064 (offset -24 lines).
1 out of 7 hunks FAILED -- saving rejects to file drivers/net/wireless/realtek/rtw88/mac.c.rej
patching file drivers/net/wireless/realtek/rtw88/Kconfig
patching file drivers/net/wireless/realtek/rtw88/Makefile
patching file drivers/net/wireless/realtek/rtw88/debug.h
patching file drivers/net/wireless/realtek/rtw88/mac.h
patching file drivers/net/wireless/realtek/rtw88/reg.h
patching file drivers/net/wireless/realtek/rtw88/sdio.c
patching file drivers/net/wireless/realtek/rtw88/sdio.h
patching file drivers/net/wireless/realtek/rtw88/main.c
Hunk #2 succeeded at 1786 (offset 2 lines).
patching file drivers/net/wireless/realtek/rtw88/main.c
Hunk #1 succeeded at 2163 (offset 2 lines).
patching file drivers/net/wireless/realtek/rtw88/ps.h
patching file drivers/net/wireless/realtek/rtw88/Kconfig
patching file drivers/net/wireless/realtek/rtw88/Makefile
patching file drivers/net/wireless/realtek/rtw88/rtw8822bs.c
patching file drivers/net/wireless/realtek/rtw88/Kconfig
patching file drivers/net/wireless/realtek/rtw88/Makefile
patching file drivers/net/wireless/realtek/rtw88/rtw8822cs.c
patching file drivers/net/wireless/realtek/rtw88/Kconfig
patching file drivers/net/wireless/realtek/rtw88/Makefile
patching file drivers/net/wireless/realtek/rtw88/rtw8821cs.c
*********** FAILED COMMAND ***********
patch -d "${PKG_BUILD}" -p1 < ${i} 1>&${VERBOSE_OUT}
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/unpack "${PKG_NAME}" "${PARENT_PKG}"
**************************************
FAILURE: scripts/build linux:host has failed!
```

### Additional context
N/A

gonciarz, 2023-04-05:
Patch, original file and rejected one:
[https://file.io/44G6E2MzTLNW](https://file.io/44G6E2MzTLNW)
[https://easyupload.io/t5v7tg](https://easyupload.io/t5v7tg)

heitbaum, 2023-04-06:
Hi @gonciarz - known issue that will be resolved when Linux Kernel is next bumped for Amlogic in master. Amlogic in master is still 6.1.14, whereas the patch shared across the builds supports 6.1.19. LE11 has had 7603 applied (supports this patch)

closed 2023-04-06 by heitbaum

gonciarz, 2023-04-06:
@heitbaum two questions:
1. stupid question: where can I find amlogic repository? And what is a typical timeline till they catch up?
2. I believe I've seen a case statement in your building scripts. Depending on platform you may apply two different directories.
If we need to wait for an Amlogic update, can we create two directories of patches. I know it's an overhead but at least kind of workaround

```
linux]$ tree
.
├── package.mk
├── patches
│   ├── default
│   │   ├── linux-020-eld-constraints-for-compressed-formats.patch
│   │   ├── linux-051-ouya_controller_support.patch
│   │   ├── linux-058.05-hid_sony-add_autorepeat_for_PS3_remotes.patch
│   │   ├── linux-062-imon_pad_ignore_diagonal.patch
│   │   ├── linux-121-rtw88-linux-next-6-2.patch
│   │   ├── linux-122-rtw88-fix-rcu-lock.patch
│   │   ├── linux-122-rtw88-sdio-pre-patches-from-next-6-3.patch
│   │   ├── linux-122-rtw88-USB-fixes.patch
│   │   ├── linux-150-dev_input_xpad_add_8bitdo_pro_2_wired_controller_support.patch
│   │   ├── linux-999.02-0001-pm-disable-async-suspend-resume-by-default.patch
│   │   └── linux-999-no-lzma-in-x86-perf-build.patch
│   └── raspberrypi
│       ├── linux-010-fix-vc4-rgb-check.patch
│       ├── linux-020-eld-constraints-for-compressed-formats.patch
│       ├── linux-051-ouya_controller_support.patch
│       ├── linux-058.05-hid_sony-add_autorepeat_for_PS3_remotes.patch
│       ├── linux-062-imon_pad_ignore_diagonal.patch
│       └── linux-999.02-0001-pm-disable-async-suspend-resume-by-default.patch
```

gonciarz, 2023-04-06:
@heitbaum I've also found a similar issue for ffmpeg patch:

```
<< ffmpeg:target seq 215 <<<
UNPACK      ffmpeg
    APPLY PATCH (common - libreelec)      packages/multimedia/ffmpeg/patches/libreelec/ffmpeg-001-libreelec.patch
patching file libavcodec/libdav1d.c
Hunk #1 succeeded at 58 (offset -8 lines).
Hunk #2 FAILED at 400.
1 out of 2 hunks FAILED -- saving rejects to file libavcodec/libdav1d.c.rej

```
Shall I open another issue?

chewitt, 2023-04-06:
Clone https://github.com/chewitt/LibreELEC.tv/commits/amlogic and build from that. It works for me ™️

gonciarz, 2023-04-06:
Thanks @chewitt, although I cannot compile the head (git rev: 630078e)
[073/286] [DONE]
Now I have a problem with flatbuffers. Do you use mark revisions that were able to build?

```
libreelec-git-amlogic/libreelec-git/src/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/build/flatbuffers-22.12.06/include/flatbuffers/flatbuffer_builder.h:582:51: note:   no known conversion for argument 1 from 'std::string' {aka 'std::__cxx11::basic_string<char>'} to 'const flatbuffers::String*'
  582 |   Offset<String> CreateSharedString(const String *str) {
      |                                     ~~~~~~~~~~~~~~^~~
ninja: build stopped: subcommand failed.
FAILURE: scripts/build flatbuffers:host during make_host (default)
*********** FAILED COMMAND ***********
ninja ${NINJA_OPTS} ${PKG_MAKE_OPTS_HOST}
**************************************
FAILURE: scripts/build flatbuffers:host has failed!

```

gonciarz, 2023-04-06:
@chewitt do you use amlogic branch as a source for nightly builds?
https://test.libreelec.tv/11.0/Amlogic/odroid-c2/

heitbaum, 2023-04-07:
> @chewitt do you use amlogic branch as a source for nightly builds?
> https://test.libreelec.tv/11.0/Amlogic/odroid-c2/

We use https://github.com/LibreELEC/LibreELEC.tv via https://github.com/LibreELEC/actions for all our nightly actions. LE12 is in its infantry and there are some bigger updates in the not too distant future that the team are working (and focussed) on, which will include kernel / u-boot / toolchain. That will address not just this - but setup the baseline for the future release.

chewitt, 2023-04-07:
LE11 nightlies are built from the LE repo `libreelec-11.0` branch. I build ad-hoc images from my own branches (mostly the `amlogic` one that I linked) which has bunch of WIP things included; currently u-boot 2023.04 and Linux 6.2 on-top of the LE11 codebase. It works for me ™️  building on an Ubuntu 22.04 host; I've done a full clean build this morning. If build on Arch isn't working for you perhaps use an Ubuntu container, there are Dockerfiles in our repo.

gonciarz, 2023-04-07:
@chewitt I'll try to narrow the problem if it's my env. A month ago I was able to build 07249c1 (LE master).
But there are just few dependencies (https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=libreelec-git)
Almost every package is build from source. I see you guys did a lot of great work.
But thank you for your suggestion. If I can't build I'll run in from container.

@heitbaum GH actions are great. 

@chewitt sorry for spamming, do you know what's the fate for Odroid C2. Is it still going to be supported by LE?
Basically LE is almost all I need for my SoC, but I like flexibility and want to be able to e.g. run some additional service (perhaps docker setup would be more than enough) and I want to be able to update to newer Kodi versions.
That's why I'm considering moving some of your work to Archlinux platform (kernel, uboot, etc.). But I see that archlinuxarm isn't maintain distro. It's almost dead. Much better choice would be to switch to armbian or even Gentoo because you build everything from scratch. But when I look at all those commits. Maintaining those low level packages is a really a pain. I really appreciate the effort you have done.
Btw is Amlogic going to merge their code to upstream kernel?

chewitt, 2023-04-07:
Amlogic maintains their own Android/buildroot codebase(s) for their paying customers. They will upstream minor things that suit them to the usptream kernel but development is mostly a community effort (nothing new about that).

Re. C2 support; read the release notes: https://libreelec.tv/2023/03/24/libreelec-nexus-11-0-1/

gonciarz, 2023-04-07:
@chewitt perfect news, I was surprised initially because C2 was able to run HEVC in 4k and 10-bit and it was dropped.
I will just try to enable junest on LE
https://github.com/fsquillace/junest
or enable docker service and I don't need anything else.
Thank you :)
Now I'm switching for my new Rockpro64 with SAS PCI-E controller for my home NAS server.