LibreELEC/LibreELEC.tv #7848: [BUG] Bluetooth audio streaming, but no sound on any output

HorstBaerbel, 2023-05-17:
### Describe the bug
I have successfully connected my mobile to LE via bluetooth (LE settings addon) and can "play" audio from it. The mobile displays the play progress, the bluetooth adapter is blinking, but there is no sound on any of the audio output devices connected to the system. Kodi menu sound is there though, so audio output in general works.
I've tried [this](https://wiki.libreelec.tv/configuration/pulseaudio#bluetooth-receiving), and I can see and set the correct bluetooth source and audio sink. They are successfully being set in ~/.config/pulse/576623bb652909e80931523c6462cb7d-default-sink and ~/.config/pulse/576623bb652909e80931523c6462cb7d-default-source.

### To Reproduce
Steps to reproduce the behavior:
1. Connect bluetooth source
2. Play auto on bluetooth source
3. No audio output

### Informations
 - LE Version: 11.0.1 x86
 - Hardware Platform: Ryzen 2400G

### Log file
Logs can be found [here](http://ix.io/4w2F)

lrusak, 2023-05-17:
This isn't the intended behavior of the Bluetooth audio connection.

You can do what you want but it requires manual setup.

See: https://wiki.libreelec.tv/configuration/pulseaudio#bluetooth-receiving

closed 2023-05-17 by lrusak

HorstBaerbel, 2023-05-17:
> This isn't the intended behavior of the Bluetooth audio connection.
> 
> You can do what you want but it requires manual setup.
> 
> See: https://wiki.libreelec.tv/configuration/pulseaudio#bluetooth-receiving
> 
> Please take any further discussion to the forum.

I've tried that setup page you linked to. It is not working for me.

lrusak, 2023-05-17:
Please use the forum for further support.