LibreELEC/LibreELEC.tv #7916: LE crash after calling EPG

eldersixpack13, 2023-06-14:
### Describe the bug

after hitting key "e" (EPG) LE crashes.
I'm seeing this bug the 3. or 4. time during the last ~10 days.
so far I haven't tried if it is reproducibly (_getting an idea during writing here_)
calling EPG in general is without errors/crashes
but there seems to be a pattern: see "steps to reproduce..." 

Enviroment here:
- NUC 13
-  connected to an DigiBit
- DigiBit get it's current over an smart home socket
- smart home socket shuts auto. OFF when current gets under 9W (no SAT-TV- movie is running) for ~15 minutes

 smart home socket (and the DigiBit) ON is via an addon (Fritz!Box SmartHome)

### To Reproduce
Steps to reproduce the behavior / What I've done:
same following steps might be superfluous

1. started LE and watched SAT-TV for some time (<5 minutes). nothing interesting there and watched a movie from mediathek. smart home socket cuts off the DigiBit . Mediathek movie finished after ~1.5 h. 
2. start DigiBit via Fritz!Box SmartHome Addon
3. started EPG, selected a SAT-TV-movie
4. after a short SAT-TV-movie play I selected EPG again **=> LE crashed**

### Informations
 - LE Version: LE12 nightly-20230609-4c5023d (Generic.x86_64)
 - Hardware Platform: NUC13

to me it or the pattern seems to be:
LE start => EPG => TV-movie-play => Mediathek-movie-play => EPG => TV-Movie-Play => EPG => crash
theory:
during the long Mediathek-movie-play the DigiBit was power cutted here, means TVH (still running) doesn't get current EPG data.
the first EPG call survived (with outdated (?)  EPG data), but not the second call

something like that ..., cause - as said -  EPG usually doesn't make any trouble 

### Log file
crash log:
https://paste.centos.org/view/21f0ebc6

### Additional context
%

eldersixpack13, 2023-07-13:
yesterday the same crash with same conditions

heitbaum, 2023-07-13:
> yesterday the same crash with same conditions

Is the segv the same on the other crashes? 

```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x0000000000e15c32 in CGUIFontTTF::RenderCharacter(CGraphicContext&, float, float, CGUIFontTTF::Character const*, unsigned int, bool, std::vector<SVertex, std::allocator<SVertex> >&) ()
[Current thread is 1 (Thread 0x7fbe3e81e940 (LWP 906))]
```

eldersixpack13, 2023-07-13:
you mean "Thread 1" of the crash ?

if so, it reads:

`Thread 1 (Thread 0x7f1bdd528940 (LWP 930)):
#0  0x0000000000e17de2 in CGUIFontTTF::RenderCharacter(CGraphicContext&, float, float, CGUIFontTTF::Character const*, unsigned int, bool, std::vector<SVertex, std::allocator<SVertex> >&) ()
#1  0x0000000000e19507 in CGUIFontTTF::DrawTextInternal(CGraphicContext&, float, float, std::vector<unsigned int, std::allocator<unsigned int> > const&, std::vector<unsigned int, std::allocator<unsigned int> > const&, unsigned int, float, bool) ()
#2  0x0000000000e1a075 in CGUIFont::DrawText(float, float, std::vector<unsigned int, std::allocator<unsigned int> > const&, unsigned int, std::vector<unsigned int, std::allocator<unsigned int> > const&, unsigned int, float) ()
#3  0x0000000000e4f029 in CGUITextBox::Render() ()
#4  0x0000000000dfcfd4 in ?? ()
#5  0x0000000000df8bce in CGUIControlGroup::Render() ()
#6  0x0000000000dfcfd4 in ?? ()
#7  0x0000000000df8bce in CGUIControlGroup::Render() ()
#8  0x0000000000dfcfd4 in ?? ()
#9  0x0000000000df8bce in CGUIControlGroup::Render() ()
#10 0x0000000000dfcfd4 in ?? ()
#11 0x0000000000df8bce in CGUIControlGroup::Render() ()
#12 0x0000000000dfcfd4 in ?? ()
#13 0x0000000000e6a758 in ?? ()
#14 0x0000000000e6220b in CGUIWindowManager::RenderPass() const ()
#15 0x0000000000e6ad12 in CGUIWindowManager::Render() ()
#16 0x0000000000ee389e in CApplication::Render() ()
#17 0x0000000000e692c2 in CGUIWindowManager::ProcessRenderLoop(bool) ()
#18 0x0000000000e93b6a in CGUIDialogBusy::WaitOnEvent(CEvent&, unsigned int, bool) ()
#19 0x0000000000eef01f in CApplication::OnMessage(CGUIMessage&) ()
#20 0x0000000000e61221 in CGUIWindowManager::SendMessage(CGUIMessage&) ()
#21 0x0000000000e6e8d3 in CGUIWindowManager::DispatchThreadMessages() ()
#22 0x0000000000eed3dd in CApplication::Process() ()
#23 0x0000000000ee322f in CApplication::Run() ()
#24 0x000000000089d2a7 in main ()`

heitbaum, 2023-07-14:
> it seems the restart of TVH before EPG call prevents the crash.
>
> it could also be that the update of glibc (I'm now on nightly-20230712-5e1975a) prevents the crash.
> don't know, I'm not an developer.
>
> But I could do a test without restart of TVH next time .

It is a library call that is causing it. [as in that it is consistent] I’m not sure if RenderCharacter uses harfbuzz or glib. I had a quick look last night and couldn’t see any code change in RenderCharacter since 2014. But if it has gone away - then glib/harfbuzz would be the guess

eldersixpack13, 2023-07-14:
yep, I initial wrote to have also glib (by me misread as glibc)  on my radar .
then I realizes my mistake (glib is not glib**c**) and I deleted that part of my comment.

Anyway, I'll proof if the fix is the update for glib/harfbuzz by leaving off the restart of TVH next time...

eldersixpack13, 2023-07-16:
without restart of TVH LE crashes
tested was with nightly-20230714-788de87

eldersixpack13, 2023-07-20:
Next crash, but it seem outdated EPG data or long time idle of EPG are not a absolutely necessary condition.

I got the crash just after first start of LE and  a short time of "surfing" the EPG Guide

eldersixpack13, 2023-10-24:
never seen since some nightlies

closed 2023-10-24 by eldersixpack13