LibreELEC/LibreELEC.tv #7917: [BUG] host nspr not checked with check-deps

mittorn, 2023-06-14:
### Describe the bug
libnss depends on nspr and host libnss build fails

### To Reproduce
Steps to reproduce the behavior:
1. Install gentoo stage3
2. Run arm target image build
3. ld fails to find -lnspr4

### Informations
 - LE Version: master
 - Hardware Platform: rk3399

heitbaum, 2023-06-14:
@mittorn - can you please share more information as I just tested separately and our build server runs the build successfully nightly.

`PROJECT=Rockchip ARCH=arm DEVICE=RK3399 scripts/build nss`

And confirming that this built the dependencies as expected.

```
docker@a1e48afe476e:/var/media/DATA/home-rudi/LibreELEC.kernel11$ PROJECT=Rockchip ARCH=arm DEVICE=RK3399 scripts/build nss:host
docker@a1e48afe476e:/var/media/DATA/home-rudi/LibreELEC.kernel11$ PROJECT=Rockchip ARCH=arm DEVICE=RK3399 scripts/build nspr:host
docker@a1e48afe476e:/var/media/DATA/home-rudi/LibreELEC.kernel11$ PROJECT=Rockchip ARCH=arm DEVICE=RK3399 scripts/build nspr
docker@a1e48afe476e:/var/media/DATA/home-rudi/LibreELEC.kernel11$
```

mittorn, 2023-06-14:
In is not docker build. I understand that it builds successfully in official environment, but check_deps supports gentoo and just missing some package checks

heitbaum, 2023-06-15:
> In is not docker build. I understand that it builds successfully in official environment, but check_deps supports gentoo and just missing some package checks

Could you please share the logs of your error to assist you in identifying the issue. We don’t use a system nspr or the like - but compile it in the build. So the issue won’t be a checkdep nspr issue as such.

```
$ find build.LibreELEC-RK3399.arm-12.0-devel/ | grep libnspr
build.LibreELEC-RK3399.arm-12.0-devel/build/nss-3.90/nspr/pr/src/libnspr4.a
build.LibreELEC-RK3399.arm-12.0-devel/build/nss-3.90/nspr/pr/src/libnspr4.so
build.LibreELEC-RK3399.arm-12.0-devel/build/nss-3.90/nspr/dist/bin/libnspr4.so
build.LibreELEC-RK3399.arm-12.0-devel/build/nss-3.90/nspr/dist/lib/libnspr4.a
build.LibreELEC-RK3399.arm-12.0-devel/build/nss-3.90/nspr/dist/lib/libnspr4.so
build.LibreELEC-RK3399.arm-12.0-devel/build/nss-3.90/nss/automation/abi-check/expected-report-libnspr4.so.txt
build.LibreELEC-RK3399.arm-12.0-devel/toolchain/lib/libnspr4.a
build.LibreELEC-RK3399.arm-12.0-devel/toolchain/lib/libnspr4.so
build.LibreELEC-RK3399.arm-12.0-devel/toolchain/armv8a-libreelec-linux-gnueabihf/sysroot/usr/lib/libnspr4.a
build.LibreELEC-RK3399.arm-12.0-devel/toolchain/armv8a-libreelec-linux-gnueabihf/sysroot/usr/lib/libnspr4.so
build.LibreELEC-RK3399.arm-12.0-devel/install_pkg/nspr-4.35/usr/lib/libnspr4.a
build.LibreELEC-RK3399.arm-12.0-devel/install_pkg/nspr-4.35/usr/lib/libnspr4.so
```

mittorn, 2023-06-15:
so maybe dist libs are ignored and it uses system libs?

heitbaum, 2023-06-24:
Closing. If further details are available / logs. Can be reopened.

closed 2023-06-24 by heitbaum