LibreELEC/LibreELEC.tv #7920: [Documentation] custom EDID instructions are broken

mittorn, 2023-06-14:
### Describe the bug
https://wiki.libreelec.tv/configuration/edid
Adding custom edid with initrd breaks boot because kernel already includes initrd and will not mount squashfs without it.
And there seems to be no ability to regerenrate initramfs correctly.
I spend half-a-day building custom libreelec which supports my display from source and did not find correct way to include edid in initramfs <build>/initramfs files just gets rewritten on every rebuild. Is there
### To Reproduce
Steps to reproduce the behavior:
1. Generate custom initrd as described in docs
2. Add it to extlinux config
3. System now panics tying to find root

### Informations
 - LE Version: master
 - Hardware Platform: RK3399

the-dreamer, 2023-06-29:
I can say from my side, custom edid is working on x86 and rpi4 well. You have not given any config file or crashlog. i would say, nobody knows what is happend on your side.

From my understanding edid.cpio is only necessary for early display detection/init. the normal edid is enough for me, as i don't need for the first 5 seconds any display. The normal edid file is used by kernel after display stack is inited. the cpio is not used after init. 

btw. the getedid script should do everything what is necessary for a working custom edid. i have always edited the extlinux config after the script because of resolution i want to have.

heitbaum, 2023-09-04:
Not bug. Please use forum for support.

closed 2023-09-04 by heitbaum