LibreELEC/LibreELEC.tv #7926: [Build Issue] u-boot-tools (AMLGX) does not build on aarch64

heitbaum, 2023-06-17:
### Describe the bug
Building AMLGX image on Ampere A1 fails.

Either Wrong flags / wrong compiler (if cross required)

```
BUILD      u-boot-tools (host)
    TOOLCHAIN      make (auto-detect)
make[1]: Entering directory '/build/build.LibreELEC-AMLGX.arm-12.0-devel/build/u-boot-tools-2022.10'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  LEX     scripts/kconfig/zconf.lex.c
  YACC    scripts/kconfig/zconf.tab.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
#
# configuration written to .config
#
make[1]: Leaving directory '/build/build.LibreELEC-AMLGX.arm-12.0-devel/build/u-boot-tools-2022.10'
make[1]: Entering directory '/build/build.LibreELEC-AMLGX.arm-12.0-devel/build/u-boot-tools-2022.10'
scripts/kconfig/conf  --syncconfig Kconfig
  UPD     include/config.h
  CFG     u-boot.cfg
gcc: error: unrecognized command-line option ‘-m64’
gcc: error: unrecognized command-line option ‘-mno-mmx’
gcc: error: unrecognized command-line option ‘-mno-sse’; did you mean ‘-fno-dse’?
make[2]: *** [scripts/Makefile.autoconf:82: u-boot.cfg] Error 1
make[1]: *** No rule to make target 'include/config/auto.conf', needed by 'include/config/uboot.release'.  Stop.
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/build/build.LibreELEC-AMLGX.arm-12.0-devel/build/u-boot-tools-2022.10'
FAILURE: scripts/build u-boot-tools:host during make_host (package.mk)
*********** FAILED COMMAND ***********
make tools-only HOSTCC="${HOST_CC}" HOSTCFLAGS="-I${TOOLCHAIN}/include" HOSTLDFLAGS="${HOST_LDFLAGS}"
**************************************
FAILURE: scripts/build u-boot-tools:host has failed!
```

Kwiboo, 2023-06-18:
Looks like we target `qemu-x86_64_defconfig` instead of `tools-only_defconfig`, maybe a change of defconfig won't require qemu?

https://u-boot.readthedocs.io/en/latest/build/tools.html

chewitt, 2023-06-18:
The next failure you'll hit is `amlogic-boot-fip` which depends on several precompiled Amlogic (closed-source) signing binaries. It's possible to use [gxlimg](https://github.com/repk/gxlimg) for GXL and newer boards, but the tool doesn't support GXBB and alternative tools that do [meson-tools](https://github.com/afaerber/meson-tools) use different syntax (and aren't tested). I've also compounded the problem by adding precompiled x86_64 versions of [aml_chksum](https://github.com/LibreELEC/aml_chksum/) into the GXBB recipe although since we have source for that (and @Kwiboo is the original author) that might be an easier hoop to jump through.

heitbaum, 2023-06-24:
Log
```
Running /home/ubuntu/LibreELEC.tv/projects/Amlogic/bootloader/release
Installing u-boot for board bananapi-m2-pro...
CLEAN      amlogic-boot-fip
    *      Removing /home/ubuntu/LibreELEC.tv/build.LibreELEC-AMLGX.arm-12.0-devel/build/amlogic-boot-fip-e96b6a694380ff07d5a9e4be644ffe254bd18512 ...
    *      Removing /home/ubuntu/LibreELEC.tv/build.LibreELEC-AMLGX.arm-12.0-devel/qa_checks/amlogic-boot-fip-* ...
UNPACK      amlogic-boot-fip
CLEAN      u-boot
    *      Removing /home/ubuntu/LibreELEC.tv/build.LibreELEC-AMLGX.arm-12.0-devel/build/u-boot-2022.10 ...
    *      Removing /home/ubuntu/LibreELEC.tv/build.LibreELEC-AMLGX.arm-12.0-devel/install_pkg/u-boot-2022.10 ...
    *      Removing /home/ubuntu/LibreELEC.tv/build.LibreELEC-AMLGX.arm-12.0-devel/qa_checks/u-boot-* ...
UNPACK      u-boot
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0001-LOCAL-configs-meson64-prevent-stdout-stderr-on-video.patch
patching file include/configs/meson64.h
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0002-LOCAL-configs-meson64-remove-amlogic-so-fdtdir-finds.patch
patching file include/configs/meson64.h
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0003-LOCAL-board-amlogic-odroid-n2-remove-amlogic-prefix-.patch
patching file board/amlogic/odroid-n2/odroid-n2.c
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0004-BACKPORT-arm64-dts-meson-nanopi-k2-readd-PHY-reset-p.patch
patching file arch/arm/dts/meson-gxbb-nanopi-k2-u-boot.dtsi
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0005-WIP-ARM-dts-import-WeTek-Hub-Play2-DTs-from-Linux-5..patch
patching file arch/arm/dts/Makefile
patching file arch/arm/dts/meson-gxbb-wetek-hub-u-boot.dtsi
patching file arch/arm/dts/meson-gxbb-wetek-hub.dts
patching file arch/arm/dts/meson-gxbb-wetek-play2-u-boot.dtsi
patching file arch/arm/dts/meson-gxbb-wetek-play2.dts
patching file arch/arm/dts/meson-gxbb-wetek.dtsi
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0006-WIP-ARM-dts-use-snps-reset-on-WeTek-devices-to-fix-E.patch
patching file arch/arm/dts/meson-gxbb-wetek-hub-u-boot.dtsi
patching file arch/arm/dts/meson-gxbb-wetek-play2-u-boot.dtsi
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0007-WIP-boards-amlogic-add-board-files-for-wetek-gxbb-de.patch
patching file board/amlogic/wetek-gxbb/MAINTAINERS
patching file board/amlogic/wetek-gxbb/Makefile
patching file board/amlogic/wetek-gxbb/wetek-gxbb.c
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0008-WIP-boards-amlogic-add-WeTek-Hub-defconfig.patch
patching file configs/wetek-hub_defconfig
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0009-WIP-boards-amlogic-add-WeTek-Play2-defconfig.patch
patching file configs/wetek-play2_defconfig
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0010-WIP-ARM-dts-add-support-for-Radxa-Zero2.patch
patching file arch/arm/dts/Makefile
patching file arch/arm/dts/meson-g12b-radxa-zero2-u-boot.dtsi
patching file arch/arm/dts/meson-g12b-radxa-zero2.dts
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0011-WIP-boards-amlogic-add-Radxa-Zero2-defconfig.patch
patching file board/amlogic/w400/MAINTAINERS
patching file configs/radxa-zero2_defconfig
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0012-WIP-doc-boards-amlogic-update-for-Radxa-Zero2.patch
Checking patch /home/ubuntu/LibreELEC.tv/build.LibreELEC-AMLGX.arm-12.0-devel/build/u-boot-2022.10/board/amlogic/w400/MAINTAINERS...
Checking patch /home/ubuntu/LibreELEC.tv/build.LibreELEC-AMLGX.arm-12.0-devel/build/u-boot-2022.10/doc/board/amlogic/index.rst...
Checking patch /home/ubuntu/LibreELEC.tv/build.LibreELEC-AMLGX.arm-12.0-devel/build/u-boot-2022.10/doc/board/amlogic/radxa-zero.rst => /home/ubuntu/LibreELEC.tv/build.LibreELEC-AMLGX.arm-12.0-devel/build/u-boot-2022.10/doc/board/amlogic/radxa-zero2.rst...
Applied patch /home/ubuntu/LibreELEC.tv/build.LibreELEC-AMLGX.arm-12.0-devel/build/u-boot-2022.10/board/amlogic/w400/MAINTAINERS cleanly.
Applied patch /home/ubuntu/LibreELEC.tv/build.LibreELEC-AMLGX.arm-12.0-devel/build/u-boot-2022.10/doc/board/amlogic/index.rst cleanly.
Applied patch /home/ubuntu/LibreELEC.tv/build.LibreELEC-AMLGX.arm-12.0-devel/build/u-boot-2022.10/doc/board/amlogic/radxa-zero.rst => /home/ubuntu/LibreELEC.tv/build.LibreELEC-AMLGX.arm-12.0-devel/build/u-boot-2022.10/doc/board/amlogic/radxa-zero2.rst cleanly.
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0013-WIP-ARM-dts-import-Beelink-GT1-DT-from-Linux-5.21.patch
patching file arch/arm/dts/meson-gxm-gt1-ultimate-u-boot.dtsi
patching file arch/arm/dts/meson-gxm-gt1-ultimate.dts
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0014-WIP-boards-amlogic-add-Beelink-GT1-defconfig.patch
patching file configs/beelink-gt1_defconfig
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0015-WIP-doc-boards-amlogic-update-for-Beelink-GT1.patch
patching file board/amlogic/q200/MAINTAINERS
patching file doc/board/amlogic/beelink-gt1.rst
patching file doc/board/amlogic/index.rst
    APPLY PATCH (project)      projects/Amlogic/patches/u-boot/u-boot-0016-WIP-ARM-dts-convert-bananapi-m5-to-dtsi.patch
patching file arch/arm/dts/meson-sm1-bananapi-m5.dts
patching file arch/arm/dts/meson-sm1-bananapi.dtsi
[... cut ...]
make[1]: Leaving directory '/home/ubuntu/LibreELEC.tv/build.LibreELEC-AMLGX.arm-12.0-devel/build/amlogic-boot-fip-e96b6a694380ff07d5a9e4be644ffe254bd18512/bananapi-m2-pro'
FAILURE: scripts/build u-boot during makeinstall_target (package.mk)
*********** FAILED COMMAND ***********
. ${FOUND_PATH}
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/build "${1}" "${PARENT_PKG}"
**************************************
*********** FAILED COMMAND ***********
UBOOT_SYSTEM="${UBOOT_SYSTEM}" ${SCRIPTS}/install u-boot 2>&1
**************************************
make: *** [Makefile:10: image] Error 2
```

heitbaum, 2023-06-24:
Update
- With #7927 and the draft #7944 we are now progressing through the the AML binaries themselves

the Ampere A1 aarch64 (and the docker container) are happy with amd64 binary. Now just rework to use qemu for those binaries.
```
docker@82ff9cc70e3e:/home/ubuntu/LibreELEC.tv$ file build.LibreELEC-AMLGX.arm-12.0-devel/build/amlogic-boot-fip-e96b6a694380ff07d5a9e4be644ffe254bd18512/bananapi-m2-pro/aml_encrypt_g12a
aml_encrypt_g12a: ELF 64-bit LSB executable, x86-64, version 1 (GNU/Linux), statically linked, for GNU/Linux 2.6.32, BuildID[sha1]=1b0866af781ed83c46006483d63b174596e935e5, with debug_info, not stripped

docker@82ff9cc70e3e:/home/ubuntu/LibreELEC.tv$ ./build.LibreELEC-AMLGX.arm-12.0-devel/build/amlogic-boot-fip-e96b6a694380ff07d5a9e4be644ffe254bd18512/bananapi-m2-pro/aml_encrypt_g12a
bash: aml_encrypt_g12a: cannot execute binary file: Exec format error

docker@82ff9cc70e3e:/home/ubuntu/LibreELEC.tv$ qemu-x86_64 build.LibreELEC-AMLGX.arm-12.0-devel/build/amlogic-boot-fip-e96b6a694380ff07d5a9e4be644ffe254bd18512/bananapi-m2-pro/aml_encrypt_g12a

        AMLOGIC-G12A-G12B-SIG-module : Ver-0.12 Built on Nov 12 2019 10:35:05
…
```

heitbaum, 2023-06-24:
Final part of the fix - as below. I’ll update https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/tools/amlogic-boot-fip/package.mk once upstream patched is merged.
- https://github.com/LibreELEC/amlogic-boot-fip/pull/20

closed 2023-07-21 by CvH