LibreELEC/LibreELEC.tv #7958: [ENHANCEMENT]

thestraycat, 2023-06-30:
A mechanism to stop libreelec cleaning all the movies from the database when the network is down and a 'clean' database is performed.

Maybe a toggle box that allows cleaning of the database ONLY when the network is present as a fail safe to blatting the whole database!

chewitt, 2023-06-30:
Closing the ticket because:

a) This is a bug/issue tracker not a feature request tracker (requests go in the forum)
b) The logic for this belongs in Kodi not LE, ergo this would be the wrong project to log the request
c) If you have flakey wifi the solution is fixing the wifi not adding software complexity and workarounds

closed 2023-06-30 by chewitt

thestraycat, 2023-06-30:
Chewitt, 

a) Fair enough but i saw another other [ENHANCEMENT] on the issues page that hadnt been closed so took a punt myself. You might want to close that one as well?

b) Ok.

c) I disagree. With issues around RPI 4B on libreelec forgetting wifi credentials on 1 in 2 reboots for hoardes of people, "fixing the wifi" dosn't seem to be an option specially as each of those threads never resolves in a fix, you yourself have attempted to help people with these issues on the libreelec forums for these very issues. 

This issue has existed for a very long time as well. To conclude, i believe the cleaning operation should at least mention that no network connection is present and do you want to continue as a fail safe. 

It dosn't make any sense to clean the database of 1000's of movies just because there was no active network connection. there's loads of scenarios where this can simply just happen... Say you're playing with your network settings, moving from wifi to lan and get something wrong, have a network outage for some reason, It seems to me that this should be the default behaviour regardless.