LibreELEC/LibreELEC.tv #7984: [BUG] Gigabit ethernet is slow after kernel 6.4

smp79, 2023-07-15:
### Describe the bug

I noticed that high bitrate videos stutter over SMB with current LE master. iperf3 didn't show any speed issues and what's really odd is that if I try to play a high bitrate video right after running iperf3 - it plays with no stuttering.
I built a test image with current master and kernel 6.3.13 and I can't reproduce the issue.

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Play '....'
3. See error

### Informations
 - LE Version: master
 - Hardware Platform: Intel TGL, Realtek Ethernet (R8169 kernel driver)

### Log file

### Additional context

smp79, 2023-07-15:
https://forum.endeavouros.com/t/poor-download-speeds-over-ethernet-on-kernel-6-4-2/43143
https://bbs.archlinux.org/viewtopic.php?id=287172

Looks like it's an issue with r8169 driver.

closed 2023-07-22 by smp79

heitbaum, 2023-08-06:
> https://forum.endeavouros.com/t/poor-download-speeds-over-ethernet-on-kernel-6-4-2/43143 https://bbs.archlinux.org/viewtopic.php?id=287172
> 
> Looks like it's an issue with r8169 driver.
> 
> Upd: reverting [torvalds/linux@2ab19de](https://github.com/torvalds/linux/commit/2ab19de62d67e403105ba860971e5ff0d511ad15) fixed it.
> 
> https://lore.kernel.org/netdev/ZJRXYtfY4jFi934A@debian.me/T/

Did this get fixed in upstream?

jernejsk, 2023-08-06:
Seems so: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v6.4.8&id=cf2ffdea08393

smp79, 2023-08-06:
Fixed in 6.4.8 / 6.5-rc3