LibreELEC/LibreELEC.tv #7986: [BUG] No sound via HDA Nvidia , HDMI

alexhass, 2023-07-16:
### Describe the bug
Generic Lagacy build have no Audio via HDMI. This is a serious problem.

### To Reproduce
Steps to reproduce the behavior:
1. install legacy on zotac ion
2. connect to tv and run tv (e.g. tvheadend)
3. no sound

### Informations
 - LE Version: tested up to nightly 15.07.2023
 - Hardware Platform: Zotac ION

### Log file

### Additional context

closed 2023-09-04 by heitbaum

alexhass, 2023-09-04:
The bug still exists in 11.0.3. 

The forum has tons of comments about the same defect.

heitbaum, 2023-09-04:
Unfortunately without much to go on in this issue unable to address this.

Nvidia is not a recommended video card as you may be aware. https://wiki.libreelec.tv/hardware/intel-x86-64-generic

Raising a bug here won’t get your issue fixed as the development team do not develop or have nvidia to test with so the forum is the right place for support - hopefully another user has experience with this issue -  if a solution can be found there - then would be able to address this in development. 

Your issue seems to be isolated to an audio issue - so suggest your forum post identifies this as the issue. 

There have been recent updates to the nvidia packages in LE - and will continue - but are only for the modern cards.

alexhass, 2023-12-04:
There are many others in forums reporting the same issue. Looks like the new driver is not working properly. In LE9.3 ist works. Have no read about any workaround or other installed driver that works yet.