LibreELEC/LibreELEC.tv #7987: LE 12 Update via GUI generates a crash (-log)

eldersixpack13, 2023-07-16:
### Describe the bug
LE12 crashes or better a crashlog (usually invisible for default user) is generated during LE12 update via GUI

### To Reproduce
Steps to reproduce the behavior:
1. trigger a update via LibreELEC setting => update
2. let the update finish
3. see .kodi/temp/

### Informations
 - LE Version: Generic.x86_64-nightly-20230714-788de87
 - Hardware Platform: Intel NUC13

### Log file
[kodi_crashlog_20230716082421.log](https://github.com/LibreELEC/LibreELEC.tv/files/12064196/kodi_crashlog_20230716082421.log)

### Additional context
 don't know since when exactly this crashes happens (have seen this in the past moons ago, though).
I usually do update via a command line script:
https://gist.github.com/eldersixpack13/5a4de7e2b6043cf950856b330287eec8

antonlacon, 2023-09-04:
DNS issues?

> 2023-09-04 14:23:35.913 T:1147    error <general>: SETTINGS: load_url # URLError(gaierror(-3, 'Temporary failure in name resolution'))

eldersixpack13, 2023-09-05:
don't know.
I see the DNS error since moon (LE 11 or 10).
a bug report (?) should be findable in the LE Forum (I do not have a account there anymore)

nuc:~ # nslookup fritz.box
Server:    192.168.100.1
Address 1: 192.168.100.1 fritz.box

Name:      fritz.box
Address 1: 2001:9e8:263:ee00:9a9b:cbff:feab:2a40 fritz.box
Address 2: 192.168.100.1 fritz.box
Address 3: fd00::9a9b:cbff:feab:2a40 fritz.box

nuc:~ # nslookup nuc.fritz.box
Server:    192.168.100.1
Address 1: 192.168.100.1 fritz.box

Name:      nuc.fritz.box
Address 1: 192.168.100.22 nuc.fritz.box

antonlacon, 2023-11-09:
The error log when the network isn't up should be significantly shorter with the next LE12 nightly. Please reopen if a nightly from November 10/11 or later still has issues (with a new log).

Thanks for reporting.

closed 2023-11-09 by antonlacon