LibreELEC/LibreELEC.tv #7990: [BUG] CEC initialization failed RPI4

DerGary, 2023-07-16:
### Describe the bug
I installed a clean new version of libreelec. CEC was working at the beginning but after a few restarts it is now broken. Everytime Kodi starts it shows the message "The CEC Adapter could not be initialized. Please check your settings". Therefore I can not use my remote control from the TV to control Kodi.

### To Reproduce
Steps to reproduce the behavior:
I don't really know how to reproduce it. I installed Libreelec 2 times and configured everything as I would and after some restarts on both installation CEC broke down and was not able to work again even after trying different cables and so on.
On both installations I additionally installed Hyperion.ng and the Arctic Horizon 2 Skin. But I don't think that it has to do with that. As this didn't immediately break CEC.

It immediately broke after I put the SD Card into my windows machine and made a backup with Win32DiskImager, but I don't know if that has something to do with it, as I think it should not.

### Informations
 - LE Version: 11.0.1
 - Hardware Platform: RPI4

### Log file
http://ix.io/4AJJ

### Additional context
cec-ctl device list:

```
LibreELEC:~ # cec-ctl --list-devices
vc4_hdmi (vc4-hdmi-0):
        /dev/cec0

vc4_hdmi (vc4-hdmi-1):
        /dev/cec1

Driver Info:
        Driver Name                : vc4_hdmi
        Adapter Name               : vc4-hdmi-0
        Capabilities               : 0x0000010e
                Logical Addresses
                Transmit
                Passthrough
                Connector Info
        Driver version             : 6.1.19
        Available Logical Addresses: 1
        DRM Connector Info         : card 0, connector 32
        Physical Address           : 4.1.0.0
        Logical Address Mask       : 0x0002
        CEC Version                : 1.4
        Vendor ID                  : 0x001582 (Pulse-Eight)
        OSD Name                   : ''
        Logical Addresses          : 1 (Allow Fallback to Unregistered)

          Logical Address          : 1 (Recording Device 1)
            Primary Device Type    : Record
            Logical Address Type   : Record
            All Device Types       : Record
            RC TV Profile          : None
            Device Features        :
                None
```

When I use cec-ctl -m to monitor the cec stream than I can watch the keys pressed on my remote to be registered by linux. Altough kodi won't register them

```
LibreELEC:~ # cec-ctl -m
Driver Info:
        Driver Name                : vc4_hdmi
        Adapter Name               : vc4-hdmi-0
        Capabilities               : 0x0000010e
                Logical Addresses
                Transmit
                Passthrough
                Connector Info
        Driver version             : 6.1.19
        Available Logical Addresses: 1
        DRM Connector Info         : card 0, connector 32
[... 146 lines cut ...]
DEBUG:   [            5041]     CLinuxCECAdapterCommunication::Close - m_fd=-1
DEBUG:   [            6041]     CLinuxCECAdapterCommunication::Open - m_fd=3 bStartListening=1
ERROR:   [            6041]     CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
DEBUG:   [            6041]     CLinuxCECAdapterCommunication::Close - m_fd=3
ERROR:   [            6041]     could not open a connection (try 7)
DEBUG:   [            6041]     CLinuxCECAdapterCommunication::Close - m_fd=-1
DEBUG:   [            7041]     CLinuxCECAdapterCommunication::Open - m_fd=3 bStartListening=1
ERROR:   [            7041]     CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
DEBUG:   [            7041]     CLinuxCECAdapterCommunication::Close - m_fd=3
ERROR:   [            7041]     could not open a connection (try 8)
DEBUG:   [            7041]     CLinuxCECAdapterCommunication::Close - m_fd=-1
DEBUG:   [            8042]     CLinuxCECAdapterCommunication::Open - m_fd=3 bStartListening=1
ERROR:   [            8042]     CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
DEBUG:   [            8042]     CLinuxCECAdapterCommunication::Close - m_fd=3
ERROR:   [            8042]     could not open a connection (try 9)
DEBUG:   [            8042]     CLinuxCECAdapterCommunication::Close - m_fd=-1
DEBUG:   [            9042]     CLinuxCECAdapterCommunication::Open - m_fd=3 bStartListening=1
ERROR:   [            9042]     CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
DEBUG:   [            9042]     CLinuxCECAdapterCommunication::Close - m_fd=3
ERROR:   [            9042]     could not open a connection (try 10)
DEBUG:   [            9042]     CLinuxCECAdapterCommunication::Close - m_fd=-1
unable to open the device on port Linux
NOTICE:  [           10042]     connection opened
ERROR:   [           10042]     could not start CEC communications
```

DerGary, 2023-08-04:
nothing?

heitbaum, 2023-09-04:
Many are / have been on summer holidays, Please use the forums for support. Please try with the latest LE11 nightly image. There have been updates since LE11.0.1. Issue can be reopened if issue identified at the forums. Thanks.

closed 2023-09-04 by heitbaum