LibreELEC/LibreELEC.tv #8001: PVR issue

numbernumero, 2023-07-21:
In all versions of LibreELEC 11 even in the latest 11.0.3, PVR Simple Client does not work perfectly. When you start a channel from the list in the first few seconds there are various flickers. This happens especially as soon as the PVR is opened. I also tried with Zatto Pvr Client and the problem is the same, in the first seconds when you start a channel freezes and flickers.
I did other tests in Kodi 20.2 in both Windows 11 and Android and on these operating systems everything works normally.
I also tried Libreelec 12 Nightly (20230716) and in this case there are no flickers

chewitt, 2023-07-21:
I'm closing the issue report because it contains no information/data of use for an investigation, and because at this point in its lifecycle LE11 is a dead codebase; there are no plans for another release). If LE12 nightlies solve your problem, use them.

closed 2023-07-21 by chewitt

numbernumero, 2023-07-21:
> I'm closing the issue report because it contains no information/data of use for an investigation, and because at this point in its lifecycle LE11 is a dead codebase; there are no plans for another release). If LE12 nightlies solve your problem, use them.

Unfortunately LibreElec 12 often crashes and starts again on its own

chewitt, 2023-07-21:
There are no plans for further Kodi releases; ergo there are no plans for further LE releases. Please take further Q&A to the forums as this tracker is for confirmed/diagnosed problems.