LibreELEC/LibreELEC.tv #8009: [Help Required] qemu 8.1-rc1 requires Python venv and pip

heitbaum, 2023-07-27:
### Describe the bug

Build of qemu is failing
- Ref #5538

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Play '....'
3. See error

### Informations
 - LE Version: [e.g. 9.2.1] 12
 - Hardware Platform: [e.g. RPi3] all

### Log file

### Additional context

- https://gitlab.com/qemu-project/qemu/-/issues/1792

vpeter4, 2023-07-30:
What exactly is the problem here and what need to get fixed/investigated?

heitbaum, 2023-07-30:
I created a draft pr - it does seem to cross compile the .so files of simplejson and Pillow as aarch64 when using a target of A64. To experience the issue first hand - cherry pick https://github.com/LibreELEC/LibreELEC.tv/pull/8016/commits/f9eb28c0380f8101ee75530ec14fb5c96a2b6976 and try scripts/build qemu:host

the changes to the cross-compile and setuptools/pip may/probably have other unforeseen items to be worked through.

I tried a number of different ways to create a “bootstrap” pyvenv including ensurepip - with limited/no/inconsistent results - and configured qemu to use this…. This draft pr at least is producing consistent results when building.

If you had a better way - that would be great. Otherwise if this is the right track … would be good to understand why the 85a018611be9a28e87ff5af2e6fd4d4fdbcd50c5 workaround is currently necessary.

there is also the #7361 to be addressed

heitbaum, 2023-07-30:
Output of .so files (build host is x86_64)
```
build.LibreELEC-A64.aarch64-12.0-devel/install_pkg$ file */usr/lib/python3.11/site-packages/*/*/*.so */usr/lib/python3.11/site-packages/*/*.so 
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_ARC4.abi3.so:            ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_chacha20.abi3.so:        ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_pkcs1_decode.abi3.so:    ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_raw_aes.abi3.so:         ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_raw_arc2.abi3.so:        ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_raw_blowfish.abi3.so:    ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_raw_cast.abi3.so:        ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_raw_cbc.abi3.so:         ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_raw_cfb.abi3.so:         ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_raw_ctr.abi3.so:         ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_raw_des3.abi3.so:        ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_raw_des.abi3.so:         ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_raw_ecb.abi3.so:         ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_raw_eksblowfish.abi3.so: ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_raw_ocb.abi3.so:         ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_raw_ofb.abi3.so:         ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Cipher/_Salsa20.abi3.so:         ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Hash/_BLAKE2b.abi3.so:           ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Hash/_BLAKE2s.abi3.so:           ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Hash/_ghash_portable.abi3.so:    ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Hash/_keccak.abi3.so:            ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Hash/_MD2.abi3.so:               ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Hash/_MD4.abi3.so:               ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Hash/_MD5.abi3.so:               ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Hash/_poly1305.abi3.so:          ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Hash/_RIPEMD160.abi3.so:         ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Hash/_SHA1.abi3.so:              ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Hash/_SHA224.abi3.so:            ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
pycryptodome-3.18.0/usr/lib/python3.11/site-packages/Crypto/Hash/_SHA256.abi3.so:            ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
[... cut ...]
Pillow-10.0.0/usr/lib/python3.11/site-packages/PIL/_imaging.so:                              ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
Pillow-10.0.0/usr/lib/python3.11/site-packages/PIL/_imagingtk.so:                            ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
simplejson-3.19.1/usr/lib/python3.11/site-packages/simplejson/_speedups.so:                  ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, stripped
```

vpeter4, 2023-07-30:
For those 3 python calls in a row: if you remove patch file https://github.com/LibreELEC/LibreELEC.tv/blob/aec466da8aaf5f79b8f18b8474d0a6e6fbaf5127/packages/lang/Python3/patches/Python3-0300-generate-legacy-pyc-bytecode.patch then python:host builds with just one call.
Tip comes from here: https://bugs.python.org/issue42446

heitbaum, 2023-07-30:
> For those 3 python calls in a row: if you remove patch file https://github.com/LibreELEC/LibreELEC.tv/blob/aec466da8aaf5f79b8f18b8474d0a6e6fbaf5127/packages/lang/Python3/patches/Python3-0300-generate-legacy-pyc-bytecode.patch then python:host builds with just one call. Tip comes from here: https://bugs.python.org/issue42446 Not sure if this is correct solution or not. Maybe patch is needed only for target?

d8ddcfad154d453dca2c30554b49c4904e9a47ae does indeed allow the successful compile. The does it work right is still the ultimate question :-)

vpeter4, 2023-07-31:
> The does it work right is still the ultimate question :-)

Looks like not because at least my Generic legacy image doesn't really work with pr 8016.

heitbaum, 2023-08-01:
> > The does it work right is still the ultimate question :-)
> 
> Looks like not because at least my Generic legacy image doesn't really work with pr 8016.

Backing out the move of the bytecode patch to target only and reintroduction of ensurepip * 3 returns my LE back to a working state. 

Looks like we need to fix the builds of the target pyc (by the host Python3) to match the required bytecode in target + drop the patch in the host build - https://peps.python.org/pep-3147/#case-4-legacy-pyc-files-and-source-less-imports

heitbaum, 2023-08-01:
Putting a ENV controlled use of the patch seems to have addressed this - 8d9376694898b699a2bb34b8cca54291399c49c3 - I’m now using a cleanly built LE with the updated 8016 on Generic nuc12. 

- should the `packages/python/devel` builds use `export DONT_BUILD_LEGACY_PYC=1`?
- should `export DONT_BUILD_LEGACY_PYC=1` be the default for all :host builds?
- using the modified patch (and environment variable)  returns Python3 back its standard pyc behaviour - which is probably more correct when not targeting the embedded device
- run logs look good

heitbaum, 2023-08-27:
Pr complete. Closing

heitbaum, 2023-09-04:
- PR was reverted #8102

closed 2024-09-03 by jernejsk