LibreELEC/LibreELEC.tv #8010: mc-bin: error while loading shared libraries: libssh2.so.1

eldersixpack13, 2023-07-27:
### Describe the bug
call of mc on an command line gives: 

mc-bin: error while loading shared libraries: libssh2.so.1: cannot open shared object file: No such file or directory

### To Reproduce
Steps to reproduce the behavior:
1. ssh into LE
2. run mc

### Informations
 - LE Version: nightly-20230727-b3ad101
 - Hardware Platform: NUC13

### Log file
nothing special in the logs

### Additional context
tried with an previous nightly doesn't fixed it.
currently I can't tell for sure when this bug went in, but I mean I got mc flawless running on at least
nightly-20230714-788de87

heitbaum, 2023-07-30:
Issue came about with change in libssh2 - https://github.com/libssh2/libssh2/commit/896154bc17f000c0a1bb89b74bc879692ac0d47c fix is 912f953b410cd9cd6f53a5d3955a62f2c191ae33 - will include this in the next system tools release.

eldersixpack13, 2023-07-30:
Okay, thanks
will close this bug then

eldersixpack13, 2023-07-31:
should it have been "System Tools" version 11.80.2.0 ? 
updated today

if so:  the bug still exists

heitbaum, 2023-08-01:
> should it have been "System Tools" version 11.80.2.0 (new update from today) ?
> 
> if so: the bug still exists
> 
> find / -iname '*libssh2*'
> 
> returns nothing

Fix will be in
- #8020
- #8019

the file shouldn’t be there. When compiled statically it will be included in the mc-bin file

eldersixpack13, 2023-08-07:
Houston, the eagle has landed
;-)

closed 2023-08-07 by eldersixpack13