LibreELEC/LibreELEC.tv #8013: [BUG] Kodi doesn't show up on AMD Generic with LE 11.0.3 (Regression)

Aztorius, 2023-07-29:
### Describe the bug
When using an AMD CPU with integrated GPU there is a major regression from LE Generic 11.0.1 to 11.0.3 => LE is starting and print version at the top left corner but Kodi doesn't show up anymore. There is no crash log, and no error in the kodi.log file nor dmesg/journalctl. Despite Kodi not showing up, logs suggest it is running in the background. Downgrade to 11.0.1 fixes the issue => regression.

### To Reproduce
Steps to reproduce the behavior:
1. Update from 11.0.1 to 11.0.3 on an AMD Generic system (Ryzen APU or maybe AMD dGPU)
2. System is updating and rebooting automatically and only print the LibreElec version at the top left corner of the screen, no Kodi interface

### Informations
 - LE Version: 11.0.3
 - Hardware Platform: AMD Ryzen 3 4300GE with Radeon Graphics

### Log file
- LE 11.0.1 (no issue) : http://ix.io/4BLG
- LE 11.0.3 (regression) : http://ix.io/4BLK
- dmesg diff between 11.0.1 and 11.0.3 : http://ix.io/4BLI

### Additional context
N/A

jpsutton, 2023-07-29:
Confirming this issue on Ryzen 5 PRO 2400GE with integrated Vega graphics. Downgrading to 11.0.1 resolved the issue (placing tar file in `.updates` folder and rebooting).

Fietspomp86, 2023-07-29:
Same issue here, with a AMD Ryzen 9 5900X.
GPU is a AMD 6900XT.

Installed legacy version, [nightly-20230728, that works fine.

Will also wait for the kernel bump.

heitbaum, 2023-07-30:
LE11 Generic-legacy there now - with kernel 6.1.42 - https://test.libreelec.tv/11.0/Generic/Generic-legacy/LibreELEC-Generic-legacy.x86_64-11.0-nightly-20230730-445ef36.img.gz

LE11 Generic will be there in the next two hours or so.

Fietspomp86, 2023-07-30:
Just installed LE11 generic nightly, LibreELEC-Generic.x86_64-11.0-nightly-20230730-445ef36.img.gz.

Still same issue unfortunately, keeps hanging with text in top left corner.

Installed generic-legacy 20230730, that works fine.

Any logs I can provide to assist?

Aztorius, 2023-07-30:
Oops sorry.

Just tested LibreELEC-Generic.x86_64-11.0-nightly-20230730-445ef36.img.gz and it is not working either.

CvH, 2023-07-30:
could you go back version wise and have a look what is the last version that still works ?
https://test.libreelec.tv/11.0/Generic/Generic/

Aztorius, 2023-07-30:
After some testing : 
- LibreELEC-Generic.x86_64-11.0-nightly-20230630-880951d.img is the last working version
- LibreELEC-Generic.x86_64-11.0-nightly-20230701-7863fdf.img is the first non working version

Aztorius, 2023-08-01:
It took me days but after compiling LE with many kernels this issue appear with kernel 6.1.36. It is working with 6.1.35, so some patch in 6.1.36 is responsible for the issue.
I may try to bisect this down to the responsible patch if necessary (and if there is a way to build LibreELEC with an un-released/un-tagged kernel version commit by commit).

heitbaum, 2023-08-01:
It looks like there are only a couple of potential candidates in the list -, https://lore.kernel.org/lkml/2023062853-viper-overtone-bfea@gregkh/

perhaps a quick revert on these. Otherwise use 6.1.35 as the base and add the bisect patch against 6.1.35 to the patches directory. 

You can build against a commit - but it does mean downloading 200MB each time - see the Amlogic linux in the package.mk

Aztorius, 2023-08-04:
Found it :
```
8d855bc67630fa2b17855d85de61b9cd4300e3ad is the first bad commit
commit 8d855bc67630fa2b17855d85de61b9cd4300e3ad
Author: Rodrigo Siqueira <Rodrigo.Siqueira@amd.com>
Date:   Thu Oct 6 16:40:55 2022 -0400

    drm/amd/display: Use dc_update_planes_and_stream
    
    [ Upstream commit f7511289821ffccc07579406d6ab520aa11049f5 ]
    
    [Why & How]
    The old dc_commit_updates_for_stream lacks manipulation for many corner
    cases where the DC feature requires special attention; as a result, it
    starts to show its limitation (e.g., the SubVP feature is not supported
    by it, among other cases). To modernize and unify our internal API, this
    commit replaces the old dc_commit_updates_for_stream with
    dc_update_planes_and_stream, which has more features.
    
    Reviewed-by: Harry Wentland <Harry.Wentland@amd.com>
    Acked-by: Qingqing Zhuo <qingqing.zhuo@amd.com>
    Signed-off-by: Rodrigo Siqueira <Rodrigo.Siqueira@amd.com>
    Tested-by: Daniel Wheeler <daniel.wheeler@amd.com>
    Signed-off-by: Alex Deucher <alexander.deucher@amd.com>
    Stable-dep-of: ea2062dd1f03 ("drm/amd/display: fix the system hang while disable PSR")
    Signed-off-by: Sasha Levin <sashal@kernel.org>

 drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm.c | 30 +++++++++++------------
 1 file changed, 15 insertions(+), 15 deletions(-)
```

CvH, 2023-08-04:
@Aztorius  Sadly that commit is also at the latest kernel so it is likely broken at the latest kernel too. We have no hardware to reproduce the problem so it is not really feasible that we report that kind of problems.
I asked for the correct procedure to report such a problem
```
you should mail author, committer and dri-devel mailing list at minimum
you can dig more relevant emails with get_maintainer.pl script in linux kernel source running over file in question
```
Would be nice if you report the problem 👍

Aztorius, 2023-08-04:
Just tested LibreELEC-Generic.x86_64-12.0-nightly-20230804-e6fe1ad
Kodi is showing up, although with huge artefacts at each new frame
Maybe updating some package like Mesa may help (22.3.7 -> 23.1.5) ? I will try this. It could be some incompatibility between an old mesa and the kernel following this change.

heitbaum, 2023-08-05:
> Just tested LibreELEC-Generic.x86_64-12.0-nightly-20230804-e6fe1ad Kodi is showing up, although with huge artefacts at each new frame Maybe updating some package like Mesa may help (22.3.7 -> [...]

LE12 base packages are shaping up - there are some big changes with glibc/binutils (pr waiting), the dev tools (medium change), these will form the base. Mesa 23.2 around the corner (but will be 23.3 by the time le12 comes out) LLVM 17 will be the compiler - currently testing LLVM-RCs. I have been running kernel 6.5rc and with rc5 all the intel glitches seem sorted out. There has been a lot of change in the graphics areas in both intel and amd recently. kernel 6.6 will likely be the LE12 kernel.

To test your LE11 build with latest mesa - `git diff master libreelec-11.0 packages/graphics/mesa/` there are no real hoops to go through. For the kernel same should be the case - just go with latest 6.4.8 or 6.5rc5 (sundays release)

Aztorius, 2023-08-06:
After some testing : 
- this bug is only a kernel issue
- this bug appeared on Linux 6.1.36 with commit 8d855bc67630fa2b17855d85de61b9cd4300e3ad
- this bug also affects Linux 6.3.13
- this bug does not affect Linux 6.4. It seems like they have backported the culprit commit to 6.3 and 6.1 but it may be missing something and causes Kodi to not show up with GBM.

This issue seems limited to LE11 only since LE12 is using Kernel 6.4+.
I will open an issue for the kernel team to fix this for 6.1.

CvH, 2023-08-06:
> this bug does not affect Linux 6.4. It seems like they have backported the culprit commit to 6.3 and 6.1
https://github.com/torvalds/linux/commit/f7511289821ffccc07579406d6ab520aa11049f5

looks like the commit is at 6.4 and 6.5 too, maybe another commit fixed it and that one was not backported

Aztorius, 2023-08-07:
This linux commit fixes the issue : bb46a6a9bab1 ("drm/amd/display: Ensure that planes are in the same order")
Maybe you could try applying it right now until it is effectively backported ?

See the drm/amd GitLab ticket about this issue here : https://gitlab.freedesktop.org/drm/amd/-/issues/2766

chewitt, 2023-08-08:
@Aztorius thanks for the confirmation!

Aztorius, 2023-08-09:
The fix should be included inside Kernel 6.1.45

See https://lore.kernel.org/stable/50a5705d-dbcb-4db4-a210-c55dabe5d7a0@amd.com/T/#u

heitbaum, 2023-08-10:
> The fix should be included inside Kernel 6.1.45
> 
> See https://lore.kernel.org/stable/50a5705d-dbcb-4db4-a210-c55dabe5d7a0@amd.com/T/#u

included in rc1 - https://lore.kernel.org/lkml/20230809103636.615294317@linuxfoundation.org/

heitbaum, 2023-08-11:
> The fix should be included inside Kernel 6.1.45
> 
> See https://lore.kernel.org/stable/50a5705d-dbcb-4db4-a210-c55dabe5d7a0@amd.com/T/#u

Pr raised

Fietspomp86, 2023-08-16:
> @Aztorius https://test.libreelec.tv/11.0/Generic/Generic/LibreELEC-Generic.x86_64-11.0-nightly-20230816-1d807c8.img.gz has 6.1.45 kernel included.

Thanks! Just tested it, on my HTPC with AMD 6900XT GPU and AMD Ryzen 9 5900X processor.
It works!

PR #8014 (2023-07-30): [le11] linux: update to 6.1.y
- https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.1.39
- https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.1.40
- https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.1.41
- https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.1.42

6.1.39 - 589 patches
6.1.40 - 223 patches
6.1.41 - 3 patches (AMD zenbleed fix)
6.1.42 - 183 patches

### errors / fixes / issues / regressions
- potentially fixes #8013

### log 
- https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/log/?h=linux-6.1.y
- https://git.kernel.org/pub/scm/linux/kernel/g [... cut]
PR #8041 (2023-08-15): linux: update to 6.1.45
- Fixes #8013
closed 2023-08-16 by heitbaum