LibreELEC/LibreELEC.tv #8025: [BUG] File descriptor leak in ffmpeg-001-v4l2-request.patch

cgutman, 2023-08-02:
### Describe the bug
While debugging an issue with the v4l2-requests code that I imported into my FFmpeg build from your patch [here](https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/multimedia/ffmpeg/patches/v4l2-request/ffmpeg-001-v4l2-request.patch), I identified a bug in a failure path that was causing `/dev/media` fds to leak over time. My code happened to be particularly sensitive to this leak, but it will impact LibreELEC as well to a lesser extent.

I would have this bug it on @jernejsk 's repo, but it doesn't have an issue tracker enabled. Since they're a LibreELEC team member, hopefully they will see it here.

### To Reproduce
The specific issue is the codepath here: https://github.com/LibreELEC/LibreELEC.tv/blob/5c23848068e1f7866e86eda319430fd717d9ebf5/packages/multimedia/ffmpeg/patches/v4l2-request/ffmpeg-001-v4l2-request.patch#L882-L883

That code is intended to be reached when `v4l2_request_probe_video_device()` is successful and the `break` is taken. However, it's also executed when the probe fails after trying all relevant `/dev/video` devices. This can happen in normal situations because there will often be multiple `/dev/media` devnodes for different codecs that won't be the ones you're looking for. In the latter case, we need to `goto fail` instead so we can `close(ctx->media_fd)`.

### Informations
 - Latest commit as of 5c23848068e1f7866e86eda319430fd717d9ebf5

### Log file

### Additional context

jernejsk, 2023-08-05:
Since you already sorted out issue, can you provide a patch? I can then include it in my branch.

heitbaum, 2023-09-06:
@cgutman - following up.

cgutman, 2023-09-06:
Apologies, I somehow missed @jernejsk's message. I have a patch [here](https://github.com/cgutman/FFmpeg/commit/7000b5396c723d6d962b08278c36b50984ab13a4), but it will probably not apply directly to your code since I have made changes to remove the udev dependency for v4l2-requests in my branch too.

jernejsk, 2023-09-06:
If you don't mind, I'll fold patch into "Add common V4L2 request API code" commit after some time.

cgutman, 2023-09-06:
Yeah, that's fine with me.

You may also want to pick up https://github.com/cgutman/FFmpeg/commit/86d5b7f665ab5f15698c596ec3c9e5c83521c6a6 if you're updating the v4l2-request patch. Feel free to fold that trivial change into the patch set however you'd like.

zhangn1985, 2023-09-07:
@jernejsk any upstream progress for request API?

jernejsk, 2023-09-07:
Not that I know.

PR #8116 (2023-09-09): ffmpeg: update v4l2-request patch
Fix memory leak during device enumeration. Thanks to @cgutman.

Closes #8025
closed 2023-09-09 by CvH