LibreELEC/LibreELEC.tv #8040: [BUG] OK Button on Remote not working

DDQW, 2023-08-11:
### Describe the bug
Chromecast Remote OK Button is not working. Every other Button is. Remote is known good cause it works fine when paired with chromecast or kodi on fedora

### To Reproduce
Steps to reproduce the behavior:
1. Pair a chromecast remote
2. try to activate something with the ok button
3. nothing works

### Informations
 - LE Version: 11.0.3
 - Hardware Platform: x86_64 - Intel N200

### Log file

### Additional context

chewitt, 2023-08-11:
Share the output of "cat /proc/bus/input/devices" .. the keycode needs mapping.

DDQW, 2023-08-11:
`
I: Bus=0019 Vendor=0000 Product=0003 Version=0000
N: Name="Sleep Button"
P: Phys=PNP0C0E/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0E:00/input/input0
U: Uniq=
H: Handlers=kbd event0 
B: PROP=0
B: EV=3
B: KEY=4000 0 0

I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=PNP0C0C/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input1
U: Uniq=
H: Handlers=kbd event1 
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=LNXPWRBN/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXPWRBN:00/input/input2
U: Uniq=
H: Handlers=kbd event2 
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0019 Vendor=0000 Product=0006 Version=0000
N: Name="Video Bus"
P: Phys=LNXVIDEO/video/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input3
U: Uniq=
H: Handlers=kbd event3 
B: PROP=0
B: EV=3
B: KEY=3e000b00000000 0 0 0

I: Bus=0003 Vendor=046a Product=b090 Version=0111
N: Name="Cherry USB keyboard"
P: Phys=usb-0000:00:14.0-1/input0
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:046A:B090.0001/input/input4
U: Uniq=
H: Handlers=sysrq kbd leds event4 
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff9f207ac14057ff febeffdfffefffff fffffffffffffffe
B: MSC=10
B: LED=7

I: Bus=0003 Vendor=046a Product=b090 Version=0111
N: Name="Cherry USB keyboard System Control"
P: Phys=usb-0000:00:14.0-1/input1
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:046A:B090.0002/input/input5
U: Uniq=
H: Handlers=kbd event5 
B: PROP=0
B: EV=13
B: KEY=c000 10000000000000 0
B: MSC=10

I: Bus=0003 Vendor=046a Product=b090 Version=0111
N: Name="Cherry USB keyboard Consumer Control"
P: Phys=usb-0000:00:14.0-1/input1
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:046A:B090.0002/input/input6
U: Uniq=
H: Handlers=kbd event6 
B: PROP=0
B: EV=13
B: KEY=2a000000000 0 0 1000002000000 3878d8011001 e000000000000 0
B: MSC=10

I: Bus=0003 Vendor=041e Product=3264 Version=0110
N: Name="Creative Technology Ltd Sound Blaster X3"
P: Phys=usb-0000:00:14.0-7/input0
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:041E:3264.0003/input/input7
U: Uniq=525A958A86523F95
H: Handlers=kbd event7 
B: PROP=0
B: EV=13
B: KEY=40 0 0 0 0 0 18000 17800000000 8e000000000000 0
B: MSC=10

I: Bus=0019 Vendor=1283 Product=0000 Version=0000
N: Name="ITE8708 CIR transceiver"
P: Phys=
S: Sysfs=/devices/pnp0/00:02/rc/rc0/input8
U: Uniq=
H: Handlers=kbd event8 
B: PROP=20
B: EV=100017
B: KEY=fff 0 4200108fc32e 237605100000000 0 700158000 419200004001 8e968000000000 10000ffc
B: REL=3
B: MSC=10

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HDA Intel PCH Mic"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:1f.3/sound/card0/input9
U: Uniq=
H: Handlers=event9 
B: PROP=0
B: EV=21
B: SW=10

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HDA Intel PCH Line Out"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:1f.3/sound/card0/input10
U: Uniq=
H: Handlers=event10 
B: PROP=0
B: EV=21
B: SW=40

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HDA Intel PCH Headphone"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:1f.3/sound/card0/input11
U: Uniq=
H: Handlers=event11 
B: PROP=0
B: EV=21
B: SW=4

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HDA Intel PCH HDMI/DP,pcm=3"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:1f.3/sound/card0/input12
U: Uniq=
H: Handlers=event12 
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HDA Intel PCH HDMI/DP,pcm=7"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:1f.3/sound/card0/input13
U: Uniq=
H: Handlers=event13 
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HDA Intel PCH HDMI/DP,pcm=8"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:1f.3/sound/card0/input14
U: Uniq=
H: Handlers=event14 
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HDA Intel PCH HDMI/DP,pcm=9"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:1f.3/sound/card0/input15
U: Uniq=
H: Handlers=event15 
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0019 Vendor=1283 Product=0000 Version=0000
N: Name="eventlircd"
P: Phys=/dev/input/event8
S: Sysfs=/devices/virtual/input/input16
U: Uniq=
H: Handlers=event16 
B: PROP=0
B: EV=5
B: REL=3

`

Btw if a developer reads this i would happily provide you with the money for a chromecast remote and some treats to make your time fixing this more comfortable, just contact me privately.

chewitt, 2023-08-11:
Ahh, so it's an IR device? .. I was assuming BT.

DDQW, 2023-08-11:
No it is BT the Chromecast Remote has no IR afaik

DDQW, 2023-08-11:
```
I: Bus=0019 Vendor=0000 Product=0003 Version=0000
N: Name="Sleep Button"
P: Phys=PNP0C0E/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0E:00/input/input0
U: Uniq=
H: Handlers=kbd event0 
B: PROP=0
B: EV=3
B: KEY=4000 0 0

I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=PNP0C0C/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input1
U: Uniq=
H: Handlers=kbd event1 
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=LNXPWRBN/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXPWRBN:00/input/input2
U: Uniq=
H: Handlers=kbd event2 
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0019 Vendor=0000 Product=0006 Version=0000
N: Name="Video Bus"
P: Phys=LNXVIDEO/video/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input3
U: Uniq=
H: Handlers=kbd event3 
B: PROP=0
B: EV=3
B: KEY=3e000b00000000 0 0 0

I: Bus=0003 Vendor=046a Product=b090 Version=0111
N: Name="Cherry USB keyboard"
P: Phys=usb-0000:00:14.0-1/input0
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:046A:B090.0001/input/input4
U: Uniq=
H: Handlers=sysrq kbd leds event4 
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff9f207ac14057ff febeffdfffefffff fffffffffffffffe
B: MSC=10
B: LED=7

I: Bus=0003 Vendor=046a Product=b090 Version=0111
N: Name="Cherry USB keyboard System Control"
P: Phys=usb-0000:00:14.0-1/input1
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:046A:B090.0002/input/input5
U: Uniq=
H: Handlers=kbd event5 
B: PROP=0
B: EV=13
B: KEY=c000 10000000000000 0
B: MSC=10

I: Bus=0003 Vendor=046a Product=b090 Version=0111
N: Name="Cherry USB keyboard Consumer Control"
P: Phys=usb-0000:00:14.0-1/input1
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:046A:B090.0002/input/input6
U: Uniq=
H: Handlers=kbd event6 
[... 309 lines cut ...]

I: Bus=0005 Vendor=045e Product=0817 Version=0134
N: Name="Ergonomic Keyboard"
P: Phys=10:f6:0a:d2:1e:13
S: Sysfs=/devices/virtual/misc/uhid/0005:045E:0817.0008/input/input32
U: Uniq=c2:07:74:53:e0:36
H: Handlers=event19 
B: PROP=0
B: EV=9
B: ABS=10000000000

I: Bus=0005 Vendor=18d1 Product=9450 Version=0110
N: Name="Chromecast Remote"
P: Phys=10:f6:0a:d2:1e:13
S: Sysfs=/devices/virtual/misc/uhid/0005:18D1:9450.0009/input/input34
U: Uniq=d8:b6:73:cc:92:3f
H: Handlers=kbd event20 
B: PROP=0
B: EV=13
B: KEY=200000200000000 0 1004002000000 101041000000 e168000000000 0
B: MSC=10

```

i noticed the remote got disconnected here is an updated proc

chewitt, 2023-08-11:
```
# chromecast remote
evdev:input:b0005v045ep0817e0134*
 KEYBOARD_KEY_c0041=enter
```
Create /storage/.config/hwdb.d/99-chromecast.hwdb with ^ that content and reboot. If you confirm it works we can add it to LE12 nightlies and you can drop the manual workaround in the future.

DDQW, 2023-08-11:
i just tried sadly no luck. still everything works except the ok button.

chewitt, 2023-08-11:
```
# chromecast remote
evdev:input:b0005v18d1p9450e0110*
 KEYBOARD_KEY_c0041=enter
```
Helps if I don't read things on a phone .. try this ^

DDQW, 2023-08-11:
still the same

chewitt, 2023-08-11:
`Bus=0005 Vendor=18d1 Product=9450 Version=0110` is converted to `evdev:input:b0005v18d1p9450e0110*` so unless there's a typo I'm not seeing why that wouldn't work. You'll have to boot Fedora, find the .hwdb files, and see what config they mapped for the device.

DDQW, 2023-08-11:
```
[FAILED] librtmp-2.4-22.20190330.gitc5f04a5.fc37.x86_64.rpm: No more mirrors to try - All mirrors were already tried without success                                                                                                       
(18/18): compat-ffmpeg4-4.4.4-1.fc38.x86_64.rpm                                                    84% [==================================================================================-               ]  16 kB/s |  41 MB     08:14 ETA
The downloaded packages were saved in cache until the next successful transaction.
You can remove cached packages by executing 'dnf clean packages'.
Error: Error downloading packages:
  librtmp-2.4-22.20190330.gitc5f04a5.fc37.x86_64: Cannot download, all mirrors were already tried without success

```

sadly i cannot reinstall kodi to fedora at the moment seems to be broken.

could i get you something useful out of the chromecast with googletv or any other logs that might provide useful information?

My offer from second post also still stands fancy a chromecast remote? its actually quite comfortable when it works :D

chewitt, 2023-08-11:
Why do you need to install Kodi to look at OS .hwdb files?

DDQW, 2023-08-11:
i wanted to confirm it still does what it is supposed to since using it on fedora was a while ago.

 i got kodi installed now it also does not work on fedora anymore only on the chromecast. maybe google changed something with a firmware update.
 
 is there a way to see what exactly the remote is sending to libreelec even so it does not get interpreted correctly?

heitbaum, 2024-03-16:
Closing as fixed. Happy to open again - if this is not the case.

closed 2024-03-16 by heitbaum

chewitt, 2024-03-16:
I'll pick the change into a hwdb file update