LibreELEC/LibreELEC.tv #8044: [BUG] rsyslog addon will not run in LibreELEC as currently built because of incorrect library paths

adriankaylor, 2023-08-15:
### Describe the bug
The rsyslog addon will not run as built.  The rsyslogd binary is looking for its libraries in "/usr/lib/rsyslog/" instead of "/storage/kodi/addons/service.rsyslog/lib/rsyslog".

With /usr/lib being read only I haven't been able create a link back to the right folder to see if the app will load. 

### To Reproduce
Steps to reproduce the behavior:
1. Install and enable rsyslog
2. ssh in and run "~/.kodi/addons/service.rsyslog/bin/rsyslogd -d -n"

### Informations
 - LE Version: 11.03 (this has been an issue for a few years though.  I posted in the forums about it a while back but got no response.)
 - Hardware Platform: Intel NUC 11 and RPi4's

### Log file
In previous builds I saw this error in journald, but as of 11.03 I'm not seeing anything.
`systemd[1]: service.rsyslog.service: Main process exited, code=exited, status=1/FAILURE`

11.03 shows this at boot
```
LibreELEC (official): 11.0.3
[FAILED] Failed to start service.rsyslog.service.
[FAILED] Failed to start service.rsyslog.service.
[FAILED] Failed to start service.rsyslog.service.
[FAILED] Failed to start service.rsyslog.service.
[FAILED] Failed to start service.rsyslog.service.
[FAILED] Failed to start service.rsyslog.service.
```

Running  "~/.kodi/addons/service.rsyslog/bin/rsyslogd -d -n" will produce

```
"9400.677393386:main thread    : /var/lib/jenkins/LE/build2/workspace/Addons/All_Addons-Generic/build.LibreELEC-Generic.x86_64-9.1-devel/rsyslog-8.37.0/tools/rsyslogd.c: rsyslogd 8.37.0 startup, module path '', cwd:/storage/.kodi/addons/service.rsyslog/bin
9400.677545059:main thread    : /var/lib/jenkins/LE/build2/workspace/Addons/All_Addons-Generic/build.LibreELEC-Generic.x86_64-9.1-devel/rsyslog-8.37.0/runtime/glbl.c: rsyslog/glbl: using '127.0.0.1' as localhost IP
9400.677650435:main thread    : /var/lib/jenkins/LE/build2/workspace/Addons/All_Addons-Generic/build.LibreELEC-Generic.x86_64-9.1-devel/rsyslog-8.37.0/runtime/obj.c: caller requested object 'net', not found (iRet -3003)
9400.677700951:main thread    : /var/lib/jenkins/LE/build2/workspace/Addons/All_Addons-Generic/build.LibreELEC-Generic.x86_64-9.1-devel/rsyslog-8.37.0/runtime/modules.c: Requested to load module 'lmnet'
9400.677762382:main thread    : /var/lib/jenkins/LE/build2/workspace/Addons/All_Addons-Generic/build.LibreELEC-Generic.x86_64-9.1-devel/rsyslog-8.37.0/runtime/modules.c: loading module '/usr/lib/rsyslog/lmnet.so'
9400.677908286:main thread    : /var/lib/jenkins/LE/build2/workspace/Addons/All_Addons-Generic/build.LibreELEC-Generic.x86_64-9.1-devel/rsyslog-8.37.0/runtime/errmsg.c: Called LogMsg, msg: could not load module '/usr/lib/rsyslog/lmnet.so', dlopen: /usr/lib/rsyslog/lmnet.so: cannot open shared object file: No such file or directory

rsyslog internal message (3,-2066): could not load module '/usr/lib/rsyslog/lmnet.so', dlopen: /usr/lib/rsyslog/lmnet.so: cannot open shared object file: No such file or directory
[v8.37.0 try http://www.rsyslog.com/e/2066 ]
Error during class init for object 'conf' - failing...
rsyslogd initializiation failed - global classes could not be initialized.
Did you do a "make install"?
Suggested action: run rsyslogd with -d -n options to see what exactly fails.
rsyslogd: run failed with error -2066 (see rsyslog.h or try http://www.rsyslog.com/e/2066 to learn what that number means)"
```

### Additional context

heitbaum, 2023-08-15:
Hi @adriankaylor - the rsyslog logs above are not from a current version of the addon. Can you please update your addon first and share the logs. The addon logs shown above are from LE-9.1

adriankaylor, 2023-08-15:
Appologies, I must have grabbed those from my original post. 

```
1898.134938743:main thread    : /build/build.LibreELEC-Generic.x86_64-11.0-devel/build/rsyslog-8.2212.0/tools/rsyslogd.c: rsyslogd 8.2212.0 startup, module path '', cwd:/storage/.kodi/addons/service.rsyslog/bin
1898.139015091:main thread    : /build/build.LibreELEC-Generic.x86_64-11.0-devel/build/rsyslog-8.2212.0/runtime/glbl.c: rsyslog/glbl: using '127.0.0.1' as localhost IP
1898.139070487:main thread    : /build/build.LibreELEC-Generic.x86_64-11.0-devel/build/rsyslog-8.2212.0/runtime/obj.c: caller requested object 'net', not found (iRet -3003)
1898.139081701:main thread    : /build/build.LibreELEC-Generic.x86_64-11.0-devel/build/rsyslog-8.2212.0/runtime/modules.c: Requested to load module 'lmnet'
1898.139088733:main thread    : /build/build.LibreELEC-Generic.x86_64-11.0-devel/build/rsyslog-8.2212.0/runtime/modules.c: loading module '/usr/lib/rsyslog/lmnet.so'
1898.139297095:main thread    : /build/build.LibreELEC-Generic.x86_64-11.0-devel/build/rsyslog-8.2212.0/runtime/errmsg.c: Called LogMsg, msg: could not load module 'lmnet', errors: trying to load module /usr/lib/rsyslog/lmnet.so: /usr/lib/rsyslog/lmnet.so: cannot open shared object file: No such file or directory
1898.139302196:main thread    : /build/build.LibreELEC-Generic.x86_64-11.0-devel/build/rsyslog-8.2212.0/runtime/operatingstate.c: osf: MSG could not load module 'lmnet', errors: trying to load module /usr/lib/rsyslog/lmnet.so: /usr/lib/rsyslog/lmnet.so: cannot open shared object file: No such file or directory: rsyslog internal message (3,-2066): could not load module 'lmnet', errors: trying to load module /usr/lib/rsyslog/lmnet.so: /usr/lib/rsyslog/lmnet.so: cannot open shared object file: No such file or directory [v8.2212.0 try https://www.rsyslog.com/e/2066 ]
Error during class init for object 'conf' - failing...
rsyslogd initialization failed - global classes could not be initialized.
Did you do a "make install"?
Suggested action: run rsyslogd with -d -n options to see what exactly fails.
rsyslogd: run failed with error -2066 (see rsyslog.h or try https://www.rsyslog.com/e/2066 to learn what that number means)
```

adriankaylor, 2023-08-15:
I started poking around with building a new version of the addon locally and figured it out.  When I added "                           --libdir=/storage/.kodi/addons/service.rsyslog/lib \" to the PKG_CONFIGURE_OPTS_TARGET in its package.mk I was able to get rid of the library errors.  Hard coding the path probably isn't the best fix, but I think it at least proves the point.

Unfortunately, there may be some other issues with the addon.  Now that it runs, the command parameters in the unit file don't seem to be correct.  Still poking around there, but I'll submit those as another bug report when I have more specific info.

heitbaum, 2023-08-15:
The following sets the environment for the modules, so no need to change the compile. So if you set this while debugging you should be good to go. 

https://github.com/LibreELEC/LibreELEC.tv/blob/c46fefd6e3d74c50ff03380fecc908ebc4b25735/packages/addons/service/rsyslog/source/system.d/service.rsyslog.service#L8

I just tested here, and with setting the ip address to 1.1.1.1 and then systemctl start service.rsyslog.service / status. Looks right. 

```

● service.rsyslog.service - System Logging Service
     Loaded: loaded (/storage/.config/system.d/service.rsyslog.service; enabled; preset: disabled)
     Active: active (running) since Tue 2023-08-15 07:30:09 UTC; 52s ago
       Docs: http://www.rsyslog.com/doc/
    Process: 302428 ExecStartPre=/bin/sh /storage/.kodi/addons/service.rsyslog/bin/rsyslog.init (code=exited, status=0/SUCCESS)
   Main PID: 302442 (rsyslogd)
      Tasks: 4 (limit: 36592)
     Memory: 3.6M
        CPU: 146ms
     CGroup: /system.slice/service.rsyslog.service
             └─302442 /storage/.kodi/addons/service.rsyslog/bin/rsyslogd -n -f /storage/.kodi/userdata/addon_data/service.rsyslo

Aug 15 07:30:09 nuc12 systemd[1]: Starting service.rsyslog.service...
Aug 15 07:30:09 nuc12 rsyslogd[302442]: [origin software="rsyslogd" swVersion="8.2304.0" x-pid="302442" x-info="https://www.rsyslog.
Aug 15 07:30:09 nuc12 systemd[1]: Started service.rsyslog.service.
```

adriankaylor, 2023-08-16:
You're right, it's not the library path, total red herring.  I was getting the debug messages about the missing libraries because I was running the rsyslogd directly from the cli so it wasn't picking up the environment settings.  Once I go that out of my head and kept digging I figure out the reason my service wouldn't start was it wasn't getting any of its config files.

The problem is profiles.  I'm using multiple profiles and the unit file is looking for the addon settings in .kodi/userdata/addon_data/service.rsyslog/[rsyslog.conf,settings.xml] but because I'm in a profile the GUI settings are going to .kodi/userdata/profiles/Adrian/addon_data/....  Is there an environment variable that points to the active profile path and not Master settings?

I tweaked the unit file and the service is starting for me now.  Still no logs on my syslog server, but I'm still poking around with that.

heitbaum, 2023-08-18:
Closing as addon is running. Please reopen/update title if still a bug. If general support please use forum.

closed 2023-08-18 by heitbaum

adriankaylor, 2023-08-19:
Will do, thanks for your help.  I think I tracked down an actual bug, but it's way different than what I originally posted.  I'll file something new for clarity.