LibreELEC/LibreELEC.tv #8045: [BUG] Docker update and docker.linuxserver.portainer

easyfab, 2023-08-15:
### Describe the bug
With new Docker update, Portainer ( installed from addons ) doesn't launch.

And with docker.linuxserver.portainer command I have this message.

 portainer
http: invalid Host header

### Informations
 - LE Version: 	LibreELEC-RPi4.aarch64-12.0-nightly-20230809-c46fefd.img.gz 
 - Hardware Platform: RPI4

easyfab, 2023-08-15:
From https://github.com/LibreELEC/LibreELEC.tv/pull/8032 and https://github.com/moby/moby/releases/tag/v24.0.5

I see : 
The Go client now avoids using UNIX socket paths in the HTTP Host: header, in order to be compatible with changes introduced in go1.20.6. https://github.com/moby/moby/pull/45962, https://github.com/moby/moby/pull/45990

Could this be related to the problem ?

heitbaum, 2023-08-15:
- Fixed in #8032
- Should be available shortly

easyfab, 2023-08-16:
All good now.

Thank you for your quick reply and for all your great work.

closed 2023-08-16 by easyfab