LibreELEC/LibreELEC.tv #8076: [BUG] RPi4: Audio output (stereo jack) has short gaps in LE 11.0.1 and 11.0.3

amanita-citrina, 2023-08-27:
### Describe the bug
When viewing video content with an AddOn (e.g. YouTube, Crackle), the audio output on the stereo jack has short gaps (silence) for perhaps 100 msec, approximately every 2 seconds. When listening to audio files from my NAS, there are no gaps.

The bug was introduced with LE 11.0.1 and is still present in 11.0.3. Sound was perfect with LE 10.x and 11.0.0.

### How to reproduce
Steps to reproduce the behavior:
1. On a Raspberry Pi 4, Install LE 11.0.1. \
   The problem first appears in nightly build https://test.libreelec.tv/11.0/RPi/RPi4/LibreELEC-RPi4.arm-11.0-nightly-20230312-707238c.img.gz 
2. Play any video and listen to the audio on the stereo jack (3.5mm) output. Gaps are easier to hear with music.
3. There are short gaps in the audio, approximately every 2 seconds.

### Information
 - LibreELEC Version: 11.0.1 or nightly builds from 12-Mar-2023 or later
 - Hardware Platform: RPi 4 Model B Rev. 1.2, 4G RAM

### Context
- I installed 11.0.0 from scratch then updated to 11.0.1.
- I tried a lot of recommended settings, including those in the LE blog for 11.0.0 and 11.0.1. Nothing helped.
- Finally, I installed the nightly builds and found that the bug was introduced with the one from 12-Mar-2023.

popcornmix, 2023-09-01:
Just to confirm, if you revert to nightly from 11-Mar-2023 audio is perfect?
12-Mar-2023 is broken?

popcornmix, 2023-09-01:
PR's around that time:
```
1411552037      Tue Mar 14 14:42:02 2023 +0400  Merge pull request #7601 from heitbaum/binaries
4f571d5045      Mon Mar 13 20:58:59 2023 +0100  Merge pull request #7598 from heitbaum/linux61
39e5c4080c      Sun Mar 12 13:39:47 2023 +0100  Merge pull request #7594 from heitbaum/linux11
f6b79cb83c      Sun Mar 12 13:02:00 2023 +0100  Merge pull request #7591 from heitbaum/kernel61
6a357827a5      Sun Mar 12 15:55:58 2023 +0400  Merge pull request #7593 from heitbaum/kernel-firmware
ba72469d76      Sun Mar 12 12:52:31 2023 +0400  Merge pull request #7592 from heitbaum/amlogic-boot-fip
5bf76626e6      Sat Mar 11 10:22:22 2023 +0100  Merge pull request #7587 from heitbaum/flatbuffers
996cc5d215      Fri Mar 10 18:36:36 2023 +0100  Merge pull request #7584 from HiassofT/le12-ia-21.0.2
```
None look very likely (I believe the kernel bumps don't affect Pi).

amanita-citrina, 2023-09-01:
I could not find a nightly build from 11-Mar-2023, but this nightly build from 10-Mar-2023 works fine:
https://test.libreelec.tv/11.0/RPi/RPi4/LibreELEC-RPi4.arm-11.0-nightly-20230310-9f7c82b.img.gz

amanita-citrina, 2023-09-01:
The nightly build from 12-Mar is from 21:56, so the latest possible commit in that one is f6b79cb83c.
I guess the earliest possible commit would be one of the 7 commits on 10-Mar.

popcornmix, 2023-09-01:
So
https://test.libreelec.tv/11.0/RPi/RPi4/LibreELEC-RPi4.arm-11.0-nightly-20230310-9f7c82b.img.gz is good
https://test.libreelec.tv/11.0/RPi/RPi4/LibreELEC-RPi4.arm-11.0-nightly-20230312-707238c.img.gz is bad
(the git hashes are in filenames)
```
707238c7dd      Sun Mar 12 15:27:27 2023 +0100  Merge pull request #7596 from heitbaum/binaries11 [BAD]
daf339777c      Sun Mar 12 15:26:44 2023 +0100  Merge pull request #7595 from heitbaum/linux11.0
0f10694c26      Sun Mar 12 11:31:44 2023 +1100  Merge pull request #7590 from CvH/11.0/kodi-20.1
84d83d1083      Sat Mar 11 15:34:20 2023 +0100  Merge pull request #7585 from CvH/11.0/kodi_up_20230310
9f7c82be4b      Fri Mar 10 11:19:49 2023 +0100  Merge pull request #7566 from HiassofT/le11-kernel-6.1.14 [GOOD]
```

#7585 updates kodi from 20.0-Nexus to cfad9250
#7590 updated kodi from cfad9250 to 20.1-Nexus
#7595 kernel bump but not for pi
#7596 inputstream.adaptive: update 20.3.4-Nexus to 20.3.5-Nexus peripheral.joystick: update 20.1.7-Nexus to 20.1.8-Nexus

 I guess kodi bump and inputstream bump are both candidates.

amanita-citrina, 2023-09-02:
I will happily run any debug build (#7590?), but I do not have the means to build myself.

popcornmix, 2023-09-05:
I have made some builds around that time ([here](https://drive.google.com/uc?id=1-7rXYJ1PRMDG1O9byCueJdiMKxVGEX6a&export=download)):
LibreELEC-RPi4.aarch64-11.0-devel-20230905154540-707238c.tar
LibreELEC-RPi4.aarch64-11.0-devel-20230905154927-daf3397.tar
LibreELEC-RPi4.aarch64-11.0-devel-20230905155026-0f10694.tar
LibreELEC-RPi4.aarch64-11.0-devel-20230905155208-84d83d1.tar
LibreELEC-RPi4.aarch64-11.0-devel-20230905153611-9f7c82b.tar
Can you try them? If I've got this right, then 707238c7dd will be bad, 9f7c82be4b will be good,
and one of the intermediate builds will identify the PR that broke it.

amanita-citrina, 2023-09-05:
It seems that I can only download this one:

`LibreELEC-RPi4.aarch64-11.0-devel-20230905155208-84d83d1.tar`

Can you provide (individual) links for the other builds?

popcornmix, 2023-09-05:
https://drive.google.com/uc?id=1w0Lt5sRa7zFbEUp-pHh6FssQVz2s1Avj&export=download
https://drive.google.com/uc?id=19_ShPs9rSJSC-dfz-hBvMEheSeU5spEb&export=download
https://drive.google.com/uc?id=1ALDlovYFLqeZM8cdlvt9eny2dpdNSfOw&export=download
https://drive.google.com/uc?id=1WDYCSRO0x5zeVPDgiShN45k9-wx2N5V0&export=download
https://drive.google.com/uc?id=1-7rXYJ1PRMDG1O9byCueJdiMKxVGEX6a&export=download

amanita-citrina, 2023-09-05:
Well... LibreELEC-RPi4.aarch64-11.0-devel-20230905155208-84d83d1.tar resulted in (paraphrasing):
> Not compatible with RPi4.arm hardware
> Current system: RPi4.arm
> Update system: RPi4.aarch64
> Update canceled

There was a suggestion to create /storage/.update/.nocompat, but I dare not do that.

popcornmix, 2023-09-05:
Ah - my build system has switched to aarch64.
I can kick off an arm build (but I won't see results until tomorrow).

amanita-citrina, 2023-09-05:
Same problem with LibreELEC-RPi4.aarch64-11.0-devel-20230905154927-daf3397.tar.

Do you have any suggestions?

amanita-citrina, 2023-09-05:
No problem, thanks for building. I should have time to test tomorrow.

popcornmix, 2023-09-05:
The build was quicker than I expected.
[This](https://drive.google.com/uc?id=1qnT0v3THSCSzeBXxBd5CuaMEIixT-9CQ&export=download) should contain the 5 builds (for 32-bit arm).

amanita-citrina, 2023-09-05:
Here are the results:
```
BAD:  LibreELEC-RPi4.aarch64-11.0-devel-20230905155026-0f10694.tar
BAD:  LibreELEC-RPi4.aarch64-11.0-devel-20230905155208-84d83d1.tar
GOOD: LibreELEC-RPi4.aarch64-11.0-devel-20230905153611-9f7c82b.tar
```
So it seems that 84d83d1083 is the culprit. What now?

popcornmix, 2023-09-07:
These are kodi commits in that update:
```
cfad925044      Fri Mar 10 14:10:45 2023 +0100  [Windows] Fix SMB read chunk size when is not used FileCache
fc4234b396      Fri Mar 10 18:34:41 2023 +1000  [Backport] Sync controller add-ons with controller-topology-project
6de9250a3f      Thu Mar 9 16:50:01 2023 +0100   ActiveAESink: workaround for TrueHD and DD+ broken after pause or seek in AudioTrack RAW (Android only)
644cf0d7ba      Tue Mar 7 21:26:17 2023 +0100   Revert "AESinkAudioTrack: Use most simple pause() logic"
038f4e730d      Mon Mar 6 23:55:25 2023 -0800   [Backport] Fixes and improvements from "show actual controllers" PR
570d8e7c0b      Mon Mar 6 20:49:04 2023 -0800   [Backport] Controller Dialog: Fix "Ignore Input" buttons not being saved
ae03c8e9fe      Mon Mar 6 19:58:21 2023 +0100   [backport] NFSv3 and NFSv4 fixes and improvements
004dd56b63      Mon Mar 6 09:34:32 2023 +0000   [Addons] Having no repos is no error
e2aa74faa5      Mon Mar 6 05:26:23 2023 +1000   [Windows][Installer] fix the missing .html files of Chorus2 with proper exclusion of .h
cc00d8f519      Sat Mar 4 20:55:19 2023 +0100   AESinkAudioTrack: Revert sanity checking
24eaa5e6b2      Sat Mar 4 16:29:49 2023 +0000   [backport][Skin][Estuary] Fix player settings adjustments if autoclose video osd is active
c2e5a221b6      Sat Mar 4 17:02:22 2023 +0100   [video] Fix watched status not taken into account when queueing items.
2c3d97d34f      Thu Mar 2 11:14:38 2023 +0000   [Backport] guilib: Fix warning in GUIListGroup control
22153da8a0      Wed Mar 1 18:47:25 2023 +1000   [backport][Valgrind] Fix some jumps depending on uninitialized values
af68c437f2      Mon Feb 27 22:17:44 2023 +0100  AudioTrack: Increase robustness
d8bbaa4cad      Mon Feb 27 14:21:25 2023 +0000  [Backport][GUIImage] Fix regression for color diffuse if no info is set
e71c00df43      Mon Feb 27 10:17:22 2023 +1000  StringUtils: fix sorting by name with ascii punctuation and symbols o…
82e957ec43      Sun Feb 26 19:27:53 2023 +0100  VideoPlayerAudio: use simple algorithm for self-learning max allowed a/v Out-Of-Sync
543c82e1e1      Sun Feb 26 11:55:32 2023 +0100  [PVR] Rework PVR windows late init.
db47f82ea1      Fri Feb 24 15:47:41 2023 -0800  [Backport] Port Dialog: Fix resetting multitaps
b5e2926b81      Fri Feb 24 09:58:55 2023 +0000  [GUI][GUIImage] Fix inconsistencies for colordiffuse
636d4e2cb3      Fri Feb 24 09:04:52 2023 +0100  [backport][libass] Temporary font folder checking
f5de6ceb65      Tue Feb 21 02:28:30 2023 -0800  [Backport] Savestate rendering
246fd7f19f      Tue Feb 21 07:50:08 2023 +1000  [backport][Android] Fix symbol relocation errors on Android x86
93376f096b      Mon Feb 20 16:12:58 2023 +0100  [backport][Subtitles][TX3G] Fix styles lookup
bf0c616eda      Mon Feb 20 16:12:26 2023 +0100  [backport][Subtitles] Fix RTL on subtitles
0e7b997584      Sun Feb 19 11:45:35 2023 +0000  [settings][airplay] Fix password setting order/dependencies
45b31ef353      Fri Feb 17 14:02:16 2023 +1000  [backport] CSMBDirectory: fix return value for smbc_getxattr
c20c209e9d      Thu Feb 16 23:56:55 2023 +0000  [backport][upnp]: load library art before broadcasting remote watched state
a5e84dc09f      Fri Feb 17 00:15:20 2023 +0100  [Nexus] Update translation related workflows
77464668fc      Thu Feb 16 22:46:41 2023 +0000  [backport][upnp]: fix state reporting
3a7a8b1772      Tue Feb 14 22:31:56 2023 -0800  [Backport] RetroPlayer: Faster in-game savestate dialog and autosaves
5c91216733      Tue Feb 14 22:31:29 2023 -0800  [Backport] Savestate Manager: Break dependency on DialogSelect and add "Saved with"
402e7bc5b2      Mon Feb 13 08:17:13 2023 +0000  [backport][pvr] Always retirieve the max DB ID for providers when reading them from the DB - Nexus
740c681447      Sun Feb 12 16:34:55 2023 +0100  Audio Backports from testing group
ea5d936a5b      Sun Feb 12 14:03:29 2023 +0100  [backport][Subtitles][Libass] Fix line spacing to avoid overlap box
d40e1d69b5      Sun Feb 12 14:03:16 2023 +0100  [backport][Subtitles][WebVTT] Fix overlapped subtitles on malformed segments
1794875a0b      Sun Feb 12 14:02:56 2023 +0100  [backport][Subtitles][WebVTT] Fixed segmented webvtt sync
8ea7280764      Sun Feb 12 14:02:39 2023 +0100  [backport][Subtitles][TagSami] Handle basic color names
391bd218bd      Sun Feb 12 14:02:25 2023 +0100  [Backport][Subtitles] Fix subtitles in livestreams with high start timestamps
fb2ff3afaa      Sun Feb 12 11:48:36 2023 +0100  render the .notdef characters to fix drawing of text with missing unicode font
885f1505d6      Sun Feb 12 17:01:03 2023 +1000  [backport] tools/depends: Python update to 3.11.2 
5ce2d74a6c      Fri Feb 10 12:54:08 2023 +0100  fixed: PVR timer validation was broken for certain (TVHeadend) timer rules
ca100ec1c3      Thu Feb 9 14:14:25 2023 -0800   [backport] RetroPlayer: Improve reading from write-only memory
296733727a      Thu Feb 9 10:11:26 2023 +1000   [backport][python][xbmcgui] Fix ControlEdit constructor args
1a1d744a4d      Sat Feb 4 14:07:54 2023 -0800   [Backport] RetroPlayer: Code improvements
8fcd090f0d      Fri Feb 3 17:07:37 2023 +0000   [dllwrapper] wrap dll_stat64 and dll_fstat64
e9f58cbf13      Thu Feb 2 18:35:55 2023 +0100   [Windows] handle WM_INITMENU event to disable move from system menu when window is in full screen
a0255ae194      Tue Jan 31 19:42:27 2023 -0800  [Backport] Game OSD: Address savestate manager usability issues
[... cut ...]
d463c0769d      Mon Jan 23 18:16:56 2023 +0100  [backport] TrueHD fixes + AudioTrack fixes + PAPlayer fixes
1853fb7dea      Sun Jan 22 10:55:44 2023 +0100  [PVR] Fix sort by channel backend order.
09db987a1d      Sat Jan 21 14:55:02 2023 +0100  [video][music] Fixes for playback related context menu item apperance
ec5b71680e      Mon Jan 16 21:52:52 2023 +0100  DXVA: fix AMD SD interlaced workaround
052d4afc08      Sun Jan 15 21:05:17 2023 +0100  EndTime: fix overflow when doing large comparisons
389e701cb9      Sun Jan 15 19:13:34 2023 +1000  [Nexus] Bump version 20.0 Final
```

popcornmix, 2023-09-07:
Quite a few commit are network (nfs/smb) related. Can you confirm you get the issue playing a file locally (e.g. copy the file to sdcard or directly attached usb stick) to rule them out.

amanita-citrina, 2023-09-07:
We can rule out smb/nfs. I downloaded a short file to the SD card:
```
GOOD: LE 11.0.0
BAD:  LibreELEC-RPi4.aarch64-11.0-devel-20230905155208-84d83d1.tar
```
We could try to zoom in on the offending kodi commit, e.g., with three debug builds using the kodi commits at 25%, 50% and 75% of the commit list. This should reveal the exact commit after 3-4 rounds. What do you think?

amanita-citrina, 2023-10-06:
I found out that the bug was introduced in kodi commit 82e957e (see updated description).
Maybe I can find the source of the problem, but any help is greatly appreciated.

popcornmix, 2023-10-07:
I'd suggest creating a kodi issue. Then you can get feedback from author of that commit.

amanita-citrina, 2023-10-08:
I finally found the root of the problem. In config.txt, I had `audio_pwm_mode=1`. Apparently, this legacy setting requires continuous audio sync correction, which became audible when kodi increased the possible sync offset to 100ms.

Setting `audio_pwm_mode=2` solved my problem. Audio is now fine with LE 11.0.3.

I see no reason to use the legacy setting, so I will close the issue.
@popcornmix: Thank you for your help! I hope I can make more meaningful contributions to LE in the future.

closed 2023-10-08 by amanita-citrina

amanita-citrina, 2023-10-11:
...one more thing: The setting `audio_pwm_mode=1` is recommended in the blog post for  [LE 10.0.0](https://libreelec.tv/2021/08/26/libreelec-matrix-10-0/). It may have been correct at the time, but now it might cause problems.