LibreELEC/LibreELEC.tv #8077: [BUG] systemd: found ordering cycle

rschupp, 2023-08-27:
### Describe the bug
from journal:

Aug 09 13:51:00 LibreELEC systemd[1]: machine-id.service: Found ordering cycle on local-fs.target/start
Aug 09 13:51:00 LibreELEC systemd[1]: machine-id.service: Found dependency on local-fs-pre.target/start
Aug 09 13:51:00 LibreELEC systemd[1]: machine-id.service: Found dependency on systemd-tmpfiles-setup-dev.service/start
Aug 09 13:51:00 LibreELEC systemd[1]: machine-id.service: Found dependency on machine-id.service/start
Aug 09 13:51:00 LibreELEC systemd[1]: machine-id.service: Job local-fs.target/start deleted to break ordering cycle starting with machine-id.service/start

### How to reproduce
boot machine

### Information
 - LibreELEC Version: 12
 - Hardware Platform: RPi4

### Context
/usr/lib/systemd/system/machine-id.service:
```
Before=systemd-journald.service systemd-tmpfiles-setup-dev.service shutdown.target
After=local-fs.target
```
hence (> is After):
**systemd-tmpfiles-setup-dev.service > machine-id.service > local-fs.target**

/usr/lib/systemd/system/systemd-tmpfiles-setup-dev.service:
```
Before=sysinit.target local-fs-pre.target systemd-udevd.service
```
**local-fs-pre.target > systemd-tmpfiles-setup-dev.service** > machine-id.service > local-fs.targe

/usr/lib/systemd/system/local-fs.target:
```
After=local-fs-pre.target
```
**local-fs.target > local-fs-pre.target** > systemd-tmpfiles-setup-dev.service > machine-id.service > local-fs.targe 💣

chewitt, 2023-08-28:
```
RPi4:~ # journalctl -l | grep machine
Aug 28 18:35:43 LibreELEC systemd[1]: local-fs.target: Found dependency on machine-id.service/start
Aug 28 18:35:43 LibreELEC systemd[1]: local-fs.target: Found dependency on machine-id.service/start
Aug 28 18:35:43 LibreELEC systemd[1]: Starting machine-id.service...
Aug 28 18:35:43 LibreELEC systemd[1]: Finished machine-id.service.
```
^ from a current nightly-ish build on RPi hardware

rschupp, 2023-08-28:
> Aug 28 18:35:43 LibreELEC systemd[1]: local-fs.target: Found dependency on machine-id.service/start

That **is** one line in a series of lines that show the problem. Search for "ordering cycle" in `journalctl -b` or do
```
pi4:~ # systemd-analyze verify default.target
var-log.mount: Found ordering cycle on local-fs-pre.target/start
var-log.mount: Found dependency on systemd-tmpfiles-setup-dev.service/start
var-log.mount: Found dependency on machine-id.service/start
var-log.mount: Found dependency on local-fs.target/start
var-log.mount: Found dependency on var-log.mount/start
var-log.mount: Job local-fs-pre.target/start deleted to break ordering cycle starting with var-log.mount/start
```
The unit where this is reported may vary.

PR #8086 (2023-09-01): systemd/busybox: fix unit permissions after systemd v254 bump
machine-id.service fix for:
```
systemd[1]: local-fs.target: Found ordering cycle on local-fs-pre.target/start
systemd[1]: local-fs.target: Found dependency on systemd-tmpfiles-setup-dev.service/start
systemd[1]: local-fs.target: Found dependency on machine-id.service/start
systemd[1]: local-fs.target: Found dependency on local-fs.target/start
systemd[1]: local-fs.target: Job local-fs-pre.target/start deleted to break ordering cycle starting with local-fs.target/start
```
Resolve #8077

var-log.mount fix for:
```
systemd[1]: local-f [... cut]
closed 2023-09-01 by chewitt