LibreELEC/LibreELEC.tv #8101: [BUG] LE12 - Fails to create GUI x86-64 After 2023 April 27

smf007, 2023-09-03:
### Describe the bug

With the LE12 nightlies, after 2023 April 27 both Generic and Generic-Legacy fail to produce a GUI.  Last working snapshot that works is April 27 d287ee22cc79f07eb79df30dba55d0d23bdd8827.  I have also tested the latest master as there was a kernel bump, however that has not changed things.

### How to reproduce
Steps to reproduce the behavior:
Update to any nightly of master after 2023 April 27, Generic or Generic-Legacy

### Information
 - LibreELEC Version: 12
 - Hardware Platform: x86-64

### Log file

Generic
https://pastebin.com/GKnXFa2g
Generic-Legacy
https://pastebin.com/Zs0xawSE

### Context

heitbaum, 2023-09-04:
Issue is with your mysql connection.

Please move the support request to the forums:

```
2023-09-03 12:08:18.646 T:901     debug <general>: Initialize, updating databases...
2023-09-03 12:08:18.647 T:901      info <general>: Running database version Addons33
2023-09-03 12:08:18.648 T:901      info <general>: Running database version ViewModes6
2023-09-03 12:08:18.650 T:901      info <general>: Running database version Textures13
2023-09-03 12:08:18.650 T:901     debug <general>: connect replacing configured host 192.168.250.120 with resolved host 192.168.250.120
2023-09-03 12:08:18.701 T:901      info <general>: MYSQL: Connected to version 8.0.34-0ubuntu0.20.04.1
2023-09-03 12:08:18.771 T:901      info <general>: Running database version MyMusic82
2023-09-03 12:08:18.773 T:901     debug <general>: connect replacing configured host 192.168.250.120 with resolved host 192.168.250.120
2023-09-03 12:08:18.882 T:901     error <general>: SQL: [MyVideos122] The table does not exist
                                                   Query: SELECT idVersion FROM version
 
 
2023-09-03 12:08:18.889 T:901     error <general>: Process error processing job 
2023-09-03 12:08:21.988 T:892     debug <general>: Sink changed
2023-09-03 12:08:48.889 T:901     debug <general>: Thread JobWorker 139845989025472 terminating (autodelete)
````

closed 2023-09-04 by heitbaum

smf007, 2023-09-04:
The error is occurs earlier in the log file.  Confirmed with a blank ~/.kodi as well, so definitely not mySQL

smf007, 2023-09-04:
Log with blank profile but with debug enabled, generic build
https://paste.kodi.tv/aqofozukun.kodi

heitbaum, 2023-09-04:
The first log shows the edid issue. Do you have another screen to test with? Can you create an edid file on the “working LE” and does it work on the more recent LE?

smf007, 2023-09-04:
I've gone through and captured the EDID however it doesn't make a difference.

I should add that the boot screen no longer shows as well.  In generic-legacy I note that X11 has the following even when the EDID is provided.

> 2023-09-04 10:56:37.542 T:937   warning <general>: CXRandR::Query - output HDMI2 has no current mode, assuming disconnected

It seems to be more lower level, like gpu mode setting in the kernel.

Generic-Legacy with EDID being forced
https://paste.kodi.tv/yitiromude.kodi

Generic with EDID being forced
https://paste.kodi.tv/layokoremi.kodi

I'll try to get another display to test, however I don't think it's the display causing this.  Is there any kernel switches that can be added to force a view mode?

heitbaum, 2023-09-04:
As you seem to have got it working via hard coding. Maybe testing with another display/cable is probably best at this stage- that you are not seeing anything on the boot screen. Also options on updating syslinux. Suggest moving this to the forums where you will get others input - e.g. something similar https://forum.libreelec.tv/thread/26955-no-hdmi-output-since-le-11/

smf007, 2023-09-09:
I did a bunch more testing.  What I've found is the system is actually booting fine and appears to be correctly setting the display.  If I remove "quite" from the kernel options, I'm clearly getting output.

I've also tried other displays/monitors, enabling/disabling modesetting, forcing modes, and more.  In all those cases it looks like the updated kernel is correctly setting things up.  It's looking more like a Kodi issue.  Specifically, this combination of processor/bios/kernel is resulting in Kodi attempting the wrong display settings.... which might also be why the GBM version is also failing when selecting the video plane.  Even when the Kodi setup has a working resolution in it's configuration files, Kodi is doing something else.

In the case of X11, using xrandar to set the resolution after Kodi has started is more a hack around the problem.  It would further explain why hardware decoding is missing.

I've tried a similar vintage of system (Sandybridge i7 instead of i5, different motherboards) and things worked.  So this is a very corner case issue it seems.

So at this time my best guess is the kernel update uncovered a bug in Kodi that this specific configuration hits, a very small corner case.  I'll take this over to Kodi, perhaps Lukas has some command to run that could shed some light on how Kodi is misinterpreting things.