LibreELEC/LibreELEC.tv #8124: Audio Flac file - file corrupted!

pirrys, 2023-09-08:
Hi, I am experiencing a weird phenomenon with High resolution audio file, they will play perfectly fine until randomly a "scratch"will be heard, once it happens the scratches can be heard over again at the same time.

The interesting part is that the Flac file became corrupted, I have listen to the same audio file on a different support with the same annoying noises while the original file in my music library on laptop plays perfectly.
What I use:
Raspberry pi 4 with DAC+
Libreelec latest version
Music library stocked on a USB memory flash drive.

Thank you.

Typical audio file:

Audio
Format: FLAC
Format/Info: Free Lossless Audio Codec
Duration: 4 min 57 s
Bit rate mode: Variable
Bit rate: 3 137 kb/s
Channel(s): 2 channels
Channel layout: L R
Sampling rate: 96.0 kHz
Bit depth: 24 bits
Compression mode: Lossless
Stream size: 111 MiB (100%)
Writing library: libFLAC 1.3.1 (UTC 2014-11-25)

heitbaum, 2024-03-11:
Is this issue still occurring with latest LE12 20240311-86dba4b images. If so please share the debug logs. Alternatively please raise this on the forum for support.

heitbaum, 2024-03-16:
Closing as inactive. Happy to open again - but will need debug logs, ideally a reproducible case. Using the forum for support as others may have seen the same issue.

closed 2024-03-16 by heitbaum

pirrys, 2024-10-07:
Hi, I guess I have been away for a while, not sure if the thread can be reopen?
I still encounter the same problem here and there. Some audio file get randomly corrupted with scratch noises. 
Thanks.

heitbaum, 2024-10-07:
Suggest using the forum for support - more eyes will have a look at it.