LibreELEC/LibreELEC.tv #8125: [BUG] rsyslog addon compiled against wrong prefix

jsaathof, 2023-09-08:
### Describe the bug
The rsyslog addon cannot load modules and therefore cannot start. 

The modules are searched in _/usr/lib/rsyslog_ which is not the correct folder. The module is installed in _/storage/.kodi/addons/service.rsyslog_. According to the manpage of rsyslog the modules are searched in the default directory prefix/lib/rsyslog. The prefix has been set to _/usr_. 

The module should be recompiled with _/storage/.kodi/addons/service.rsyslog_ as prefix.

### How to reproduce
Steps to reproduce the behavior:
1. Login via SSH
2. Run "/storage/.kodi/addons/service.rsyslog/bin/rsyslogd -n"
3. See error

### Information
 - LibreELEC Version: 11.0.3
 - Hardware Platform: Intel NUC7i3BNK

### Log file
```
LibreELEC:~ # /storage/.kodi/addons/service.rsyslog/bin/rsyslogd -n
rsyslog internal message (3,-2066): could not load module 'lmnet', errors: trying to load module /usr/lib/rsyslog/lmnet.so: /usr/lib/rsyslog/lmnet.so: cannot open shared object file: No such file or directory [v8.2308.0 try https://www.rsyslog.com/e/2066 ]
Error during class init for object 'conf' - failing...
rsyslogd initialization failed - global classes could not be initialized.
Did you do a "make install"?
Suggested action: run rsyslogd with -d -n options to see what exactly fails.
rsyslogd: run failed with error -2066 (see rsyslog.h or try https://www.rsyslog.com/e/2066 to learn what that number means)
```

### Context
The manpage on a different node reads:

```
FILES
       /etc/rsyslog.conf
              Configuration file for rsyslogd.  See rsyslog.conf(5) for exact information.
       /dev/log
              The Unix domain socket to from where local syslog messages are read.
       /var/run/rsyslogd.pid
              The file containing the process id of rsyslogd.
       prefix/lib/rsyslog
              Default directory for rsyslogd modules. The prefix is specified during compilation (e.g. /usr/local).
```
The prefix supplied during compilation shows it is set to _/usr_ because rsyslogd searches for the modules in that location. The addon should be recompiled with _/storage/.kodi/addons/service.rsyslog_ as prefix.

heitbaum, 2023-09-08:
Closing. Please see #8044

closed 2023-09-08 by heitbaum

jsaathof, 2023-09-09:
Sorry, somehow I could find that ticket. Bun since this is not fixed and does not work "out of the box" when installing the addon I still think this is a bug. The addon was not updated. 
I have reinstalled the addon before creating this issue. Even removed the folder in _/storage/.kodi/addons_ to be sure no incorrect information was left behind. I think it is weird to make an addon work by having to SSH into libreelec and changing things.

heitbaum, 2023-09-09:
> Sorry, somehow I could find that ticket. Bun since this is not fixed and does not work "out of the box" when installing the addon I still think this is a bug. The addon was not updated. I have [...]

Because of the way rsyslog plugins are done - the supported startup is via the systemd script (and this is automated)
You should not need to ssh to activate the addon.
addons may require reboots (known issue).

the following lines are part of the startup.

https://github.com/LibreELEC/LibreELEC.tv/blob/c4db2d1d4a51c9b4f871a861a0f9fc62fb0b6154/packages/addons/service/rsyslog/source/system.d/service.rsyslog.service#L8-L10

If you were looking to improve the addon build - to directly specify the RSYSLOG_MODDIR during build - then PR would be welcome.

jsaathof, 2023-09-09:
I will check again.

adriankaylor, 2023-09-13:
@jsaathof This ended up being a problem with how settings were read when you were not using the default profile.  All of the library stuff above was a problem with my testing.  Check the forums.  I just posted a novel about it with a workaround if you are also using a profile other than Master.