LibreELEC/LibreELEC.tv #8140: [BUG] SystemTools (Regression in MC)

eldersixpack13, 2023-09-12:
### Describe the bug
MC from SystemTools Version 11.80.2.**3** is malfunction (here)
Version 11.80.2.2 is okay 
and I hope upcoming 11.80.2.4 will be okay too (I'm unsure)

### How to reproduce
Steps to reproduce the behavior:
1. ssh into LE
2. run mc and try to copy/move a file to the other pane
3. NOTHING will happen !

### Information
 - LibreELEC Version: 12 (nightly e17bf16)
 - Hardware Platform: NUC13

### Log file
nothing in the logs

### Context
the box is a new setup from today
- installed LE11
- removed .kodi
- put last LE12 nightly under .update
- ...

heitbaum, 2023-09-13:
@eldersixpack13 - can you please raise the bug with the mc team. Thanks. It will get packaged when bumped - or if there is a patch - I will look at if it can be included before the release.

eldersixpack13, 2023-09-13:
sure if I could help to lower your work !

but it doesn't make sense to me
Why ?

mc is since 13.08.2023 on Version 4.8.30
it (seem to me) it hasn't changed since (?!)

the breakage went in from SystemTools Version 11.80.2.2 to 11.80.2.3

so my understanding: it's/it seems an LE bug

will test and report 11.80.2.4 though ...

eldersixpack13, 2023-09-14:
for the record:
11.80.2.4 isn't working too

last working is 11.80.2.2

rellla, 2023-09-26:
Just pinging this closed issue...
Didn't bisect it, but I can confirm the above behaviour.
Additionally mc isn't building here with curren mastr atm due to not finding pcre2/pcre:
```
checking if PCRE2 is wanted... yes
checking for pcre2_compile_8 in -lpcre2-8... no
checking for pcre2.h... no
configure: WARNING: Cannot find pcre2 library, trying pcre one...
checking lib pcre... (testing)
checking for pcre_study in -lpcre... no
checking for pcre_compile in -lpcre... no
checking lib pcre... no, (WARNING)
configure: error: Neither pcre2 nor pcre library found!
```

heitbaum, 2023-09-27:
> Just pinging this closed issue... Didn't bisect it, but I can confirm the above behaviour. Additionally mc isn't building here with current master atm due to not finding pcre2/pcre:
> 
> ```
> checking if PCRE2 is wanted... yes
> checking for pcre2_compile_8 in -lpcre2-8... no
> checking for pcre2.h... no
> configure: WARNING: Cannot find pcre2 library, trying pcre one...
> checking lib pcre... (testing)
> checking for pcre_study in -lpcre... no
> checking for pcre_compile in -lpcre... no
> checking lib pcre... no, (WARNING)
> configure: error: Neither pcre2 nor pcre library found!
> ```

Hi @rellla - using Ubuntu jammy the builds are completing fine. Just ran a test now - http://ix.io/4Hzb 

in the config.log
```
configure:20788: checking if PCRE2 is wanted
configure:20811: result: yes
configure:20840: checking for pcre2_compile_8 in -lpcre2-8
configure:20863: /var/media/DATA/home-rudi/LibreELEC.kernel11/build.LibreELEC-Generic.x86_64-12.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc -o conftest -march=x86-64 -Wall -pipe  -O2 -fomit-frame-pointer -DNDEBUG   -Wl,--as-needed -fuse-ld=gold -lcrypto -lssl conftest.c -lpcre2-8   >&5
configure:20863: $? = 0
configure:20873: result: yes
configure:20882: checking for pcre2.h
configure:20882: /var/media/DATA/home-rudi/LibreELEC.kernel11/build.LibreELEC-Generic.x86_64-12.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc -c -march=x86-64 -Wall -pipe  -O2 -fomit-frame-pointer -DNDEBUG  conftest.c >&5
configure:20882: $? = 0
configure:20882: result: yes
```

So guessing a localized issue, or something unexpected coming through.

heitbaum, 2023-09-27:
Whilst I had mc running - I have done the tests and can’t reproduce the reported error.
I tested with F6
[image]
[image]
[image]

rellla, 2023-09-27:
@heitbaum Thanks for your answer. Yes, i'm building on Debian 11, but never had a problem so far. My guess, is, that the build host system is not the reason.
This is, what I get: https://pastebin.com/raw/9mCN3PyW
Smells, like "internal compiler error" doesn't sound good. Any hints i could search for the reason? 

@eldersixpack13 Does this occur, when working via ssh or locally? I'm connected via ssh/putty. Can you bisect it to a specific commit?

eldersixpack13, 2023-09-27:
ssh (linux box => LE box) 
bisect: no !

bug went in: 11.80.2.2 to 11.80.2.3

heitbaum, 2023-09-27:
My build / buildhost matching yours `PROJECT=Rockchip ARCH=arm DEVICE=RK3399 s/build mc`

```
configure:20788: checking if PCRE2 is wanted
configure:20811: result: yes
configure:20840: checking for pcre2_compile_8 in -lpcre2-8
configure:20863: /var/media/DATA/home-rudi/LibreELEC.kernel11/build.LibreELEC-RK3399.arm-12.0-devel/toolchain/bin/armv8a-libreelec-linux-gnueabihf-gcc -o conftest -march=armv8-a+crc -mtune=cortex-a72.cortex-a53 -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mfloat-abi=hard -mfpu=crypto-neon-fp-armv8 -Wall -pipe  -O2 -fomit-frame-pointer -DNDEBUG   -Wl,--as-needed -fuse-ld=gold -lcrypto -lssl conftest.c -lpcre2-8   >&5
configure:20863: $? = 0
configure:20873: result: yes
configure:20882: checking for pcre2.h
configure:20882: /var/media/DATA/home-rudi/LibreELEC.kernel11/build.LibreELEC-RK3399.arm-12.0-devel/toolchain/bin/armv8a-libreelec-linux-gnueabihf-gcc -c -march=armv8-a+crc -mtune=cortex-a72.cortex-a53 -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mfloat-abi=hard -mfpu=crypto-neon-fp-armv8 -Wall -pipe  -O2 -fomit-frame-pointer -DNDEBUG  conftest.c >&5
configure:20882: $? = 0
configure:20882: result: yes
```

the difference on your log is that `-I/usr/include` and `-L/usr/lib` have come from somewhere. Do you have pcre2-dev on your buildhost?

```
configure:20863: /home/andreas/git/VDRSternELEC2/LibreELEC.tv/build.LibreELEC-RK3399.arm-12.0-devel/toolchain/bin/armv8a-libreelec-linux-gnueabihf-gcc -o conftest -march=armv8-a+crc -mtune=cortex-a72.cortex-a53 -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mfloat-abi=hard -mfpu=crypto-neon-fp-armv8 -Wall -pipe  -O2 -fomit-frame-pointer -DNDEBUG  -I/usr/include  -Wl,--as-needed -fuse-ld=gold -lcrypto -lssl -L/usr/lib conftest.c -lpcre2-8   >&5
CROSS COMPILE Badness: /usr/include in INCLUDEPATH: /usr/include
cc1: internal compiler error: in add_path, at incpath.cc:481
0x7f51d6470d09 __libc_start_main
	../csu/libc-start.c:308
```

heitbaum, 2023-09-27:
> ssh (linux box => LE box) bisect: no !
> 
> Tried to get some logs with component logging and compared the ldd's: nothing !
> 
> bug went in: 11.80.2.2 to 11.80.2.3

Can you try a simpler test - so as to isolate file system boundaries / permissions ….
This is the test I did.
```
(As root)
cd /storage
mkdir 123
cd 123
mkdir a
mkdir b
touch a/file
mc
# left pane = a
# right pane = b
F6 on file —> b
```

rellla, 2023-09-27:
Good hint. After a "apt-get remove libpcre2-dev" on my build host, the build completes. But why is that interfering with a toolchain/cross-build? After removing libpcre2-dev i don't have the /usr/* dirs included anymore...
Removing libpcre2-dev seems to be a workaround, but shouldn't it also build when it's there?
Something has introduced this regression, because it worked before. 
Maybe i should look into mc sources...

rellla, 2023-09-27:
For the copy problem, i will do an update an see if i can help - if it's not solved before.

rellla, 2023-09-27:
> Can you try a simpler test - so as to isolate file system boundaries / permissions …. This is the test I did.
>
> ```
> (As root)
> cd /storage
> mkdir 123
> cd 123
> mkdir a
> mkdir b
> touch a/file
> mc
> # left pane [...]

Doesn't work here. With my "old" setup.

rellla, 2023-09-27:
@heitbaum: solved the build error:

we need a "--with-pcre2=${SYSROOT_PREFIX}/usr" configure option.
See https://github.com/MidnightCommander/mc/commit/49624e473bef82ab56e5e89271f6c4b4cd0f9e5c#diff-ab925d7ab80c31592b79585268b1893c5d35c08cd807a25ec5152a71000ee58eR11

eldersixpack13, 2023-09-27:
@heitbaum 
tried your test and dir b remains empty

Rights:
root@nuc:~$ ls -lR 123
123:
total 8
drwxr-xr-x    2 root     root          4096 Sep 27 19:17 a
drwxr-xr-x    2 root     root          4096 Sep 27 19:17 b

123/a:
total 0
-rw-r--r--    1 root     root             0 Sep 27 19:17 file

123/b:
total 0
root@nuc:~$

heitbaum, 2023-09-27:
> Good hint. After a "apt-get remove libpcre2-dev" on my build host, the build completes. But why is that interfering with a toolchain/cross-build? After removing libpcre2-dev i don't have the /usr/* [...]

:-) probably never worked - as pcre2 was new in 4.8.30 - 6d322bfdfe - and we have been progressively removing pcre #6798

heitbaum, 2023-09-27:
> @heitbaum: solved the build error:
> 
> we need a "--with-pcre2=${SYSROOT_PREFIX}/usr" configure option. See [MidnightCommander/mc@49624e4#diff-ab925d7ab80c31592b79585268b1893c5d35c08cd807a25ec5152a71000ee58eR11](https://github.com/MidnightCommander/mc/commit/49624e473bef82ab56e5e89271f6c4b4cd0f9e5c#diff-ab925d7ab80c31592b79585268b1893c5d35c08cd807a25ec5152a71000ee58eR11)

If you would like to PR it. No need to bump system-tools addon (as it is a non impacting configuration for the buildhosts) but will fix the self build with pcre2-dev installed

rellla, 2023-09-28:
done: https://github.com/LibreELEC/LibreELEC.tv/pull/8172

eldersixpack13, 2023-10-03:
Question to the universe:

Will this bug be fixed and until then treated as "Open" **or I should stop reporting bugs here** ? 
(cause the maintainer can't reproduce the bugs I reported and (partly) closed them immediately (!) after I reported them)

Sorry, I'm pissed !!!

heitbaum, 2023-10-03:
> Question to the universe:
> 
> Will this bug be fixed and/or until then treated as "Open" **or I should stop reporting bugs here** ? (cause the maintainer can't reproduce the bugs and (partly) closed them immediately (!) after I reported them)
> 
> Sorry, I'm pissed !!!

Hi @eldersixpack13 - ref: https://github.com/LibreELEC/LibreELEC.tv/issues/8140#issuecomment-1716943695 You may want to see if it is a localized issue as `root@nuc:~$` is not the standard prompt. Otherwise at a complete loss and working with the mc developers/strace/gdb may resolve this. Alternatively rolling back to your working version. Happy to accept a fix - if it is a compile option that needs to be changed, otherwise any code issue would need to be addressed in upstream.

eldersixpack13, 2023-10-03:
cit.: "_You may want to see if it is a localized issue as root@nuc:~$ is not the standard prompt._"

quiet clear:
root@nuc:~$ is the command prompt when running mc in background (key CRTL+o)

nuc:~ # is the default prompt (mc off)

anyway, it remains "root is the default LE user (at least here)"
no ?

cat /etc/passwd
systemd-timesync:x:191:191:systemd-timesync:/:/bin/false
systemd-network:x:193:193:systemd-network:/:/bin/sh
dbus:x:81:81:System message bus:/:/bin/sh
**root**:x:0:0:Root User:**/storage**:/bin/sh
nobody:x:65534:65534:Nobody:/:/bin/sh
system:x:430:430:service:/var/run/connman:/bin/sh
avahi:x:495:495:avahi-daemon:/var/run/avahi-daemon:/bin/sh
===

cit.: "_... working with the mc developers/strace/gdb may resolve this. Alternatively rolling back to your working version_"
for the last part of the sentence: it's the only thing I can do here !
for the first part:
when I begin to tell the mc developers the breakage went in by updating LE **(!)** SystemTools Releases 11.80.2.2 to 11.80.2.3 
I'm sure, they will tell me: Yup, sounds like an LE problem, esp. cause the mc version hasn't changed, IIRC. 
Please open a bug report there (=> LE) ...
No ?
===
please verify if this bug should be re-opened by setting up a free box, as I did, and again test mc  (on new install it should be SystemTools Version 11.80.3.0 I guess, what is also NOT working (here))

Do not test with your devel box, as above.

after that and esp. if you (or someone others running a clean and new nightly with Systemtools) can't reproduce my bug with mc,
I'll go the mc-developer line.

otherwise if this bug remains closed you could also close all my other open bugs !
Advantage for you: you don't need to deal with *my* bugs anymore.
Disadvantage: you might be flooded with my mc bugs and some more currently buried in nightly when LE12 nightly becomes LE12 Release.

**my Thoughts are** 
Why should I test nightly and report bugs, when they're getting closed immediately after I opened them 
OR 
for the kodi developers: see that fixes (code against flickering from sarbes) are getting moved from one milestone to the next and junked (my view) afterwards !

I guess it has been noticed that english is not my native language, so reporting bugs is HARD WORK here, too ...

tnmeyer, 2023-10-07:
I can reproduce this bug as well.
It was triggered with the transition from pcre to pcre2. Root cause seems to be an improperly configured pcre2.
"...-DPCRE2_SUPPORT_JIT=ON..." is needed in the pcre2 package.mk for mc to work correctly.
Or, as an alternative, disable pcre support in mc completely (use --with-search-engine=glib instead for example).

heitbaum, 2023-10-08:
> I can reproduce this bug as well. It was triggered with the transition from pcre to pcre2. Root cause seems to be an improperly configured pcre2. "...-DPCRE2_SUPPORT_JIT=ON..." is needed in the [...]

Thanks @tnmeyer for getting to the bottom of this. I have included it in a system-tools (1).

closed 2023-10-08 by CvH