LibreELEC/LibreELEC.tv #8148: [BUG] Forum No way to create an new post no response from Admin Contact

Pau563, 2023-09-15:
### Describe the bug
Created a an account with the LibreElec.TV Forum and I can log in and view but there is No way to create an new post, no response from Admin Contact, two attempts. 

I have gone all over the pages and cannot find anything as there is also no reply buttons so I can't even ask a existing member. 

### How to reproduce
Create an account, you will then be told you will receive a confirmation email. None received but was logged automatically. I have also logged in  several times of the last few days. 

Try to make a new post/question and there are not tabs to do so. There are other tabs for things as filter and sort but not to create. 

### Information
 LibreElec Forum

heitbaum, 2023-09-15:
@Pau563 - can you please share your forum username?

chewitt, 2023-09-15:
Please email `contact@libreelec.tv` from the same email address used to register the forum account and we can manually enable the account.

closed 2023-09-15 by chewitt