LibreELEC/LibreELEC.tv #8207: Widevine 4.10.2662.3 fails to load

Dapsy2000, 2023-10-13:
### Describe the bug
My video addons like Prime Video or RTL+ stopped playing video a few days ago.
For this reason I did an update for all installed addons including InputStream Adaptive but still the same issue.

I opened an issue on https://github.com/xbmc/inputstream.adaptive/issues/1398 and I got a comment that the issue is LibreELEC related with the advise to linked the issue. They also provided some information about GLIBC_ABI_DT_RELR: https://github.com/xbmc/inputstream.adaptive/wiki/Widevine-ARM64-support

### How to reproduce
Steps to reproduce the behavior:

LibreELEC:~ # ldd /storage/.kodi/cdm/libwidevinecdm.so
**/storage/.kodi/cdm/libwidevinecdm.so: error while loading shared libraries: /storage/.kodi/cdm/libwidevinecdm.so: DT_RELR without GLIBC_ABI_DT_RELR dependency**

or

Start Playback on Prime Video (https://github.com/Sandmann79/xbmc) 
or RTL+ Addon (https://github.com/Collabsvito/REpo-AiO/tree/master/MATRIX/plugin.video.rtlgroup.de)

### Information
 - LibreELEC Version: 11.0.3 with Kodi v20.2 (Nexus)
 - Hardware Platform: RPi3
 - Info about Widevine CDM / InputStream Helper /  InputStream Adaptive:
 
   2023-10-13 15:27:25.809 T:1100 warning : [script.module.inputstreamhelper]
   Kodi version 20.2 is running on a Linux system with arm architecture.
   
   InputStream Helper is at version 0.6.1+matrix.1
   InputStream Adaptive is at version 20.3.13.1
   
   Widevine CDM is at version 4.10.2662.3 and was installed on 2023-07-29 17:41
   It was extracted from Chrome OS image ELM with version 15572.57.0
   It was last checked for updates on 2023-07-29 17:42
   It is installed at /storage/.kodi/cdm/
   
   Please report issues to: https://git.io/JfKJb

### Log file

Start Playback on Prime Video

2023-10-13 15:25:44.389 T:1073 critical : [script.module.inputstreamhelper] **['ldd', '/storage/.kodi/cdm/libwidevinecdm.so'] cmd failed.**
2023-10-13 15:25:44.389 T:1073 critical : [script.module.inputstreamhelper] **Failed to check for missing Widevine libraries.**
2023-10-13 15:25:44.389 T:1073 warning : [script.module.inputstreamhelper] User had declined an update on 2023-10-11 16:56

more see: https://github.com/xbmc/inputstream.adaptive/issues/1398

### Context

lrusak, 2023-10-13:
Are you using the right widevine lib? Don't you just need the arm version not the aarch64 version?

Dapsy2000, 2023-10-13:
> Are you using the right widevine lib? Don't you just need the arm version not the aarch64 version?

Yes, I‘m using the right widevine lib because I updated all addons including inputstreamhelper via WebGUI on RPi LibreELEC 11.0.3 with Kodi v20.2 (Nexus).

lrusak, 2023-10-13:
> > Are you using the right widevine lib? Don't you just need the arm version not the aarch64 version?
> 
> Yes, I‘m using the right widevine lib because I updated all addons including inputstreamhelper via WebGUI on RPi LibreELEC 11.0.3 with Kodi v20.2 (Nexus).

You shouldn't need the chrome OS widevine lib then IIRC. I would try removing widevine and have the widevine helper redownload the lib.

HiassofT, 2023-10-13:
I can't reproduce that here with a fresh install of ORF TVThek addon from kodi repo and inputstream.adaptive 20.3.13.1 on RPi3B+ / LE 11.0.3

Please use the forum for support provide a full debug log, the snippet you quoted is a harmless, misleading bug in inputstreamhelper and more likely  the amazon prime addon is failing again

closed 2023-10-13 by HiassofT

leecher1337, 2024-06-15:
https://thebrokenrail.com/2022/12/31/xfinity-stream-on-linux.html