LibreELEC/LibreELEC.tv #8208: [BUG] AMD GPU display flicker

TianZerL, 2023-10-13:
### Describe the bug

LE12 with AMD GPU will cause flicker, see photo in `Context`.

I have tested with lots of nightly builds (Generic.x86_64, date from 9.2 to 10.10), all got same issue.

I also tested with tow different computers, but no luck:
- Laptop with Ryzen 3500u and Vage8, which has a built-in 1080p 60 hz screen.
- Desktop with i7 8700K and 5700XT, which conneted to a 1080p 144hz screen via DP.

In the desktop, I tried change resolution and fresh rate, it can suppress the flicker somtimes but reappears if I go to another page (eg. from settings to main page).

Not sure if it is an AMD GPU only issue, I don't have an Intel GPU machine to test.

### How to reproduce
Steps to reproduce the behavior:
1. Boot a LE12 Generic.x86_64 nightly build with an AMD GPU. (I use live mode)

### Information
 - LibreELEC Version: 12
 - Hardware Platform: X86

### Log file

http://ix.io/4IV3
### Context

[image]

TianZerL, 2023-10-14:
@lrusak 

> Might be able to use the env variable `mesa_glthread=false`

Thanks for reply. I can confirm it works.

> related https://gitlab.freedesktop.org/mesa/mesa/-/issues/7977

This should be fixed in mesa 22.3.5, but it doesn't seem to be in branch 23.

Is there a way that I can set `mesa_glthread=false` before installing?
I tried add a `.config/kodi.conf` file in my USB media's `storage` partition, but it is not work.

lrusak, 2023-10-14:
> @lrusak
>
>
>
>
>
>
> Thanks for reply. I can confirm it works.
>
>
>
>
>
>
> This should be fixed in mesa 22.3.5, but it doesn't seem to be in branch 23.
>
>
>
> Is there a way that I can set `mesa_glthread=false` [...]

Not really. We should probably pull the fix into our tree or set the env variable in the build.

mglae, 2023-10-14:
> Boot a LE12 Generic.x86_64 nightly build with an AMD GPU. (I use live mode)

> Is there a way that I can set mesa_glthread=false before installing?
> I tried add a .config/kodi.conf file in my USB media's storage partition, but it is not work.

Use "run" mode - but start with unmodified storage partition.

RobGries, 2023-11-08:
I am also having the same trouble on the latest nightly build on an HP EliteDesk 705 G4 (Ryzen 5 PRO 2400GE APU with Vega 11 Graphics) .

I am trying to set the `mesa_glthread=false` environment variable, but I am afraid that it is not affecting the problem with flicker/graphics corruption. I tried setting it as an environment variable with `export mesa_glthread=false` and restarting kodi from ssh with `systemctl restart kodi` but that did nothing. I feel that I am missing something, perhaps it isn't a normal envvar?

heitbaum, 2024-03-11:
Is this issue still occurring with latest LE12 20240311-86dba4b images?

dod38fr, 2024-03-16:
This LE12 20240311-86dba4b image is working fine on my AMD system (with or without `export mesa_glthread=false`).

heitbaum, 2024-08-17:
Retested on LE13 nightly (no changes to mesa_glthread required)
```
=== tested on ===
Generic.x86_64-devel-20240815104328-b569b7a
Linux ser7 6.11.0-rc3 #1 SMP Thu Aug 15 09:45:31 UTC 2024 x86_64 GNU/Linux
Starting Kodi (22.0-ALPHA1 (21.90.700) Git:f02afef0f39f8a3770d0e9fd5483e8a53a1814d7). Platform: Linux x86 64-bit
Using Release Kodi x64
Kodi compiled 2024-08-14 by GCC 14.2.0 for Linux x86 64-bit version 6.11.0 (396032)
Running on LibreELEC (heitbaum): devel-20240815104328-b569b7a 13.0, kernel: Linux x86 64-bit version 6.11.0-rc3
FFmpeg version/source: 6.0.1
Host CPU: AMD Ryzen 7 7840HS w/ Radeon 780M Graphics     , 16 cores available
[    0.000000] DMI: AZW SER/SER, BIOS SER7PRO_P5C8V28 08/14/2023
[    0.000000] DMI: Memory slots populated: 2/2
CApplication::CreateGUI - using the gbm windowing system
EGL_VENDOR = Mesa Project
GL_VERSION = OpenGL ES 3.2 Mesa 24.2.0
libva info: VA-API version 1.22.0
libva info: Trying to open /usr/lib/dri/radeonsi_drv_video.so
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: Mesa Gallium driver 24.2.0 for AMD Radeon 780M (radeonsi, gfx1103_r1, LLVM 18.1.8, DRM 3.58, 6.11.0-rc3)
```

closed 2024-08-17 by heitbaum