LibreELEC/LibreELEC.tv #8217: [BUG] Regular crashes since LE 11

bverhagen, 2023-10-14:
### Describe the bug

Since I upgraded from LE 10 to LE 11 in March, I have been having regular Kodi crashes (almost daily). Three patch upgrades later, it still is crashing in the same way, so I (finally) started to debug a bit myself. The LE11 setup is exactly the same as it was for LE 10 before (same add-ons, upgraded to newer Kodi), where it was running very stable.

The crashes never occur during playback, but always seem to occur when pushing the 'back' button in the Kodi interface (I do this using a Logitech Harmoney Hub connected over bluetooth to the raspberry pi 4 LE is running on). I got the impression it happens the very first time I push 'back' after long periods of stand-by. After the automatic restart, everything works fine and as expected.

At first I thought it was the entire device crashing. However, during debugging, I noticed how my SSH session remained open when a crash occurred. The uptime of the device confirms it is not the entire device crashing and restarting.

This let me to the Kodi crash logs themselves, where there are indeed daily crashlogs. The most recent one is attached to this post. The full crash log is found below, but every one of them seems to crash with the same stacktrace:

```

Using host libthread_db library "/usr/lib/libthread_db.so.1".
Core was generated by `/usr/lib/kodi/kodi.bin --standalone -fs'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0xf7537614 in PyEval_ReleaseThread () from /usr/lib/libpython3.11.so.1.0
[Current thread is 1 (Thread 0xcc6ff280 (LWP 30254))]
```

I do not find any similar reports online. It appears to be related to the upgrade to Python 3.11 that came with LE 11.

Is this a known problem? What can I do to further debug this? Should I report this further upstream at the kodi development Github?

### How to reproduce
Steps to reproduce the behavior:
1. Wait until Kodi goes to standby
2. Switch on TV etc.
3. Wake up kodi by pushing a button (the interface becomes no longer dimmed)
4. Push the 'back button'
5. Nothing happens for multiple seconds, then the Libreelect logo appears
6. A restarted Kodi appears

### Information
 - LibreELEC Version: 11.0.3 (had it also in 11.0 and 11.0.1)
 - Hardware Platform: RPi 4

### Log file

[kodi_crashlog_20231014213055.log](https://github.com/LibreELEC/LibreELEC.tv/files/12907945/kodi_crashlog_20231014213055.log)

### Context

See the description of the bug

heitbaum, 2023-10-16:
Please use forum for support. Many more people are on the forum. If a confirmed bug. Then happy to reopen or have a new issue created.

closed 2023-10-16 by heitbaum

bverhagen, 2023-10-16:
With 'confirmed' you mean you need other users need to complain about this too on the forums?

Other than that, it is obviously a confirmed bug: Kodi (and by extension libreElec) should not crash, even when plugins do the weirdest things (which I do think is unlikely the case here, as it happens everywhere in Kodi and was stable for LE 10). The proof Kodi crashes you have with the crashlog I provided.

heitbaum, 2023-10-16:
Hi @bverhagen - the crash is indeed happening. But without a why - this report will not be able to get a fix. The forum is setup to dig into the problem and help try and fix it. It does not need others to experience the same “bug”. First thing the forum will ask for is a full set of debug logs.