LibreELEC/LibreELEC.tv #8223: Bluetooth audio stop after couple minutes on rpi4

iamcoolj83, 2023-10-17:
### Describe the bug
When playing Mpeg-Ts files (AC3 audio) (Tv records) Bluetooth driver crash. Video and audio start, play correctly then after some minutes you lose audio, then after one minute of no audio headphones return in pairing mode (bluetooth are unresponsive on the Rpi4, tried onboard BT, two different Bt dongle, different headphones and issue remain the same, only with mpeg-ts. It seem ok with DTS or AAC audio on my mkv files (h264 and h265) I have listened several movies and they were ok, but as soon I watch tv records (mpeg-ts Ac3 audio) it crash. Sink timer expire, but It doesn't seem to be a sync issue, the driver/hardware get unresponsive. The same file will play just fine on Le 9.2.8. 

### How to reproduce
Steps to reproduce the behavior:
Play an mpeg-ts ac3 on bluetooth headphones, wait until bluetooth crash

### Information
 - LibreELEC Version: 10.0.4 to 11.0.3.,   9.2.8 are ok (I use this one until things work on latest), didn't tried 12
 - Hardware Platform: RPi4

### Log file

[Log](https://pastebin.com/DizbvxVA)

### Context

Then pastekodi doesn't work on putty on 11.0.3 why?? didn't get a link.

iamcoolj83, 2023-10-17:
Added infos: The mediainfo from a source video example.
[Mediainfo](https://pastebin.com/DuMzw39a)

heitbaum, 2023-10-25:
There have been a number of Bluetooth changes in LE12, and other significant changes (64bit userland), Alsa, pulseaudio, pipewire with aac. So would suggest you try LE12 out and if still an issue - then it can be looked at.

iamcoolj83, 2023-10-26:
Yes I will do and let you know soon.

iamcoolj83, 2023-10-31:
Fixed in Le12 work like a charm

iamcoolj83, 2023-10-31:
Still the same with Le12, seems to be only with one specific channel that I record tv from. Will need more testing.

heitbaum, 2024-03-11:
Is this issue still occurring with latest LE12 20240311-86dba4b images?

heitbaum, 2024-03-16:
Closing as inactive. Happy to open again - but will need debug logs, ideally a reproducible case. Using the forum for support may get more answers as others may have seen the same issue.

closed 2024-03-16 by heitbaum