LibreELEC/LibreELEC.tv #8255: [BUG] Pirewire/Wireplumber build failure

smp79, 2023-10-27:
### Describe the bug

> Program /usr/bin/gdbus-codegen found: NO

It tries to find it on a host OS.

mglae, 2023-10-28:
Similar issue when building gtk3 for chrome. Try:

```
Subject: [PATCH] glib: binaries must be executed from toolchain

---
 packages/devel/glib/package.mk | 3 +++
 1 file changed, 3 insertions(+)

diff --git a/packages/devel/glib/package.mk b/packages/devel/glib/package.mk
index 6428c2ec08..2441d7293d 100644
--- a/packages/devel/glib/package.mk
+++ b/packages/devel/glib/package.mk
@@ -35,4 +35,7 @@ post_makeinstall_target() {
   rm -rf ${INSTALL}/usr/lib/glib-2.0
   rm -rf ${INSTALL}/usr/lib/installed-tests
   rm -rf ${INSTALL}/usr/share
+
+  # glib binaries must be executed from toolchain
+  sed -e "s#bindir=\${prefix}/bin#bindir=${TOOLCHAIN}/bin#" -i "${SYSROOT_PREFIX}/usr/lib/pkgconfig/"{gio,glib}-2.0.pc
 }
```

heitbaum, 2023-10-28:
glib:host is called out as a dependency in wireplumber. I have added this to pipewire.

heitbaum, 2023-10-28:
There is something else. Not sure yet but something about meson 1.2 or glib 2.78. glib 2.77.3 and meson 1.1.1 with the draft patch work.

heitbaum, 2023-10-28:
If you apply the dependency patch and revert meson to 1.2.1 (I think it works) - `git revert 1768c1f00d edd8fcea5d`. I need to do some further clean builds to make sure though. Then to take a look at why meson 1.2.2 caused a change/regression.

heitbaum, 2023-10-28:
The issue (change of behavior) is https://github.com/mesonbuild/meson/commit/f0b2ab8bb02f282499c6d0e5b9f6e5d55bb1419c - meson now uses the .pc files for gio-2.0, glib-2.0, gobject-introspection-1.0. Updated the pr with your @mglae fix, and fixed a couple of dependancies that I noticed burning build testing. I tested gtk3 on Generic-legacy [Fixed], and pipewire of Generic and various aarch64 - so cross compiling is good.

closed 2023-10-29 by mglae