LibreELEC/LibreELEC.tv #8271: [BUG] Wayland build fails

heitbaum, 2023-11-01:
### Describe the bug

Since https://github.com/xbmc/xbmc/pull/23612 the Wayland build fails
- https://github.com/xbmc/xbmc/commit/f8573a7247e41cb5c1c7d738a898f5a88c40d348
```
CMake Error at cmake/scripts/linux/ExtraTargets.cmake:39 (add_dependencies):
  Cannot add target-level dependencies to non-existent target "libkodi".

  The add_dependencies works for top-level logical targets created by the
  add_executable, add_library, or add_custom_target commands.  If you want to
  add file-level dependencies see the DEPENDS option of the add_custom_target
  and add_custom_command commands.
Call Stack (most recent call first):
  CMakeLists.txt:581 (include)
```

### How to reproduce
Steps to reproduce the behavior:
1. `PROJECT=Generic ARCH=x86_64 DEVICE=wayland make image`

### Information
 - LibreELEC Version: [e.g. 9.2.1] LE12
 - Hardware Platform: [e.g. RPi3] wayland 

### Log file

### Context

heitbaum, 2024-10-03:
Further info 
`PROJECT=Generic ARCH=x86_64 DEVICE=wayland s/build kodi:host` fails
But 
`PROJECT=Generic ARCH=x86_64 DEVICE=wayland s/build kodi:target` doesn’t fail

The host build can be made to succeed by commenting the following line (2nd last) in the file `cmake/scripts/linux/ExtraTargets.cmake`

`add_dependencies(lib${APP_NAME_LC} generate-wayland-extra-protocols)`

then removing the comment before executing the :target build.

jernejsk, 2024-10-03:
I guess it's time to update our hack patch for quick addon builds. I'll look into that.

PR #9357 (2024-10-05): Kodi wayland build fixes
Refresh Kodi headers only patch to fix #8271 and promote libxkbcommon to common kodi target dependency so it can be build separately.
closed 2024-10-05 by heitbaum