LibreELEC/LibreELEC.tv #8278: [BUG] curl - 429 Unknown Error

mvrk69, 2023-11-02:
### Describe the bug
Hi,
I have an iptv list from a tv channel of my country that when try to access with curl from libreelec on rpi4 it gives error 429, but for example from my laptop with Fedora 37 downloads fine.

### How to reproduce
Steps to reproduce the behavior:
RPi4:
curl -A "Firefox" https://streaming-live-app.rtp.pt/liverepeater/smil:rtp1HD.smil/playlist.m3u8 -o playlist.m3u8

cat playlist.m3u8
<?xml version="1.0" encoding="utf-8"?>
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Strict//EN"
 "http://www.w3.org/TR/xhtml1/DTD/xhtml1-strict.dtd">
<html>
  <head>
    <title>429 Unknown Error</title>
  </head>
  <body>
    <h1>Error 429 Unknown Error</h1>
    <p>Unknown Error</p>
    <h3>Error 54113</h3>
    <p>Details: cache-vie6339-VIE 1698960209 1876796552</p>
    <hr>
    <p>Varnish cache server</p>
  </body>
</html>

-----------------------------------
Laptop - Fedora 37 (x86_64):
curl -A "Firefox" https://streaming-live-app.rtp.pt/liverepeater/smil:rtp1HD.smil/playlist.m3u8 -o playlist.m3u8

cat playlist.m3u8
#EXTM3U
#EXT-X-VERSION:5
#EXT-X-STREAM-INF:BANDWIDTH=2540000
chunklist_b2540000_slpor.m3u8
#EXT-X-STREAM-INF:BANDWIDTH=1290000
chunklist_b1290000_slpor.m3u8
#EXT-X-STREAM-INF:BANDWIDTH=640000
chunklist_b640000_slpor.m3u8

### Information
 - LibreELEC Version: 11.0.3
 - Hardware Platform:  RPi4

eldersixpack13, 2023-11-02:
same here on LE12 Generic (nightly-20231101-e7215dd)

heitbaum, 2023-11-03:
I have had this before, with Hue - https://github.com/zim514/script.service.hue/pull/251 

This error is being issued by the site or the site’s CDN - as it shows the “cache” test in the result -
This CF article discusses it.
https://community.cloudflare.com/t/429-too-many-requests-when-using-curl-or-proxy/397106

mvrk69, 2023-11-03:
question is what does the curl on libreelec differently from the curl on my fedora linux laptop that triggers that too many requests error from the site.

heitbaum, 2023-11-03:
> question is what does the curl on libreelec differently from the curl on my fedora linux laptop that triggers that too many requests error from the site. It will “probably” be a difference in the >?

You could check the headers and see what is the difference. If it is coming from the same up address, it must be something that the 2 versions of curl are doing differently. Here is the output from my LE12 dev system.

```
# curl --version
curl 8.4.0 (x86_64-libreelec-linux-gnu) libcurl/8.4.0 OpenSSL/3.2.0 zlib/1.3 libidn2/2.3.4 nghttp2/1.58.0 librtmp/2.3
Release-Date: 2023-10-11
Protocols: file ftp ftps http https rtmp rtsp
Features: alt-svc AsynchDNS HSTS HTTP2 HTTPS-proxy IDN IPv6 Largefile libz NTLM SSL threadsafe TLS-SRP UnixSockets

# curl -v -A "Firefox" https://streaming-live-app.rtp.pt/liverepeater/smil:rtp1HD.smil/playlist.m3u8 -o playlist.m3u8
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying 151.101.82.192:443...
* Connected to streaming-live-app.rtp.pt (151.101.82.192) port 443
* ALPN: curl offers h2,http/1.1
} [5 bytes data]
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
} [512 bytes data]
*  CAfile: /run/libreelec/cacert.pem
*  CApath: none
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Server hello (2):
{ [122 bytes data]
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
{ [19 bytes data]
* TLSv1.3 (IN), TLS handshake, Certificate (11):
{ [2781 bytes data]
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
{ [264 bytes data]
* TLSv1.3 (IN), TLS handshake, Finished (20):
{ [36 bytes data]
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
} [1 bytes data]
* TLSv1.3 (OUT), TLS handshake, Finished (20):
} [36 bytes data]
* SSL connection using TLSv1.3 / TLS_AES_128_GCM_SHA256
* ALPN: server accepted h2
* Server certificate:
*  subject: CN=*.rtp.pt
*  start date: Sep  7 16:33:16 2023 GMT
*  expire date: Oct  8 16:33:15 2024 GMT
*  subjectAltName: host "streaming-live-app.rtp.pt" matched cert's "*.rtp.pt"
*  issuer: C=BE; O=GlobalSign nv-sa; CN=AlphaSSL CA - SHA256 - G4
*  SSL certificate verify ok.
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
{ [201 bytes data]
* using HTTP/2
* [HTTP/2] [1] OPENED stream for https://streaming-live-app.rtp.pt/liverepeater/smil:rtp1HD.smil/playlist.m3u8
* [HTTP/2] [1] [:method: GET]
* [HTTP/2] [1] [:scheme: https]
* [HTTP/2] [1] [:authority: streaming-live-app.rtp.pt]
* [HTTP/2] [1] [:path: /liverepeater/smil:rtp1HD.smil/playlist.m3u8]
* [HTTP/2] [1] [user-agent: Firefox]
* [HTTP/2] [1] [accept: */*]
} [5 bytes data]
> GET /liverepeater/smil:rtp1HD.smil/playlist.m3u8 HTTP/2
> Host: streaming-live-app.rtp.pt
> User-Agent: Firefox
> Accept: */*
> 
{ [5 bytes data]
< HTTP/2 429 
< retry-after: 0
< content-type: text/html; charset=utf-8
< accept-ranges: bytes
< date: Fri, 03 Nov 2023 09:32:09 GMT
< via: 1.1 varnish
< x-served-by: cache-mel11264-MEL
< x-cache: MISS
< x-cache-hits: 0
< x-timer: S1699003930.931689,VS0,VE0
< req: /liverepeater/smil:rtp1HD.smil/playlist.m3u8
< x-ip: 122.199.31.3
< access-control-allow-credentials: true
< alt-svc: h3=":443";ma=86400,h3-29=":443";ma=86400,h3-27=":443";ma=86400
< content-length: 430
< 
{ [5 bytes data]
100   430  100   430    0     0  27544      0 --:--:-- --:--:-- --:--:-- 28666
* Connection #0 to host streaming-live-app.rtp.pt left intact
```

mvrk69, 2023-11-03:
RPI4 (LibreELEC):

rpi4:~ # curl -v -A "Firefox" https://streaming-live-app.rtp.pt/liverepeater/smil:rtp1HD.smil/playlist.m3u8 -o playlist.m3u8
```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying 151.101.130.192:443...
* Connected to streaming-live-app.rtp.pt (151.101.130.192) port 443 (#0)
* ALPN: offers h2
* ALPN: offers http/1.1
*  CAfile: /run/libreelec/cacert.pem
*  CApath: none
* TLSv1.0 (OUT), TLS header, Certificate Status (22):
} [5 bytes data]
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
} [512 bytes data]
* TLSv1.2 (IN), TLS header, Certificate Status (22):
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Server hello (2):
{ [122 bytes data]
* TLSv1.2 (IN), TLS header, Finished (20):
{ [5 bytes data]
* TLSv1.2 (IN), TLS header, Supplemental data (23):
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
{ [19 bytes data]
* TLSv1.2 (IN), TLS header, Supplemental data (23):
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Certificate (11):
{ [2781 bytes data]
* TLSv1.2 (IN), TLS header, Supplemental data (23):
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
{ [264 bytes data]
* TLSv1.2 (IN), TLS header, Supplemental data (23):
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Finished (20):
{ [36 bytes data]
* TLSv1.2 (OUT), TLS header, Finished (20):
} [5 bytes data]
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
} [1 bytes data]
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
} [5 bytes data]
* TLSv1.3 (OUT), TLS handshake, Finished (20):
} [36 bytes data]
* SSL connection using TLSv1.3 / TLS_AES_128_GCM_SHA256
* ALPN: server accepted h2
* Server certificate:
*  subject: CN=*.rtp.pt
*  start date: Sep  7 16:33:16 2023 GMT
*  expire date: Oct  8 16:33:15 2024 GMT
*  subjectAltName: host "streaming-live-app.rtp.pt" matched cert's "*.rtp.pt"
*  issuer: C=BE; O=GlobalSign nv-sa; CN=AlphaSSL CA - SHA256 - G4
*  SSL certificate verify ok.
* Using HTTP2, server supports multiplexing
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
} [5 bytes data]
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
} [5 bytes data]
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
} [5 bytes data]
* h2h3 [:method: GET]
* h2h3 [:path: /liverepeater/smil:rtp1HD.smil/playlist.m3u8]
* h2h3 [:scheme: https]
* h2h3 [:authority: streaming-live-app.rtp.pt]
* h2h3 [user-agent: Firefox]
* h2h3 [accept: */*]
* Using Stream ID: 1 (easy handle 0x1a0a740)
[... 132 lines cut ...]
< cache-control: max-age=2
< access-control-allow-origin: *
< access-control-allow-methods: OPTIONS, GET, POST, HEAD
< access-control-allow-headers: Content-Type, User-Agent, If-Modified-Since, Cache-Control, Range
< x-source: w2
< x-stream-type: video
< x-node: n1
< via: 1.1 varnish, 1.1 varnish
< accept-ranges: bytes
< date: Fri, 03 Nov 2023 09:57:39 GMT
< age: 0
< x-served-by: cache-lis1490043-LIS, cache-vie6377-VIE
< x-cache: BYPASS, MISS, MISS
< x-cache-hits: 0, 0
< x-timer: S1699005459.455012,VS0,VE56
< req: /liverepeater/smil:rtp1HD.smil/playlist.m3u8
< x-ip: 188.95.123.238
< access-control-allow-credentials: true
< alt-svc: h3=":443";ma=86400,h3-29=":443";ma=86400,h3-27=":443";ma=86400
< content-length: 221
< 
{ [221 bytes data]
100   221  100   221    0     0   1340      0 --:--:-- --:--:-- --:--:--  1347
* Connection #0 to host streaming-live-app.rtp.pt left intact
```

heitbaum, 2023-11-03:
Nothing obvious. Only 1 TLS header different. What are the curl versions?

```
diff -Nu bh bh1
--- bh  2023-11-03 10:43:06.591129460 +0000
+++ bh1 2023-11-03 10:47:22.703010559 +0000
@@ -1,7 +1,7 @@
 * Connected to streaming-live-app.rtp.pt (151.101.130.192) port 443 (#0)
 * ALPN: offers h2
 * ALPN: offers http/1.1
-*  CAfile: /run/libreelec/cacert.pem
+*  CAfile: /etc/pki/tls/certs/ca-bundle.crt
 *  CApath: none
 * TLSv1.0 (OUT), TLS header, Certificate Status (22):
 } [5 bytes data]
@@ -60,14 +60,14 @@
 * h2h3 [:authority: streaming-live-app.rtp.pt]
 * h2h3 [user-agent: Firefox]
 * h2h3 [accept: */*]
-* Using Stream ID: 1 (easy handle 0x1a0a740)
+* Using Stream ID: 1 (easy handle 0x556d65b8f5a0)
 * TLSv1.2 (OUT), TLS header, Supplemental data (23):
 } [5 bytes data]
 > GET /liverepeater/smil:rtp1HD.smil/playlist.m3u8 HTTP/2
 > Host: streaming-live-app.rtp.pt
 > user-agent: Firefox
 > accept: */*
->
+> 
 * TLSv1.2 (IN), TLS header, Supplemental data (23):
 { [5 bytes data]
 * TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
@@ -78,24 +78,30 @@
 } [5 bytes data]
 * TLSv1.2 (IN), TLS header, Supplemental data (23):
 { [5 bytes data]
-* TLSv1.2 (IN), TLS header, Supplemental data (23):
-{ [5 bytes data]
-< HTTP/2 429
-< retry-after: 0
-< content-type: text/html; charset=utf-8
+< HTTP/2 200 
+< content-type: application/vnd.apple.mpegurl
+< access-control-expose-headers: Date, Server, Content-Type, Content-Length
+< cache-control: max-age=2
+< access-control-allow-origin: *
+< access-control-allow-methods: OPTIONS, GET, POST, HEAD
+< access-control-allow-headers: Content-Type, User-Agent, If-Modified-Since, Cache-Control, Range
+< x-source: w2
+< x-stream-type: video
+< x-node: n1
+< via: 1.1 varnish, 1.1 varnish
 < accept-ranges: bytes
-< date: Fri, 03 Nov 2023 10:27:19 GMT
-< via: 1.1 varnish
-< x-served-by: cache-vie6360-VIE
-< x-cache: MISS
-< x-cache-hits: 0
-< x-timer: S1699007240.933341,VS0,VE0
+< date: Fri, 03 Nov 2023 09:57:39 GMT
+< age: 0
+< x-served-by: cache-lis1490043-LIS, cache-vie6377-VIE
+< x-cache: BYPASS, MISS, MISS
+< x-cache-hits: 0, 0
+< x-timer: S1699005459.455012,VS0,VE56
 < req: /liverepeater/smil:rtp1HD.smil/playlist.m3u8
 < x-ip: 188.95.123.238
 < access-control-allow-credentials: true
 < alt-svc: h3=":443";ma=86400,h3-29=":443";ma=86400,h3-27=":443";ma=86400
-< content-length: 429
-<
-{ [429 bytes data]
-100   429  100   429    0     0   2689      0 --:--:-- --:--:-- --:--:--  2698
+< content-length: 221
+< 
+{ [221 bytes data]
+100   221  100   221    0     0   1340      0 --:--:-- --:--:-- --:--:--  1347
 * Connection #0 to host streaming-live-app.rtp.pt left intact
```

mvrk69, 2023-11-03:
In any case, digging into their website, i saw that now they are using streaming-live.rtp.pt instead of streaming-live-app.rtp.pt, and with that one works fine on rpi4.

rpi4:~ # curl -A "Firefox" https://streaming-live.rtp.pt/liverepeater/smil:rtp1HD.smil/playlist.m3u8 -o playlist.m3u8
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   221  100   221    0     0    548      0 --:--:-- --:--:-- --:--:--   548

rpi4:~ # cat playlist.m3u8
#EXTM3U
#EXT-X-VERSION:5
#EXT-X-STREAM-INF:BANDWIDTH=2540000
chunklist_b2540000_slpor.m3u8
#EXT-X-STREAM-INF:BANDWIDTH=1290000
chunklist_b1290000_slpor.m3u8
#EXT-X-STREAM-INF:BANDWIDTH=640000
chunklist_b640000_slpor.m3u8

closed 2023-11-04 by heitbaum