LibreELEC/LibreELEC.tv #8322: [BUG] Intel NUC 13 Pro doesn't switch to HDR with my Sony A95L[BUG]

AboDa7eM1, 2023-11-12:
### Describe the bug
Hi I have an Intel NUC 13 Pro with Core i7 1360P connected to the Sony A95L.
on LibreELEC-Generic.x86_64-12.0-nightly-20231108

When trying to run any HDR Video the display dose not switch to HDR mode. Debug log attached for more information.

I hope if you can tell me where the issue is.

### How to reproduce
Steps to reproduce the behavior:
Run any HDR File with Intel NUC 13

### Information
 - LibreELEC Version: [Generic.x86_64-12.0-nightly-20231108]
 - Hardware Platform: [Intel NUC 13]

### Log file
https://paste.kodi.tv/tefayidini

lrusak, 2023-11-12:
The log file doesn't indicate that it even tried to switch to HDR. This is likely because the hardware doesn't support it, the Linux driver doesn't support HDR for the hardware, or the file doesn't contain a valid HDR type that the driver supports.

On Linux only HLG and HDR10 are supported.

closed 2023-11-12 by lrusak

smp79, 2023-11-13:
It's a 60p video. For some reason only 24p and only HEVC codec would switch to HDR. @lrusak is this something you are aware of? When I did my custom LE HDR builds from your [repo](https://github.com/lrusak/xbmc/commits/drmprime-2img-no-ffmpeg-bump) - the earlier iterations of that repo did not have such limitations, eg 50p/60p videos did pass the HDR to the TV, HLG videos also switched the TV to HLG HDR mode. This is not the case with the current Kodi master.

smp79, 2023-11-13:
60p videos switched to HDR on Intel when DRM PRIME was used (with older lrusak repo). With current Kodi master DRM PRIME doesn't do much on Intel (no HW acceleration, no HDR).

lrusak, 2023-11-13:
> It's a 60p video. For some reason only 24p and only HEVC codec would switch to HDR. @lrusak is this something you are aware of? When I did my custom LE HDR builds from your [...]

We don't care about the refresh rate or the codec.

We do check the edid which may only offer HDR at specific modes. In the log posted it shows @60fps both pq and hlg are supported.

There is some limitations based on the HDR metadata though where some media might not contain complete info.

I've only tested HDR on a NUC7 Gemini lake model with native hdmi 2.0 support (no adapter board).

AboDa7eM1, 2023-11-14:
Why did close it? you guys did not solve anything???
I am just reporting this to let you guys of my experience and find solution or at least the route couse.

Okay I found different fork of OpenELEC called AlexELEC and HDR worked fine with 60Hz and 24Hz files
https://alexelec.tv/alexelec-pc-11-2-3-x86_64-gbm-drm-prime/

here a log for the same file I tested in LibreELEC "Samsung 4K HDR 60p"
https://paste.kodi.tv/omofehoquh

another one with 24p file
https://paste.kodi.tv/pigijoqare

smp79, 2023-11-17:
> There is some limitations based on the HDR metadata though where some media might not contain complete info.

You are right, this is not a 60p issue. I re-encoded the sample 60p video with x265 and with the re-encoded video TV switches to an HDR mode. 

[Re-encoded sample](https://www.dropbox.com/scl/fi/0lvccew7j65xh3juepd9o/HDR-60p.mkv?rlkey=mggcywcqgt3bbbyrqj3ai8lj2&dl=1) (TV switch to HDR mode)
[Source sample](https://www.dropbox.com/scl/fi/b5387t5i1xvokivg85uy8/HDR-60p-original.mkv?rlkey=apmvojen97r96rdhpl8vdom67&dl=1) (TV does not switch to HDR mode)