LibreELEC/LibreELEC.tv #8323: [BUG]

DDQW, 2023-11-13:
### Describe the bug
Flickering and color"pumping" (changes between saturated and unsaturated colors) while using the netflix plugin. creator of plugin says it is an external (kodi or elec) problem even though it only happens with the netflix plugin

### How to reproduce
Steps to reproduce the behavior:
open netflix plugin
play something

### Information
latest 20 alpha but it also happend with stable of 19 and 20

### Log file
https://paste.kodi.tv/nufefeduqi.kodi

### Context

heitbaum, 2024-03-11:
Is this issue still occurring with latest LE12 20240311-86dba4b images. If so please share the updated debug logs. Alternatively please raise this on the forum for support.

heitbaum, 2024-03-16:
Closing as inactive. Happy to open again - but will need debug logs, ideally a reproducible case. Using the forum for support may get more answers as others may have seen the same issue.

closed 2024-03-16 by heitbaum

DDQW, 2024-04-02:
Sadly still not working. I have an account on the libreelec board for a while now but for some reason i have no option to post. 
If you are an admin there please contact me over private message i don't want to post my account name here :D

heitbaum, 2024-04-02:
> Sadly still not working. I have an account on the libreelec board for a while now but for some reason i have no option to post. If you are an admin there please contact me over private message i don't want to post my account name here :D

@chewitt / @CvH can you help?

CvH, 2024-04-02:
@DDQW whats your forum name ?

DDQW, 2024-04-02:
https://forum.libreelec.tv/core/user/31924-mordred/

CvH, 2024-04-02:
@DDQW please try again, should work now

DDQW, 2024-04-02:
sadly also no. i have subscribe, mark all as read and rssfeed as options. No way to create or answer anywhere (currently i am in general support)

i did log out and relogin and a cache reset with strg+f5

CvH, 2024-04-03:
@DDQW pls try again, if this is not working I can remove the account and you can reregister - its unclear why you can't do anything at the forum (we had this before at other accounts).

DDQW, 2024-04-03:
still no feel free to nuke it and thx for the sup

CvH, 2024-04-04:
@DDQW done

DDQW, 2024-06-28:
i tried again a few days ago same mail still not able to post anything. do you maybe work with a tld whitelist so it refuses to accept my personal domain?

CvH, 2024-06-28:
can you drop me a mail at myname (c...) @libreelec.tv ?
I am not too sure what happen but we can fix this.

DDQW, 2024-07-02:
the mail is out. it might have gone to spam you might want to check it.

DDQW, 2025-03-16:
https://forum.libreelec.tv/thread/29577-x86-64-netflix-colorpumping-and-flickering/

in case someone stubmbles over this it now is in the libreelec forum