LibreELEC/LibreELEC.tv #8332: [BUG] Bluetooth connect during automatic background scanning crashes Libreelec

2019-05-10, 2023-11-15:
### Describe the bug
LE always scans for BT devices, even when you just want to connect yor already known, paired, trusted BT device.
This is a surefire way to crash LE - in 8 out of 10 attempts LE crashes and takes ages to restart.
The only way to avoid crashes is to wait several minutes (which here means until 2 to 3 devices from my neighbours' cars, TVs, laptops, phones appear) and only then connect to _my_ device. 

I really have no interest in any random device being in the vicinty by pure accident, but only in devices _I_ added/want to add.
Given that this is nothing I (or probably anyone else) do everyday, there's no good reason for automatic scanning for devices.
Instead LE should limit itself to already known devices by default and look for new devices only if asked to do that explicitly.

### How to reproduce
Steps to reproduce the behavior:
1. Start Libreelec
2. Go to Bluetooth
3. See your paired and trusted BT device
4. try to connect
5. see LE crash hard

### Information
 - LibreELEC Version: [e.g. 11.0.3]
 - Hardware Platform: [e.g. RPi4]

### Log file

### Context

heitbaum, 2024-01-07:
Could yo7nplease retest with LE12 and advise if still a problem

heitbaum, 2024-03-11:
Closing this LE11 issue. Is this issue still occurring with latest LE12 20240311-86dba4b images. If so please share the debug logs. Changes have been made to Bluetooth in LE12 that may have addressed this. Alternatively please raise this on the forum for support.

closed 2024-03-11 by heitbaum

2019-05-10, 2024-03-18:
Give that as of yesterday no update to 12 is offered, I am still unable to comment on wether the issue is fixed or not.

chewitt, 2024-03-19:
> Give that as of yesterday no update to 12 is offered

No update .. other than this one: https://libreelec.tv/2024/03/17/libreelec-nexus-12-0-beta1/