LibreELEC/LibreELEC.tv #8335: [BUG] Choppy playback with VDPAU and NVidia

salvojettison, 2023-11-16:
Hello everyone, thank you for your awesome work in developing LibreELEC!

### Describe the bug
I was using LibreELEC-Generic.x86_64-10.0.4 on some old laptop with Intel Core2Duo and NVidia Geforce 9300M GS (uses nvidia-legacy-340.180). I have seen, that the NVidia support has been continued in LibreELEC-Generic-legacy.x86_64-11.0.3, so I gave it a try. I am playing H264 on 1280x720 or 1920x1080. Seemingly HW Acceleration is enabled, I have CPU load about 10-30%, but the playback is choppy. It shows around 10-30 FPS. Same video on LE 10 plays seamless with the same VDPAU acceleration at ~60Hz. CPU load is similar, maybe a bit higher.
If I disable VDPAU HW acceleration, the playback is fine, but of course it results in higher CPU load. In V10.0.4 the playback is fine.

I have tried to disable C states, but it made no difference in the quality of playback.

### How to reproduce
Steps to reproduce the behavior:
Fresh install of Generic-legacy.x86_64-11.0.3.
Start video with HW acceleration enabled (VDPAU)
My test video was this: https://drive.google.com/file/d/0BwxFVkl63-lEUGZnR1JHUUU0WEk/view?usp=sharing&resourcekey=0-YrNv_pEiY4Ukaw_X5rckRw
It behaves similar with other h264 files.

### Information
 - LibreELEC Version: [Generic-legacy.x86_64-11.0.3]
 - Hardware Platform: [amd64]

### Log file
http://ix.io/4LFS

### Context
I have noticed that vdpauinfo results in segfault, while on LE V10.0.4 it gives back proper result.

Thank you for your help!

salvojettison, 2023-12-06:
Thank you for the recommendation and the instructions, I will try it as soon as I will have some time. Extra thanks for the hints how to add vdpau support!
However my experience with the same system under Debian based distros, that the nouveau driver is considerably worse at video playback than nvidia 340.180. I wonder, if that will be different under LibreELEC.

heitbaum, 2024-01-07:
Please test and advise with LE12 nightlies. Generic-Legacy still has nvidia 340.108.

heitbaum, 2024-03-11:
Closing this issue. If this issue is still occurring with latest LE12 20240311-86dba4b images - if so please share the debug logs. Alternatively please raise this on the forum for support.

closed 2024-03-11 by heitbaum