LibreELEC/LibreELEC.tv #8364: [BUG] LE 12 Update via GUI generates a crash (-log)

eldersixpack13, 2023-11-25:
### Describe the bug
see Bug #7987

LE 12 nightly update from 52b450b to 8b39055, but it happend earlier too

### How to reproduce
Steps to reproduce the behavior:
1. update via GUI
2. reboot

### Information
 - LibreELEC Version: LE12 
 - Hardware Platform: Intel NUC13

### Log file
log during update:
[kodi_crash.log](https://github.com/LibreELEC/LibreELEC.tv/files/13465352/kodi_crash.log)

log after update:
[kodi.log](https://github.com/LibreELEC/LibreELEC.tv/files/13465353/kodi.log)

### Context
regarding the logs:  I guess the crash happens during the update (see nightly git hashes)

Network => Maximum Wait Time: 5 sec

antonlacon, 2023-11-29:
So the steps are: update via gui, and find a kodi crash log in /storage/.kodi/temp? I can't reproduce with an RPi3. I used the November 24 -> November 25 nightly RPi2 builds as in your logs.

Would you try logging in via ssh and running `systemctl --no-block reboot` and see if it produces a crash log? This is the reboot command being used when the addon performs an update.

eldersixpack13, 2023-11-29:
Thanks for looking into it.

"systemctl --no-block reboot" (without quotes) does not produce a crash log.

I'm no developer (just an IT-Monkey (Support, Admin) retired, using ~20 years Linux (Redhat 6.x, Fedora since FC1) 
To me it seems  /usr/lib/libpython is running wild.

Box was new installed ~3 month ago, LE11, upgrade to LE12, ...

eldersixpack13, 2023-11-29:
as said tested with Maximum Wait Time: 5 sec

=> crash

[kodi_crash.log](https://github.com/LibreELEC/LibreELEC.tv/files/13506304/kodi_crash.log)

heitbaum, 2024-01-07:
Is this still happening with current nightlies.

eldersixpack13, 2024-01-07:
Yes !

eldersixpack13, 2024-01-18:
seems to be fixed since nightly (18.01.2024)
Thanks !

heitbaum, 2024-03-16:
@mglae - is this still an issue, or did it get fixed in the current LE12b1?

eldersixpack13, 2024-03-16:
Still an issue here. 
_In January I made a wrong assumption after 3 in row without crashlog_

heitbaum, 2024-05-25:
Status request:
Is this still occurring in LE12 (released)? 
Is this occurring in LE13 nightlies?

eldersixpack13, 2024-05-25:
yes on LE13 nightly 738b75b

heitbaum, 2024-10-29:
@eldersixpack13 are you still experiencing this?

eldersixpack13, 2024-10-29:
Yes

heitbaum, 2025-02-27:
Is this still occurring? (LE13 nightly)

eldersixpack13, 2025-02-27:
@heitbaum I guess we both are having a sorta "telepathic wire"

thought on this topic last weekend, after 3 updates without crash.
but no, the 4th update crashed my thoughts and the LE Box
in short: update crashes are still happening
and
apart I get ironed by a bus or so I have this topic on my radar and will close it, when ....