LibreELEC/LibreELEC.tv #8365: [BUG] error <general>: CCurlFile::Open - <http://127.0.0.1:9981/imagecache/38> Failed with code 404:

eldersixpack13, 2023-11-25:
### Describe the bug
I see: "error <general>: CCurlFile::Open - <http://127.0.0.1:9981/imagecache/38> Failed with code 404:"

### How to reproduce
Steps to reproduce the behavior:
1. take a look at the Kodi log files

### Information
 - LibreELEC Version: LE12 nightly
 - Hardware Platform: Intel NUC 13

### Log file
[kodi.log](https://github.com/LibreELEC/LibreELEC.tv/files/13465373/kodi.log)

### Context
- I tried to fix it via the TVH settings regarding icons/images
- changed IP adress
- ... ? 
but peanuts !

maybe related:
https://forum.libreelec.tv/thread/26351-kodi-is-not-loading-channel-icons-from-tvheadend-server/?postID=174588&highlight=imagecache#post174588

CvH, 2023-11-26:
I can just guess but this sounds related https://github.com/xbmc/xbmc/issues/23958#issuecomment-1817540588

eldersixpack13, 2023-11-26:
@CvH 

I'm no developer, but I don't think so.

The provided thread - for my taste - deals with icons regarding the Movie/Pics folders.

thereby involved Dir's and DB'S:
/storage/.kodi/userdata/Thumbnails
and
/storage/.kodi/userdata/Database/Textures13.db

Before I do backups I clean/delete content of Thumbnails and the Textures13.db (to get smaller backups). 
Anew Textures13.db is greated after reboot and Thumbnails folder is filled by opening files in the Movie/Picture Folder.

My Bug deals with **channel icons**:
/storage/.kodi/userdata/addon_data/service.tvheadend43/imagecache/meta
and
/storage/picons/tvh

for example one file content of .../meta:
cat /storage/.kodi/userdata/addon_data/service.tvheadend43/imagecache/meta/38:

{
	"url": "file:///storage/picons/tvh/wdrhdkoln.png",
	"accessed": 1701017281
}

what points to and **empty** /storage/picons/tvh/

Cause the image cache error messages are listed multiple times over and over again in the log - my guess: triggered by channel switching -  to me it seams the bug is in HTSP Client or TVH

see also:
/storage/.kodi/userdata/addon_data/service.tvheadend43/service.log

example for imagecache/38 from **today's** service.log (**entries do not fit with my above kodi.log; other day !**) :

2023-11-**26** 17:48:06.724 [  ERROR] http: 127.0.0.1: HTTP/1.1 GET (1) /imagecache/38 -- 404
2023-11-26 17:48:06.727 [  ERROR] http: 127.0.0.1: HTTP/1.1 GET (1) /imagecache/38 -- 404
2023-11-26 17:48:06.730 [  ERROR] http: 127.0.0.1: HTTP/1.1 GET (1) /imagecache/38 -- 404
2023-11-26 17:48:06.731 [  ERROR] http: 127.0.0.1: HTTP/1.1 HEAD (2) /imagecache/38 -- 404
2023-11-26 17:48:06.731 [  ERROR] http: 127.0.0.1: HTTP/1.1 GET (1) /imagecache/38 -- 404
2023-11-26 17:48:06.732 [  ERROR] http: 127.0.0.1: HTTP/1.1 GET (1) /imagecache/38 -- 404
2023-11-26 17:48:06.734 [  ERROR] http: 127.0.0.1: HTTP/1.1 GET (1) /imagecache/38 -- 404
2023-11-26 17:48:06.740 [  ERROR] http: 127.0.0.1: HTTP/1.1 GET (1) /imagecache/38 -- 404
2023-11-26 17:48:06.743 [  ERROR] http: 127.0.0.1: HTTP/1.1 GET (1) /imagecache/38 -- 404
2023-11-26 17:48:06.746 [  ERROR] http: 127.0.0.1: HTTP/1.1 GET (1) /imagecache/38 -- 404
2023-11-26 17:48:06.746 [  ERROR] http: 127.0.0.1: HTTP/1.1 HEAD (2) /imagecache/38 -- 404
2023-11-26 17:48:06.746 [  ERROR] http: 127.0.0.1: HTTP/1.1 GET (1) /imagecache/38 -- 404
2023-11-26 17:48:06.749 [  ERROR] http: 127.0.0.1: HTTP/1.1 GET (1) /imagecache/38 -- 404
2023-11-26 17:48:06.751 [  ERROR] http: 127.0.0.1: HTTP/1.1 GET (1) /imagecache/38 -- 404

heitbaum, 2024-01-07:
Is this the issue/fix? https://github.com/xbmc/xbmc/commit/9f02ffd0ae693dcdcd5d6e8bb715134e8efc3803

CvH, 2024-01-07:
> Is this the issue/fix? [xbmc/xbmc@9f02ffd](https://github.com/xbmc/xbmc/commit/9f02ffd0ae693dcdcd5d6e8bb715134e8efc3803)

i guess so https://github.com/xbmc/xbmc/issues/23958#issuecomment-1880019549

closed 2024-01-07 by CvH

eldersixpack13, 2024-01-08:
This bug is **NOT** fixed !
tested with nightly-20240108-f9912ab

TVH service.log:
[service.log](https://github.com/LibreELEC/LibreELEC.tv/files/13866116/service.log)

PR #8493 (2024-01-07): kodi: update to githash bdc4ab8
may resolve #8365

```
=== tested on ===
Generic.x86_64-devel-20240107083254-5cf49e4
Linux nuc12 6.6.10 #1 SMP Sat Jan  6 04:32:31 UTC 2024 x86_64 GNU/Linux
Starting Kodi (21.0-BETA2 (20.90.821) Git:bdc4ab89ae787d359f47a0148ef95d512e0be4a9). Platform: Linux x86 64-bit
Using Release Kodi x64
Kodi compiled 2024-01-07 by GCC 13.2.0 for Linux x86 64-bit version 6.6.10 (394762)
Running on LibreELEC (heitbaum): devel-20240107083254-5cf49e4 12.0, kernel: Linux x86 64-bit version 6.6.10
FFmpeg version/source: 6.0.1
Host CPU: 12th Gen Intel(R) Core(TM) i7-1260P, 16 cores [... cut]