LibreELEC/LibreELEC.tv #8420: Using the zoom on 12.0 / RPI5 makes the picture disapear définitively

dmeziere, 2023-12-13:
### Describe the bug

Adjusting the zoom while watching an UHD/4k video to get rid of upper and lower black bands turns off the picture (but the audio stays). It will last until reboot (i used Kore on my phone to reboot). Fortunately, the zoom level is reminded, and after a certain amount of reboot, the zoom level will be set.

### How to reproduce

Steps to reproduce the behavior:
1. Load a video (Mine was UHD/4k 130 GB)
2. Call the menu with the gear icon
3. Go to video / zoom and adjust it until the upper and lower black bands disapear
4. On the 4th / 5th zoom raise, the picture disapear

### Information
 - LibreELEC Version: LibreELEC-RPi5.aarch64-12.0-nightly-20230930-dd26ba0.img.gz
 - Hardware Platform: RPi5 with official box, fan equipped

davidoppen, 2023-12-15:
I‘m having the same issue on RPi5 with the latest nightly of 11.0 (14.12.).

heitbaum, 2024-01-07:
> ### Describe the bug
> Adjusting the zoom while watching an UHD/4k video to get rid of upper and lower black bands turns off the picture (but the audio stays). It will last until reboot (i used Kore [...]

Can you please share the logs. And update to the current LE12 nightlies.

heitbaum, 2024-03-11:
Closing. Is this issue still occurring with latest LE12 20240311-86dba4b images. If so please share the updated debug logs. Alternatively please raise this on the forum for support.

closed 2024-03-11 by heitbaum

dmeziere, 2024-03-11:
The bug is still present with LibreELEC-RPi5.aarch64-12.0-nightly-20240311-86dba4b.

heitbaum, 2024-03-12:
> The bug is still present with LibreELEC-RPi5.aarch64-12.0-nightly-20240311-86dba4b. I don't know how to give you a debug log. I see a logfiles folder when connected with ssh, but it is empty.

Hi @dmeziere - the forum will be able to guide you - https://forum.libreelec.tv/ - also will have a few more eyes on the issue.

dmeziere, 2024-03-12:
That said, I don't understand what makes this bug closed ?