LibreELEC/LibreELEC.tv #8446: [BUG] LE12 nightlies for Rock64 broken

rdorsch, 2023-12-22:
### Describe the bug
Hello,

the latest nightlies for Rock64 are broken, after installation, the libreelec logo and the version string appear, then the display powers of. This happens for a fresh install and an upgrade from LE11.

journalctl reveals

Dec 22 11:20:00 LibreELEC kernel: ------------[ cut here ]------------
Dec 22 11:20:00 LibreELEC kernel: [CRTC:49:crtc-0] vblank wait timed out
Dec 22 11:20:00 LibreELEC kernel: WARNING: CPU: 0 PID: 11 at drivers/gpu/drm/drm_atomic_helper.c:1679 drm_atomic_helper_wait_for_vblanks.part.0+0x224/0x248
Dec 22 11:20:00 LibreELEC kernel: Modules linked in: 8021q cfg80211 rfkill rockchip_vdec(C) rockchip_iep hantro_vpu gpio_ir_recv v4l2_vp9 v4l2_h264 videobuf2_dma_contig videobuf2_memops v4l2_mem2mem spi_rockchip videobuf2_v4l2 videobuf2_common rk_crypto crypto_engine pkcs8_key_parser fuse
Dec 22 11:20:00 LibreELEC kernel: CPU: 0 PID: 11 Comm: kworker/u8:0 Tainted: G        WC         6.6.7 #1
Dec 22 11:20:00 LibreELEC kernel: Hardware name: Pine64 Rock64 (DT)
Dec 22 11:20:00 LibreELEC kernel: Workqueue: events_unbound commit_work
Dec 22 11:20:00 LibreELEC kernel: pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Dec 22 11:20:00 LibreELEC kernel: pc : drm_atomic_helper_wait_for_vblanks.part.0+0x224/0x248
Dec 22 11:20:00 LibreELEC kernel: lr : drm_atomic_helper_wait_for_vblanks.part.0+0x224/0x248
Dec 22 11:20:00 LibreELEC kernel: sp : ffff8000817ebcc0
Dec 22 11:20:00 LibreELEC kernel: x29: ffff8000817ebcc0 x28: 000000000000038a x27: 0000000000000000
Dec 22 11:20:00 LibreELEC kernel: x26: 0000000000000001 x25: 0000000000000000 x24: ffff000000959800
Dec 22 11:20:00 LibreELEC kernel: x23: 0000000000000001 x22: 0000000000000000 x21: 0000000000000000
Dec 22 11:20:00 LibreELEC kernel: x20: ffff00001e899200 x19: ffff000000950080 x18: ffffffffffffffff
Dec 22 11:20:00 LibreELEC kernel: x17: 000000040044ffff x16: 00000032b5503510 x15: ffffffffffffffff
Dec 22 11:20:00 LibreELEC kernel: x14: 0000000000000000 x13: ffff800081590b20 x12: 000000000000281a
Dec 22 11:20:00 LibreELEC kernel: x11: 0000000000000d5e x10: ffff8000815e8b20 x9 : ffff800081590b20
Dec 22 11:20:00 LibreELEC kernel: x8 : 00000000ffffefff x7 : ffff8000815e8b20 x6 : 0000000000000001
Dec 22 11:20:00 LibreELEC kernel: x5 : ffff0000fe7598d0 x4 : 0000000000000000 x3 : 0000000000000027
Dec 22 11:20:00 LibreELEC kernel: x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff000000360d80
Dec 22 11:20:00 LibreELEC kernel: Call trace:
Dec 22 11:20:00 LibreELEC kernel:  drm_atomic_helper_wait_for_vblanks.part.0+0x224/0x248
Dec 22 11:20:00 LibreELEC kernel:  drm_atomic_helper_commit_tail_rpm+0x80/0x98
Dec 22 11:20:00 LibreELEC kernel:  commit_tail+0xa0/0x168
Dec 22 11:20:00 LibreELEC kernel:  commit_work+0x10/0x18
Dec 22 11:20:00 LibreELEC kernel:  process_one_work+0x134/0x270
Dec 22 11:20:00 LibreELEC kernel:  worker_thread+0x2ac/0x484
Dec 22 11:20:00 LibreELEC kernel:  kthread+0xdc/0xe0
Dec 22 11:20:00 LibreELEC kernel:  ret_from_fork+0x10/0x20
Dec 22 11:20:00 LibreELEC kernel: ---[ end trace 0000000000000000 ]---

### How to reproduce
Either install or upgrade to the nightlies of LE12 for rock64, e.g. LibreELEC-RK3328.aarch64-12.0-nightly-20231222-581a2fe-rock64.img.gz. I am not sure, when the problem was introduced.

### Information
 - LibreELEC Version: 12.0-nightly-20231222-581a2fe
 - Hardware Platform: rock64

### Log file
Since there is no user interface accessible anymore, I have only ssh access:
I uploaded the full journalctrl -b output and the kodi.log

https://bokomoko.de/~rd/Libreelec/2023-12-22-rock64-failure/

### Context
I connected the device to a 4k screen, but that works flawless with 11.0.3

rdorsch, 2023-12-22:
I did bisect the issue:

LibreELEC-RK3328.arm-12.0-nightly-20230629-0e83d09-rock64.img.gz works
LibreELEC-RK3328.arm-12.0-nightly-20230705-db76177-rock64.img.gz fails

There are quite a few commits during that period, does one look suspicious to anybody:

rd@h370:~/tmp.nobackup/git/LibreELEC.tv$ git whatchanged --since=2023-06-29T18:23 --until=2023-07-05T09:19|cat -
commit 3b0a28df252b04c2e1fb38aa5e1013b742367742
Author: Rudi Heitbaum <rudi@heitbaum.com>
Date:   Sat Jun 24 10:22:54 2023 +0000

    Revert "amlogic: add ffmpeg patches for improved HEVC seek"
    
    This reverts commit 57249674d9fc02493f8cb030ea529d1003d840d6.
    
    These patches are included in https://github.com/jc-kynesim/rpi-ffmpeg/tree/dev/6.0/rpi_import_1/
    since 2c9e9d043c2da0108f635d19461c0461adffb43c

:100644 000000 d3ecee3b5e 0000000000 D  projects/Amlogic/patches/ffmpeg/ffmpeg-hevc-001.patch
:100644 000000 e1de77c269 0000000000 D  projects/Amlogic/patches/ffmpeg/ffmpeg-hevc-002.patch
:100644 000000 8f5844c284 0000000000 D  projects/Amlogic/patches/ffmpeg/ffmpeg-hevc-003.patch

commit 4d3cf7f88465879f508a09190fdace13166b7455
Author: Rudi Heitbaum <rudi@heitbaum.com>
Date:   Sat Jul 1 09:38:00 2023 +0000

    linux (Amlogic): update to 6.1.36

:100644 100644 e850dd62f4 71dc6b5136 M  packages/linux/package.mk
:100644 000000 f934783d1c 0000000000 D  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0050-FROMLIST-v1-mmc-meson-gx-fix-deferred-probing.patch

commit 54a94ebdf2dd9c1ba45d3f494390de8b5a9d8165
Author: Rudi Heitbaum <rudi@heitbaum.com>
Date:   Sat Jul 1 08:51:48 2023 +0000

    linux (Amlogic): update to 6.1.28 to support shared patches

:100644 100644 e68affaeda e850dd62f4 M  packages/linux/package.mk
:100644 000000 06834480de 0000000000 D  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0013-FROMGIT-FIXES-drm-meson-fix-1px-pink-line-on-GXM-whe.patch

commit 24356d267a4a2ac5e3491f937fa30f5881f3d6c3
Author: Christian Hewitt <christianshewitt@gmail.com>
Date:   Tue Feb 28 19:26:08 2023 +0000

    linux: update Amlogic Linux 6.1.x patches

:100644 100644 efaf88ae8f e68affaeda M  packages/linux/package.mk
:100644 100644 03926984fd 74ebd01acc M  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0001-LOCAL-set-meson-gx-cma-pool-to-896MB.patch
:100644 100644 5922a41aad e919d19953 M  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0002-LOCAL-set-meson-g12-cma-pool-to-896MB.patch
:100644 100644 c4b3467369 b8437d85d4 M  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0003-LOCAL-arm64-fix-Kodi-sysinfo-CPU-information.patch
:100644 100644 404fbe35ad da89855fd1 M  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0004-LOCAL-arm64-meson-add-Amlogic-Meson-GX-PM-Suspend.patch
:100644 100644 e110bab6d8 3eade2ab65 M  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0005-LOCAL-arm64-dts-meson-add-support-for-GX-PM-and-Virt.patch
:100644 100644 6bc2cf51e8 91295503bf M  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0006-LOCAL-arm64-dts-meson-add-rtc-vrtc-aliases-to-Khadas.patch
:100644 100644 3330b62da2 f0717cab55 M  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0007-LOCAL-arm64-dts-meson-add-rtc-vrtc-aliases-to-Khadas.patch
:100644 100644 d41979c349 1a1eab50af M  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0008-LOCAL-arm64-dts-meson-add-rtc-vrtc-aliases-to-Minix-.patch
:100644 100644 395f00e564 48beb80f41 M  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0009-LOCAL-ALSA-Assign-internal-PCM-chmap-ELD-IEC958-kctl.patch
:100644 100644 e41376cf42 f57591283e M  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0010-LOCAL-usb-hub-disable-autosuspend-for-Genesys-Logic-.patch
:100644 100644 3d7d2f65df b046c64bd3 M  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0011-LOCAL-of-partial-revert-of-fdt.c-changes.patch
:100644 100644 fd416bee7a 6a8a2f3aac M  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0012-FROMGIT-FIXES-drm-meson-meson_venc-Relax-the-support.patch
:100644 100644 06072948de 06834480de M  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0013-FROMGIT-FIXES-drm-meson-fix-1px-pink-line-on-GXM-whe.patch
:100644 100644 f16f537932 d9742dba27 M  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0014-FROMGIT-6.2-Revert-arm64-dts-meson-sm1-odroid-hc4-di.patch
:100644 100644 b39e465251 1093f79244 R087       projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0016-FROMGIT-6.3-arm64-dts-meson-add-Broadcom-WiFi-to-P21.patch    projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0015-FROMGIT-6.3-arm64-dts-meson-add-Broadcom-WiFi-to-P21.patch
:100644 000000 a5f93a7da1 0000000000 D  projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0015-FROMGIT-6.3-arm64-dts-meson-remove-CPU-opps-below-1G.patch
:100644 100644 d4a57f5cb0 cd9b564713 R090       projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0017-FROMGIT-6.3-arm64-dts-meson-move-pwm_ef-node-in-P212.patch    projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0016-FROMGIT-6.3-arm64-dts-meson-move-pwm_ef-node-in-P212.patch
:100644 100644 57237e6708 bd0fddd8d9 R090       projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0018-FROMGIT-6.3-arm64-dts-meson-remove-WiFi-BT-nodes-fro.patch    projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0017-FROMGIT-6.3-arm64-dts-meson-remove-WiFi-BT-nodes-fro.patch
:100644 100644 6eae91ae46 6a18442973 R095       projects/Amlogic/devices/AMLGX/patches/linux/amlogic-0019-FROMGIT-6.3-arm64-dts-meson-add-audio-playback-to-S9.patch    projects/Amlogic/devices/AMLGX/patches/lin [... cut ...]
    mediainfo: update to 23.06

:100644 100644 47a0e9ea4d 4be95c7fe2 M  packages/addons/addon-depends/multimedia-tools-depends/mediainfo/package.mk

commit efcfc739223f59f2c6f87f1b66fecc4f0d98ed18
Author: Ian Leonard <antonlacon@gmail.com>
Date:   Thu Jun 29 22:16:44 2023 -0400

    LibreELEC-settings: update to b920d5d
    
    Signed-off-by: Ian Leonard <antonlacon@gmail.com>

:100644 100644 7511e2db1c 3e8f0e16d8 M  packages/mediacenter/LibreELEC-settings/package.mk
rd@h370:~/tmp.nobackup/git/LibreELEC.tv$

rdorsch, 2023-12-22:
For reference: the last working version has kernel

LibreELEC:~ # uname -a
Linux LibreELEC 6.3.8 #1 SMP Thu Jun 29 20:19:38 UTC 2023 aarch64 GNU/Linux
LibreELEC:~ # 

The next image (which fails) has 6.4.1 (not sure if the kernel upgrade is the root cause though).

heitbaum, 2023-12-22:
> For reference: the last working version has kernel
> 
> LibreELEC:~ # uname -a Linux LibreELEC 6.3.8 #1 SMP Thu Jun 29 20:19:38 UTC 2023 aarch64 GNU/Linux LibreELEC:~ #
> 
> The next image (which fails) has 6.4.1 (not sure if the kernel upgrade is the root cause though).

- If it is the kernel then it will be one on the reworked RK patches I’m guessing https://github.com/LibreELEC/LibreELEC.tv/pull/7837

rdorsch, 2023-12-22:
At least there are a few good candidates in there: most suspicious looks for me

https://github.com/LibreELEC/LibreELEC.tv/pull/7837#issuecomment-1605868312

since after the LibreELEC boot logo with the image version, the monitor reports that there is no HDMI signal anymore.

Are there more debug logs which might be useful to find out where the problem comes from aside from the ones I posted in the bug report?

heitbaum, 2024-03-11:
Is this issue still occurring with latest LE12 20240311-86dba4b images. I expect that it probably is - as there have been no real changes apart from the update to the kernel to the latest 6.6.y

rdorsch, 2024-03-11:
I have not tested since my last comment, since I have neither seen anything relevant in the commit logs nor a comment here. I try to do a test in the next days.

rdorsch, 2024-03-16:
Am Montag, 11. März 2024, 06:19:07 CET schrieb Rudi Heitbaum:
> Is this issue still occurring with latest LE12 20240311-86dba4b images. I
> expect that it probably is - as there have been no real changes apart from
> the update to the kernel to the latest 6.6.y

As you expected, the issue is still present in ~/Downloads/LibreELEC-
RK3328.aarch64-12.0-nightly-20240313-d108ae1-rock64.img.gz

But I found by chance that the problem occurs if conneced to a 4k display. If 
connected to a full HD display, the image works (and probably always worked). 
I.e. most likely only 4k is broken.

heitbaum, 2024-03-16:
Playing this back - the correct bug subject could be:
- Since #7837 kernel 6.4.y Rock64 does not display on a 4K screen

looking at the code that changed in the 6.4.y it is around the clock rates and the resolutions…. So could make sense.

I’m not sure the next steps on what to collect (data wise) but Ill reach out and hopefully we can progress this now …

rdorsch, 2024-03-16:
What I noticed in addition on full HD, that for dash streams, there are large 
audio sync errors, for a full kodi.log see

https://bokomoko.de/~rd/Libreelec/rock64_large_audio_sync_error_LE12_kodi.log

But that is likely an unrelated issue.

rdorsch, 2024-03-16:
Am Samstag, 16. März 2024, 16:55:46 CET schrieb Rudi Heitbaum:
> Playing this back - the correct bug subject could be:
> - Since #7837 kernel 6.4.y Rock64 does not display on a 4K screen
>
> looking at the code that changed in the 6.4.y it is around the clock rates [...]

let me please know, if I can help with something.

Rainer

heitbaum, 2024-12-10:
Now that we are on the 6.12 kernel that will be part of LE13 - can you please install and test https://test.libreelec.tv/13.0/Rockchip/rock64/ 2024-12-09 or newer, so that we can validate it works / or next steps.

rdorsch, 2025-11-01:
I did a test with

LibreELEC-RK3328.aarch64-13.0-nightly-20251030-a4cd4bd-rock64.img.gz 

today. 

The display does not power off anymore after boot.

There is sometimes some light flickering of the display though.

And I also saw an unexpected automatic reboot in the first 5 minutes of testing, but that is maybe due to the overall state of 13.0 (?).

For reference I uploaded the [kodi_crashlog](https://bokomoko.de/~rd/Libreelec/kodi_crashlog_20251101204406.log) and the corresponding [kodi.log](https://bokomoko.de/~rd/Libreelec/kodi.old.log)

rdorsch, 2025-11-01:
I also tried to play 4k content, that failed immediately, system (maybe only the user interface) hang and the screen became green.

```
rd@h370:~/tmp.nobackup$ mediainfo Costa_Rica.mp4 
General
Complete name                            : Costa_Rica.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/mp41)
File size                                : 1.06 GiB
Duration                                 : 5 min 13 s
Overall bit rate mode                    : Variable
Overall bit rate                         : 29.0 Mb/s
Frame rate                               : 59.940 FPS
Writing application                      : Lavf61.7.100

Video
ID                                       : 1
Format                                   : VP9
HDR format                               : SMPTE ST 2086, HDR10 compatible
Codec ID                                 : vp09
Duration                                 : 5 min 13 s
Bit rate                                 : 28.9 Mb/s
Width                                    : 3 840 pixels
Height                                   : 2 160 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Variable
Frame rate                               : 59.940 FPS
Minimum frame rate                       : 55.556 FPS
Maximum frame rate                       : 62.500 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bits/(Pixel*Frame)                       : 0.058
Stream size                              : 1.06 GiB (100%)
Title                                    : ISO Media file produced by Google Inc. Created on: 11/01/2025.
Color range                              : Limited
Color primaries                          : BT.2020
Transfer characteristics                 : PQ
Matrix coefficients                      : BT.2020 non-constant
Mastering display color primaries        : R: x=0.678000 y=0.322000, G: x=0.245000 y=0.703000, B: x=0.138000 y=0.052000, White point: x=0.312700 y=0.329000
Mastering display luminance              : min: 0.0001 cd/m2, max: 1000 cd/m2
Maximum Content Light Level              : 1100 cd/m2
Maximum Frame-Average Light Level        : 180 cd/m2

Audio
ID                                       : 2
Format                                   : AAC LC
Format/Info                              : Advanced Audio Codec Low Complexity
Codec ID                                 : mp4a-40-2
Duration                                 : 5 min 13 s
Bit rate mode                            : Variable
Bit rate                                 : 128 kb/s
Maximum bit rate                         : 132 kb/s
Channel(s)                               : 2 channels
Channel layout                           : L R
Sampling rate                            : 44.1 kHz
Frame rate                               : 43.066 FPS (1024 SPF)
Compression mode                         : Lossy
Stream size                              : 4.79 MiB (0%)
Default                                  : Yes
Alternate group                          : 1

rd@h370:~/tmp.nobackup$
```

heitbaum, 2025-11-01:
> I will also suggest a clean image on an sdcard ensuring we have the latest u-boot…
>
> I started from a fresh image dd'ed on the sd card.

Was confused - given the addon was there? Is it a clean .kodi directory (it shouldn’t be the issue) but just making sure we are not chasing shadows.

heitbaum, 2025-12-06:
Is this now fixed? https://forum.libreelec.tv/thread/30168-rock64-regression-on-le-12-2-1/?postID=203642#post203642

heitbaum, 2025-12-06:
I would suggest closing this issue then. And as the testing starts work with the forum post, and then things that are discrete - create issues for them. As much is changing as you can see, and this issue whilst still occurring is probably quite different now.

My suggestions being. 
1. Does it boot out of the box in 1080 using the default config (I hope yes) 
2. then the 4K testing …, forum first…