LibreELEC/LibreELEC.tv #8447: [BUG] LD_LIBRARY_PATH not updated for addons with commit 581a2fe

nvdias0, 2023-12-22:
### Describe the bug
I have some add-ons from may own (nginx with some special configs, tvheadend, etc)
It has been working ok until today I've compiled LibreELEC from commit 581a2fe
And even (re)activating unknown sources in kodi, I've identified that LD_LIBRARY_PATH is not being updated for the installed ADD-ONS.

At the moment LD_LIBRARY is kept only with /usr/lib:/usr/lib/pulseaudio'

 - LibreELEC Version: 12 commit 581a2fe
 - Hardware Platform: RPi4

chewitt, 2023-12-22:
I'd guess related to https://github.com/LibreELEC/LibreELEC.tv/issues/7602 which means the changes are intentional.

nvdias0, 2023-12-22:
yeah ! that's it. Thanks

(a lot of changes for the addos in the next days 😁 )

heitbaum, 2023-12-22:
@nvdias0 
- any of your private/customized addons that have libraries in lib will continue to work as expected https://github.com/LibreELEC/LibreELEC.tv/pull/8342/commits/065d3f5e88e6444b3514c9c14ab41089c9d4ba49 
- If you are referencing “LE” provided .so. files in your addons - then you will need to adjust your addons with patchelf (or some of the other ways shown in #8342 and #8385)

nvdias0, 2023-12-25:
@heitbaum , thanks for the suggestions.
They will help, as I have private add-ons with a lot of libraries in ../lib. 
I need now to patch some bins - it is not the best distribution approach, but I understand the need to clean LD_LIBRARY_PATH

nvdias0, 2023-12-28:
@heitbaum 
> @nvdias0
> 
> * any of your private/customized addons that have libraries in lib will continue to work as expected [065d3f5](https://github.com/LibreELEC/LibreELEC.tv/commit/065d3f5e88e6444b3514c9c14ab41089c9d4ba49)
> * If you are referencing “LE” provided .so. files in your addons - then you will need to adjust your addons with patchelf (or some of the other ways shown in [Keep addon introduced private shared libraries private to the addon #8342](https://github.com/LibreELEC/LibreELEC.tv/pull/8342) and [Part 2 - Keep addon introduced private shared libraries private to the addon #8385](https://github.com/LibreELEC/LibreELEC.tv/pull/8385))

Sorry to get back to this issue, but looking into [065d3f5] it is not working as expected - LD_LIBRARY_PATH is not updated, even when there are *.so inside  $addon/lib dirs.

Looking into the code: changing LD_LIBRARY_PATH in a set of code between parenthesis will keep the change only inside that scope. right ? (sorry for my bad english).
I propose replacing the parenthesis by a if then clause. something like:
```
for addon in /storage/.kodi/addons/*/lib /usr/lib/kodi/addons/*/lib; do
if [ -d "$addon" ]; then
    files="$(find $addon ! -type d -name '*.so*' -maxdepth 1)"
    [ ! -z "$files" ] && LD_LIBRARY_PATH="$LD_LIBRARY_PATH:$addon"
fi
done

```

I will report this also in the [065d3f5] commit for easier tracing.
thanks

mglae, 2023-12-31:
@nvdias0 Confirmed.

Create a PR, it will be merged.

nvdias0, 2024-01-04:
> @nvdias0 Confirmed.
> 
> Create a PR, it will be merged.
done:
https://github.com/LibreELEC/LibreELEC.tv/pull/8471

closed 2024-01-05 by nvdias0