LibreELEC/LibreELEC.tv #8457: [BUG] no CEC support and no Ethernet (via USB Dongle) with AMLogic Radxa zero image

VanillaFord, 2023-12-28:
### Describe the bug
LE 11.03 and LE 11.04 image for AMLogic, Radxa-zero (4GB/16GB model) can boot fine, both from SD or EMMC.
However

1. during boot the message "unable to start ethermac.service" appears and an RT-based USB-Dongle does not receive an ethernet/IP connection.
2.  after startup, the Message appears, that pulseeight CEC adapter cannot be found and there is no CEC connection with the TV. (indeed there is no pluseight adapter present in the setup, as the platform should provide CEC support without this device).

### How to reproduce
Steps to reproduce the behavior:
1. boot LE 11.03 or 11.04, specific to radxa zero model.

### Information
 - LibreELEC Version: 11.03 and 11.04
 - Hardware Platform: Radxa zero with 4GB RAM and 16GB EMMC

### Log file
N/A

### Context
...both issues do not exist when booting CoreElec  20.1 (nexus).

chewitt, 2023-12-29:
What chipset is used in the USB Ethernet Adaptor? (and provide the VID:PID details)

VanillaFord, 2023-12-30:
...this is the dmesg output from armbian Jammy on the radxa zero with the said USB Ethernet adapter:

[    2.879458] kernel: usb 2-1: New USB device found, idVendor=0b95, idProduct=1790, bcdDevice= 2.00
[    2.879479] kernel: usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.879484] kernel: usb 2-1: Product: AX88179A
[    2.879489] kernel: usb 2-1: Manufacturer: ASIX
[    2.879493] kernel: usb 2-1: SerialNumber: 00ECCA37
[    2.978239] kernel: usbcore: registered new interface driver cdc_ether

lsusb shows:
Bus 002 Device 002: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet

chewitt, 2023-12-31:
The required kernel driver is enabled in our kernel config (CONFIG_USB_NET_AX88179_178A=m) so please share the full boot log. Also test with the image here (newer kernel): https://chewitt.libreelec.tv/testing/

VanillaFord, 2023-12-31:
...thanks for not giving up on me ;-)
I tried the newer 11.80.0 image for the radxa zero but both problems persist.

Without CEC working (mind you, its not the hdmi cable as coreelec 20.1 nexus is working) I have no means to access the boot log.
Tried to inject a wifi setting on startup with this method, but to no avail: https://forum.libreelec.tv/thread/10537-can-i-add-wifi-config-to-sd-card-pre-install-headless-setup/

heitbaum, 2024-03-11:
Is this issue still occurring with latest LE12 20240311-86dba4b images. If so please share the updated debug logs. Alternatively please raise this on the forum for support.

heitbaum, 2024-03-16:
Closing as inactive. Happy to open again - but will need debug logs, ideally a reproducible case. Using the forum for support may get more answers as others may have seen the same issue.

closed 2024-03-16 by heitbaum