LibreELEC/LibreELEC.tv #8461: [BUG]

drk1900, 2023-12-29:
### Describe the bug
"reloaded" mode in Artic Zhephyr OSD not rendered correctly

### How to reproduce
Steps to reproduce the behavior:
1. Open any movie or music file and change OSD mode to "reloaded"

### Information
 - LibreELEC-RPi4.aarch64-12.0-nightly-20231229-4e24516
 - Hardware Platform: [RPi4]

### Log file

### Context
Not happening in build LibreELEC-RPi4.aarch64-12.0-nightly-20231223 or before

CvH, 2023-12-29:
can you check what version is the first one that fails ?
So does [20231224](https://test.libreelec.tv/12.0/RPi/RPi4/LibreELEC-RPi4.aarch64-12.0-nightly-20231224-8fece3c.img.gz) and [20231227](https://test.libreelec.tv/12.0/RPi/RPi4/LibreELEC-RPi4.aarch64-12.0-nightly-20231227-74ff04a.img.gz) also shows that problem ?

https://github.com/LibreELEC/LibreELEC.tv/compare/581a2fe...4e24516

drk1900, 2023-12-29:
Yes, same issue in any build including 20231229

drk1900, 2023-12-29:
[image]

drk1900, 2023-12-29:
It looks like this, instead of the transparency gradient in 20231223 or previous

drk1900, 2023-12-29:
231213 is the last that works OK

CvH, 2023-12-29:
https://github.com/LibreELEC/LibreELEC.tv/compare/a1a3979...63347fc

likewise due a RPi kernel bump
can you share a complete logfile from 231213 and 231216

drk1900, 2023-12-29:
https://paste.libreelec.tv/harmless-jackass.log

drk1900, 2023-12-29:
https://paste.libreelec.tv/mint-goat.log

heitbaum, 2024-01-07:
- is this related to #8488

neo1973, 2024-01-07:
> is this related to https://github.com/LibreELEC/LibreELEC.tv/issues/8488

Probably not, RPi4 doesn't support fences:

```
warning <general>: [GBM] missing support for EGL_KHR_fence_sync and EGL_ANDROID_native_fence_sync - performance may be impacted
```

drk1900, 2024-01-09:
After a try with today build, 240109, no luck again, even with kernel 6.6.10, but changing the video player render method to EGL solves the issue

nekromantik, 2024-02-14:
I am seeing same on RPi5
Only on the OSD when playing movie

HiassofT, 2024-02-14:
Yes, that's a known issue and we are still thinking about how to best solve it.

Kodi prefers to use 30bit planes (10bit per R/G/B channel) so that leaves only 2 bits for alpha channel and thus the transparency looks crap.

closed 2024-02-17 by CvH