LibreELEC/LibreELEC.tv #8462: [BUG] libreelec 11.0.4 crashes on r600 where 10.0.3 is rock solid

oliv3r, 2023-12-29:
### Describe the bug
While watching tv via the pvr.hts addon, kodi occasionally crashes, or rather the video card crashes. Sadly, kodi does not get a chance to save its crashlog. But dmesg yields the following:

```
[ 2588.053609] kodi.bin[856]: segfault at 0 ip 00007f2325f02d00 sp 00007fffb401a500 error 4 in r600_dri.so[7f23255ca000+13fc000] likely on CPU 2 (core 2, socket 0)
[ 2588.053642] Code: 00 00 48 8b 5c 24 50 f6 47 03 02 74 4d 4e 8d 04 36 31 f6 45 85 c9 74 34 44 89 e8 45 89 e1 31 d2 66 2e 0f 1f 84 00 00 00 00 00 <41> 0f b6 0c 10 01 c1 66 89 0c 53 48 83 c2 01 49 39 d1 75 ec 48 85
[ 5169.646866] kodi.bin[1278]: segfault at 0 ip 00007f2b9df02d00 sp 00007ffe799b9d10 error 4 in r600_dri.so[7f2b9d5ca000+13fc000] likely on CPU 0 (core 0, socket 0)
[ 5169.646901] Code: 00 00 48 8b 5c 24 50 f6 47 03 02 74 4d 4e 8d 04 36 31 f6 45 85 c9 74 34 44 89 e8 45 89 e1 31 d2 66 2e 0f 1f 84 00 00 00 00 00 <41> 0f b6 0c 10 01 c1 66 89 0c 53 48 83 c2 01 49 39 d1 75 ec 48 85
```

Journal log says:
```
Dec 29 17:48:33 TV-Beneden sshd[1400]: Accepted password for root from 10.2.12.48 port 34450 ssh2
Dec 29 17:50:50 TV-Beneden sshd[1400]: Received disconnect from 10.2.12.48 port 34450:11: disconnected by user
Dec 29 17:50:50 TV-Beneden sshd[1400]: Disconnected from user root 10.2.12.48 port 34450
Dec 29 17:55:02 TV-Beneden connmand[511]: ntp: adjust (slew): +0.001349 sec
Dec 29 18:12:06 TV-Beneden connmand[511]: ntp: adjust (slew): +0.000402 sec
Dec 29 18:29:10 TV-Beneden connmand[511]: ntp: adjust (slew): -0.000103 sec
Dec 29 18:29:50 TV-Beneden kodi.sh[1278]: radeon: mmap failed, errno: 12
Dec 29 18:29:50 TV-Beneden kodi.sh[1278]: radeon: mmap failed, errno: 12
Dec 29 18:29:50 TV-Beneden kodi.sh[1278]: radeon: mmap failed, errno: 12
Dec 29 18:29:50 TV-Beneden kodi.sh[1278]: radeon: mmap failed, errno: 12
Dec 29 18:29:50 TV-Beneden kodi.sh[1278]: radeon: mmap failed, errno: 12
Dec 29 18:29:50 TV-Beneden kodi.sh[1278]: radeon: mmap failed, errno: 12
Dec 29 18:29:50 TV-Beneden kernel: kodi.bin[1278]: segfault at 0 ip 00007f2b9df02d00 sp 00007ffe799b9d10 error 4 in r600_dri.s
Dec 29 18:29:50 TV-Beneden kernel: Code: 00 00 48 8b 5c 24 50 f6 47 03 02 74 4d 4e 8d 04 36 31 f6 45 85 c9 74 34 44 89 e8 45 8
Dec 29 18:30:03 TV-Beneden kodi.sh[1274]: Segmentation fault (core dumped)
Dec 29 18:30:11 TV-Beneden kodi.sh[1274]: Crash report available at /storage/.kodi/temp/kodi_crashlog_20231229183003.log
Dec 29 18:30:11 TV-Beneden systemd[1]: kodi.service: Main process exited, code=exited, status=139/n/a
Dec 29 18:30:11 TV-Beneden systemd[1]: kodi.service: Failed with result 'exit-code'.
Dec 29 18:30:11 TV-Beneden systemd[1]: kodi.service: Consumed 47min 52.599s CPU time.
Dec 29 18:30:13 TV-Beneden sshd[1711]: Accepted password for root from 10.2.12.48 port 42824 ssh2
Dec 29 18:30:14 TV-Beneden systemd[1]: kodi.service: Scheduled restart job, restart counter is at 2.
Dec 29 18:30:14 TV-Beneden systemd[1]: Stopped kodi.service.
Dec 29 18:30:14 TV-Beneden systemd[1]: kodi.service: Consumed 47min 52.599s CPU time.
Dec 29 18:30:14 TV-Beneden systemd[1]: Starting kodi.service...
Dec 29 18:30:14 TV-Beneden systemd[1]: Started kodi.service.
Dec 29 18:30:15 TV-Beneden kodi.sh[1745]: libva info: VA-API version 1.17.0
Dec 29 18:30:15 TV-Beneden kodi.sh[1745]: libva info: Trying to open /usr/lib/dri/r600_drv_video.so
Dec 29 18:30:15 TV-Beneden kodi.sh[1745]: libva info: Found init function __vaDriverInit_1_17
Dec 29 18:30:15 TV-Beneden kodi.sh[1745]: libva info: va_openDriver() returns 0
```

### How to reproduce
The bug is almost impossible to reproduce, but does happen after about 20 minutes of watching tv.

### Information
 - LibreELEC Version: 11.0.4
 - Hardware Platform: AMD A10-7800 Radeon R7, 12 Compute Cores 4C+8G

heitbaum, 2024-01-07:
Can you please test with LE12 nightlies. If it is still happening in LE12 - and we can find a fix - it might be able to be fixed before LE12 release. LE11 will not have any further release (or release of code that could fix this r600)

oliv3r, 2024-01-24:
Sorry, I hadn't seen your reply before. I'm testing with today's LE 12 nightly now :)

oliv3r, 2024-01-25:
So far so good, though CPU usage is very high (125+ %) so it looks like video decode is not HW accelerated. So turning that on may make things break again.

mglae, 2024-01-27:
```
2024-01-26 18:03:21.229 T:1153     info <general>: CAddonMgr::FindAddons: vfs.rar v20.1.0.5 installed
```
Looks like vfs.rar compiled for LE11 is crashing in LE12. Please try to force update. (And there may be other addons).

Unfortunately there is no vfs.rar tagged for Kodi 21, LE is still shipping v20.1.0.6 compiled for LE12.

---
To the team: an improved implementation of the dropped #4370 seem still be required.

heitbaum, 2024-01-27:
> So far so good, though CPU usage is very high (125+ %) so it looks like video decode is not HW accelerated, but all are turned on.
> 
> 125% CPu usage is 4 * ~ 25%
> 
> edit: So got at least one crash/something:
> 
> ```
> [ 8662.895424] JobWorker[331993]: segfault at 1 ip 00007fc999b8a4f6 sp 00007fc8da9d99a8 error 4 in libc.so.6[7fc999b0d000+141000] likely on CPU 2 (core 2, socket 0)
> [ 8662.895462] Code: 00 00 66 90 66 0f ef c0 66 0f ef c9 66 0f ef d2 66 0f ef db 48 89 f8 48 89 f9 48 81 e1 ff 0f 00 00 48 81 f9 cf 0f 00 00 77 6a <f3> 0f 6f 20 66 0f 74 e0 66 0f d7 d4 85 d2 74 04 0f bc c2 c3 48 83
> ```
> 
> [kodi_crashlog_20240126202755.log](https://github.com/LibreELEC/LibreELEC.tv/files/14068700/kodi_crashlog_20240126202755.log)

It looks like your Player settings have DRMPrime enabled? As CDVDVideoCodecDRMPRIME is in the logs. What are the settings on the Player page?

I’m going to close this bug as not reproducible in LE12. But happy continue support on the “high cpu” on the forum. For the vfs.rar segv - a separate issue with Kodi if it is reproducible.

closed 2024-01-27 by heitbaum

oliv3r, 2024-02-01:
I've updated the rar plugin, and made it 'ask to unpack' (I really should just get rid of those archives in the library anyway).

And since, everything seems quite stable and well behaving. I didn't see any r600 crashes at least anymore.

As for the settings page, I've enabled all sliders :p

heitbaum, 2024-02-01:
> I've updated the rar plugin, and made it 'ask to unpack' (I really should just get rid of those archives in the library anyway).
> 
> And since, everything seems quite stable and well behaving. I didn't see any r600 crashes at least anymore.
> 
> As for the settings page, I've enabled all sliders :p

I as want to check are you using drmprime, vaapi, egl, … as some are “mutually exclusive”. I don’t have an amd in front of me at the moment, so not sure what it displays.