LibreELEC/LibreELEC.tv #8488: [BUG] Error: EGL_BAD_PARAMETER and eglClientWaitSyncKHR with 21.0b2-Omega

heitbaum, 2024-01-06:
### Describe the bug

Commit https://github.com/LibreELEC/LibreELEC.tv/commit/61ddb717d0c23f2f9f2e7b1f452e3ef6545d6c05 introduced a bug when coming out of screen save.

nightly-20231203-0ee24f5 works
nightly-20231206-2b7c647
nightly-20231208-5b2d3ec works
nightly-20231210-e521363 broken
nightly-20231211-5b9bf39 broken
nightly-20231217-1833aee broken

https://github.com/LibreELEC/LibreELEC.tv/compare/5b2d3ec...e521363

### How to reproduce
Steps to reproduce the behavior:
1. On main menu, wait for dim, then black screen
2. LG Screen comes up
3. Press back arrow on the remote 
4. Screen returns to dim (and previous image displays) 
5. Audio “clicking works”

### Information
 - LibreELEC Version: [e.g. 9.2.1] nightly-20231210-e521363
 - Hardware Platform: [e.g. RPi3] Generic

### Log file

### Context

heitbaum, 2024-01-06:
- Reverting https://github.com/xbmc/xbmc/pull/23921 fixed the issue
`cp projects/RPi/devices/RPi2/patches/kodi/000-temp-revert-fences.patch packages/mediacenter/kodi/patches/`

heitbaum, 2024-03-23:
User reported. Getting user to test - https://forum.libreelec.tv/thread/28349-kodi-black-screen-after-waking-from-sleep/?postID=190197#post190202

closed 2024-03-26 by chewitt

smp79, 2024-05-17:
I think this is triggered by a HDMI hotplug detect. Can you test if this fixes it?

```
diff --git a/drivers/gpu/drm/i915/display/intel_hotplug.c b/drivers/gpu/drm/i915/display/intel_hotplug.c
index d9ec349..3b4a947 100644
--- a/drivers/gpu/drm/i915/display/intel_hotplug.c
+++ b/drivers/gpu/drm/i915/display/intel_hotplug.c
@@ -973,10 +973,10 @@ static void queue_work_for_missed_irqs(struct drm_i915_private *i915)
 
 void intel_hpd_enable_detection_work(struct drm_i915_private *i915)
 {
-	spin_lock_irq(&i915->irq_lock);
-	i915->display.hotplug.detection_work_enabled = true;
-	queue_work_for_missed_irqs(i915);
-	spin_unlock_irq(&i915->irq_lock);
+	// spin_lock_irq(&i915->irq_lock);
+	// i915->display.hotplug.detection_work_enabled = true;
+	// queue_work_for_missed_irqs(i915);
+	// spin_unlock_irq(&i915->irq_lock);
 }
 
 void intel_hpd_disable_detection_work(struct drm_i915_private *i915)
```

heitbaum, 2024-05-19:
> I think this is triggered by a HDMI hotplug detect. Can you test if this fixes it?

I have tested using my dev tree (linux-6.9, Omega (Release), gcc-14.1.0) 
- (1) with 2ab68b2871fbb40f4a41e57cfc6a8c7a1bfa0e32, able to return to screen from “LG screen saver”
- (2) without 2ab68b2871fbb40f4a41e57cfc6a8c7a1bfa0e32, return from “LG screen saver” displays a dimmed kodi start screen and I can hear the “remote clicking”
- (3) With the above patch and without 2ab68b2871fbb40f4a41e57cfc6a8c7a1bfa0e32, return from “LG screen saver” is the same as (2)

smp79, 2024-08-08:
https://github.com/xbmc/xbmc/pull/25588 looks promising

PR #8761 (2024-03-26): kodi (Generic): add temp patch to revert xbmc/23921 commit
The change causes kodi to disbale the last image or a black screen when coming out of sleep.

- fixes #8488
- https://github.com/xbmc/xbmc/issues/24760
- https://github.com/xbmc/xbmc/pull/23921
- https://forum.libreelec.tv/thread/28349-kodi-black-screen-after-waking-from-sleep/?postID=190197#post190202