LibreELEC/LibreELEC.tv #8538: [BUG] Setting 'Allow remote control via Http' reset after restart

stemaj, 2024-01-21:
### Describe the bug

The setting 'Allow remote control via Http' is switched off again with every restart. This means I can no longer access it with my smartphone remote control and have to reactivate the setting manually.

### How to reproduce
Steps to reproduce the behavior:
1. Go to Settings / Services / "Allow remote contol via Http" and switch ON
2. Go to main menu and restart computer
3. Go to Settings / Services / "Allow remote contol via Http", you will see the switch is OFF

### Information
 - LibreELEC Version: 11.0.3, 11.0.5
 - Hardware Platform: RPi2, (also be seen on Windows)
 - Forum entry (in German): https://www.kodinerds.net/thread/78171-nach-neustart-einstellung-zur%C3%BCckgesetzt/

stemaj, 2024-01-21:
[kodi.log](https://github.com/LibreELEC/LibreELEC.tv/files/14002744/kodi.log)

heitbaum, 2024-03-11:
Unable to reproduce. Please test with a fresh install. Please raise this on the forum for support if still an issue.

closed 2024-03-11 by heitbaum

krisrok, 2024-03-24:
Same problem here on RPi4, persists on 11.0.6.

krisrok, 2024-03-24:
Ok, thanks to [this](https://stackoverflow.com/a/49474463) here‘s what worked for me:

Create a file `/storage/.kodi/userdata/advancedsettings.xml` with the following contents:

```
<advancedsettings>
    <services>
        <webserver>true</webserver>
    </services>
</advancedsettings>
```

**Note**: This hides the toggle from the UI but now it finally stays on.

stemaj, 2024-03-25:
Thank you very much krisrok. 🙂 it works for me too. i can live very well
with this workaround.

Cris--, 2024-09-18:
Same issue with LibreELEC-RPi5.aarch64-13.0-nightly-20240918-5b73cfc