LibreELEC/LibreELEC.tv #8549: [BUG] All Platform: global umask can lead to corrupted/not working images

the-dreamer, 2024-01-23:
### Describe the bug

For an unknown reason, my global umask was set to 077 which leads to no group or world permission in the image folder where for example all systemd conf files has only user permissions, no group, no world and most services cannot start properly. No udev, no network etc.. dmesg/systemlogs mentions missing world permission for systemd conf files. 

This will happen for all platform. I did try it for RPi and x64

I have two systems with fedora 39 and i wanted to switch to the newer system and had this issue on the newer system. I needed nearly one month to find out this issue and the root cause. My login.defs file, where the default is set, was edited in 2017 and had 077 as default umask. A more recent login.defs.rpmnew file had the default 022. With this value all files in build/image/system/ has now group and world permissions. 

It looks likes for me, there is no permission checking during the build and any umask settings can lead to unexpected behaviour and different build result on different systems. 

### How to reproduce
Steps to reproduce the behavior:

1. change umaks in login.defs / reboot system ?
2. do a build, it is not necessary to do a clean build. Image folder is rebuild always
3. check permissions with ls -All build*/image/system/etc/systemd/ folder

heitbaum, 2024-03-11:
The CI/CD Dockerfile images in use have an umask of 0022 set (default.) For a change to the build system - please raise a PR for review by the team.

closed 2024-03-11 by heitbaum

the-dreamer, 2024-03-11:
My point is not to have a different umask. 

The report ist about, if umask is different, the image will be broken/unpredictable as the umask is not set for the build folders and the build system is assuming a default umask. 

For my knowledge, a umask can be set for a subfolder. But it needs to be set. I am not very firm with the Libreelec development system to create a pr for that. As told, i even need one month to identify my issue i had with my system.