LibreELEC/LibreELEC.tv #8567: Logitech mouse and keyboard stop working overnight[BUG]

arcasys, 2024-01-31:
### Describe the bug
After a certain period of time both mouse and keyboard stop working.
Mouse Logitech M310 and keyboard K330 are paired with a Logitech Unifying receiver.
Detaching and reattaching the receiver does not help.
After a reboot everything is back to normal.

### How to reproduce
Steps to reproduce the behavior:

1.  dmesg (after reboot)
`# dmesg | grep logitech
[    4.445891] logitech-djreceiver 0003:046D:C52B.0003: hiddev96,hidraw0: USB HID v1.11 Device [Logitech USB Receiver] on usb-3f980000.usb-1.2/input2

[    4.578527] logitech-hidpp-device 0003:046D:4055.0004: input,hidraw1: USB HID v1.11 Mouse [Logitech Wireless Device PID:4055] on usb-3f980000.usb-1.2/input2:1

[    4.587362] logitech-hidpp-device 0003:046D:1024.0005: input,hidraw2: USB HID v1.11 Mouse [Logitech M310] on usb-3f980000.usb-1.2/input2:2

[    4.596918] logitech-hidpp-device 0003:046D:4016.0006: input,hidraw3: USB HID v1.11 Keyboard [Logitech K330] on usb-3f980000.usb-1.2/input2:3

[  126.925966] logitech-hidpp-device 0003:046D:4016.0006: HID++ 2.0 device connected.

[ 7456.477316] logitech-djreceiver 0003:046D:C52B.0009: hiddev96,hidraw0: USB HID v1.11 Device [Logitech USB Receiver] on usb-3f980000.usb-1.2/input2

[ 7456.612825] logitech-hidpp-device 0003:046D:4055.000A: input,hidraw1: USB HID v1.11 Mouse [Logitech Wireless Device PID:4055] on usb-3f980000.usb-1.2/input2:1

[ 7456.627803] logitech-hidpp-device 0003:046D:1024.000B: input,hidraw2: USB HID v1.11 Mouse [Logitech M310] on usb-3f980000.usb-1.2/input2:2

[ 7456.639775] logitech-hidpp-device 0003:046D:4016.000C: input,hidraw3: USB HID v1.11 Keyboard [Logitech K330] on usb-3f980000.usb-1.2/input2:3
[ 7472.470051] logitech-hidpp-device 0003:046D:4016.000C: HID++ 2.0 device connected.
'

2. Wait overnight 
Mouse and keyboard have stopped working.
dmesg will produce the same result.

3. Check with `udevadm monitor`
No evenr will be reported when the receiver is detached and reattached.  

4.  Reboot
Everything is back to normal

### Information
 - LibreELEC Version: 11.0.5
 - Hardware Platform: RPi3+

### Log file

### Context

heitbaum, 2024-03-11:
Is this issue still occurring with latest LE12 20240311-86dba4b images. If so please share the updated debug logs. Alternatively please raise this on the forum for support.

heitbaum, 2024-03-16:
Closing as inactive. Happy to open again - but will need debug logs, ideally a reproducible case. Using the forum for support may get more answers as others may have seen the same issue.

closed 2024-03-16 by heitbaum