LibreELEC/LibreELEC.tv #8583: [BUG] Dolby Vision HDR to SDR tone maping don't work on Intel UHD 600

valujin, 2024-02-03:
### Describe the bug
When playing Dolby Vision HDR content on Intel UHD 600 connected to SDR TV there is no tone mapping and image looks unwatchable. Both with HW and SW decoding.

### How to reproduce
1. Install LE12 nightly Generic build
2. Open any video file with Dolby Vision HDR

### Information
 - LibreELEC Version: LE12-devel [78a0ba1](https://github.com/LibreELEC/LibreELEC.tv/commit/78a0ba125593ca065dc1ba49dc583b0be4e6279a) Generic (GBM)
 - Hardware Platform: Dell Wyse 5070 (Celeron J4105, UHD 600)

### Log file
[pastekodi logfile](https://paste.libreelec.tv/lasting-amoeba.log)

### Context
Kodi nightly build on Ubuntu 24.04 and Fedora 39 on the same hardware doesn't have such a problem and DV HDR tone mapping works with both HW and SW decoding.
Also, I have the same Generic (GBM) build on my old SandyBridge laptop with Nvidia G210m - tone mapping works perfectly.

heitbaum, 2024-03-11:
Please raise this at https://forum.libreelec.tv/ forum for support.

closed 2024-03-11 by heitbaum

valujin, 2024-03-11:
> Please raise this at https://forum.libreelec.tv/ forum for support.

This is not a feature request, it's a bug that occurs only on Intel Graphics and only in LibreELEC. When I have spare time I'll investigate this deeper and make a PR if there will be any solution.

heitbaum, 2024-03-11:
> > Please raise this at https://forum.libreelec.tv/ forum for support.
> 
> This is not a feature request, it's a bug that occurs only on Intel Graphics and only in LibreELEC. When I have spare time I'll investigate this deeper and make a PR if there will be any solution.

There are some current threads on the forum re HDR - might have some direction on this.

heitbaum, 2024-03-12:
detail of fix/workaround @ https://forum.libreelec.tv/thread/28294-bug-dolby-vision-hdr-to-sdr-tone-maping-don-t-work-on-intel-uhd-600/?postID=189743#post189743